prusa3d/Prusa-Firmware #4135: Unable to flash firmware/bootloader on Einsy RAMBo 1.2e

michael-borck, 2023-04-09:
I'm trying to flash the most recent stock firmware on my recently bought Einsy Rambo.  Purchase fr0m Aliexpress.  The writing on board is "Einsy RAMBo 1.2e.  It has. ATMEGA2560 controller.

I initially used Prusa Slicer to flash the firmware and failed.  Issue [#2486 Unable to flash firmware on Einsy Rambo 1.1b](https://github.com/prusa3d/Prusa-Firmware/issues/2486) suggests I need to flash a custom bootloader.   Following the [instructions in #2486](https://github.com/prusa3d/Prusa-Firmware/issues/2486#issuecomment-585147008) the output suggests the signature is wrong:

      avrdude: Device signature = 0x1e9801 (probably m2560)
      avrdude: Expected signature for ATmega328P is 1E 95 0F
               Double check chip, or use -F to override this check.
 
Here a Pastebin of [Arduino IDE 2.0.4, ArduinoISP Output](https://pastebin.com/LdYTUktJ).

From the boards manager in the IDE I have install "Prusa Research AVR MK3 RAMBo EINSy  version 1.01" and "Prusa Research AVR MMU Boards version 1.06".    These were obtained from the additional board manager URL: [https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json](https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json)

Is that the correct URL or is it outdated?  If so, how do I find/use the latest URL for the Additona Board Manager in the IDE.

A couple of search's implies that I need I need to specify using a ATMEGA2560 with the following command.

    "/Users/michael/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/Users/michael/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -cstk500v1 -P/dev/cu.usbmodem14101 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m

I am now out of my depth.  I have no idea how to do this from the IDE (or command line), if it is correct,  if it will load the Prusa Research bootloader installed from the instruction above, or even if the fuse settings are correct.  I read somewhere that I should check the datasheet for the ATMEGA2560, but again out of my depth.

It could be user error (hope not).  I have tried with board powered from the Arduino UNO as well as powering the Einsy from a 24V PSU.

In case it makes a difference, I am using Arduin0 IDE 2.0.4 on an (old 2015) MacBook Pro.  I am trying to repair a MK3S.  Postage cost and exchange rate prevents me form buying an Einsy board direct from Prusa Research.  Here is a picture of the board.

[image]

Any help is appreciated.

If this is the incorrect forum, please let me know and if possible, suggest a more appropriate place I can seek help.

gudnimg, 2023-04-09:
What does your setup look like? Are you using an Arduino Uno to program the bootloader? (or something else?)

You don’t need to specify any command on a terminal as the buttons on the UI should do all the work for you.

When flashing the board, its best to use PrusaSlicer because it’s a known good setup. So we can rule out Arduino 2.0.4 (We havent tested it)

michael-borck, 2023-04-09:
If I power on the board, all I get is blank LCD screen.  I think you should see something like "Original Prusa i3 Prusa Research".  Googling somewhere suggest a blank screen means no bootloader. Hence why I was trying to install a bootloader.

I am using an Arduino Uno to program the bootloader.   I think because the board came from Aliexpress, there was no boot loader on the board.  Issue #2486 suggests without the custom boot loader is the reason flashing via the Prusa Slicer is failing.    So, I am using the Arduino to flash the custom bootloader.  Best I can tell is the flashing of the bootloader is failing because the expected chip signature is wrong.   Once I have a bootloader flashed, I can then use Prusa Slicer to update the firmware.

Do I have the process wrong?

gudnimg, 2023-04-09:
Can you double check you used “Arduino as ISP” and not “ArduinoISP”?

It looks like flashing the bootloader didnt work according to your ArduinoISP log output.

michael-borck, 2023-04-09:
Here is what is selected in the IDE menu.

 
[image]

michael-borck, 2023-04-10:
I seem to have resolved this after reading #910 and (again) following the instructions in #2486 everything is now working. I have been able to flash the bootloader, and then use Prusa Slicer to flash the latest stock firmware.  

The only difference was I used different jumper wire.  So, my best guess is the wiring was at fault, either a broken wire, faulty contact, or I incorrectly Jumper'd the two boards (and got lucky not to break either the Uno or Einsy.  

  Thanks @gudnimg for being patient with me.

closed 2023-04-10 by michael-borck