prusa3d/Prusa-Firmware #4148: [BUG] Wrong image shown on MK4 screen when printing via PrusaLink

acwright, 2023-04-15:
**Printer type** - MK4
**Printer firmware version** - 4.5.4

**MMU upgrade** - None
**MMU upgrade firmware version** - None

**Describe the bug**
Wrong image shown  on MK4 screen when printing from PrusaSlicer via PrusaLink. 
  
**To Reproduce**
1. Select MK4 physical printer setup with PrusaLink
2. Slice model using PrusaSlicer 
3. Send print to printer by choosing "Upload and Print" dialog option
  
**Expected behavior**
Printer starts printing with correct model image shown on MK4 screen
  
I have attached an image that shows the problem. As you can see I am printing an articulated octopus model and the screen shows the last completed print (Prusa keychain; printed via USB drive) instead of the current print.

[image]

acwright, 2023-04-15:
[image]

EDIT: Added another photo to show the correct image is shown in the PrusaLink http page.

mix579, 2023-04-16:
Same happens on an XL. Printing from PrusalInk shows the last USB print.