prusa3d/Prusa-Firmware #4179: [BUG] issue when compiling firmware with pt100 thermistor

nguyenhaiac, 2023-04-30:
**Printer type** - [MK3S]
**Printer firmware version** -  3.12.1

**Describe the bug**
I'm trying to compile the firmware to use it with PT100 thermistor. However, when I compile, I got this error:

`/home/hainguyen/Prusa-Firmware/Firmware/thermistortables.h:1120:1: error: narrowing conversion of '7.46e+2' from 'double' to 'short int' inside { } [-Wnarrowing]`

I've been searching and it seems I can't find a solution. Please help.
  
**To Reproduce**
Uncommenting the define PT100 line in the mk3s variant

Prusa-Support, 2023-07-04:
Have you been able to resolve the problem?

Please try again with the help of the Prusa Forum.
https://forum.prusa3d.com/forum/?wpfs=pt100
This component is not directly supported by us but several users in our official forum make use of it and may be available to help you out.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-23:
We can't confirm the presence of a bug and several users are using this mod in the community.

Please allow me to close this issue as stale and please consult the community on our [Forum](https://forum.prusa3d.com/).
You will find already existing discussions about this printer modification.

Michele Moramarco
Prusa Research

closed 2023-08-23 by Prusa-Support