prusa3d/Prusa-Firmware #4185: [BUG REPORT]  Incorrect file path in lang/config.sh for those who want to make changes to the original firmaware

jack020202, 2023-05-01:
**PRINTER TYPE** - [MK3S]
**PRINTER FIRMWARE VERSION** -  [3.12.2 with some little changes]

**DESCRIBE THE BUG**
This bug don't allow the creation of the file `Firmware-intl.hex` for multi-language Prusa firmware version 3.12.2, if you use `PF-build.sh` file.
For those who want to make changes to the original firmware.
The problem is a simple incorrect file path in the file `config.sh` (in the directory `Prusa-Firmware-3.12.2/lang/`). The `config.sh` start searching a file path named `PF-build-env-1.0.8/1.8.19-1.0.6-linux-64`, the problem is that the file start searching an old version of the file path (`PF-build-env-1.0.6/1.8.5-1.0.4-linux-64`). But the real problem is that the `fw-build.sh` file also uses this file to build the `Firmware-intl.hex` file. Without this file it is impossible to work on a Prusa printer (in my case a MK3S with some little changes) using another language, this problem only happens if you want to make changes to the original Prusa firmware.

**HOW TO FIX IT**
To fix this you need to change line number 34 in the config.sh file. From this:

`export ARDUINO=../../PF-build-env-1.0.6/1.8.5-1.0.4-linux-64 `

to this:

`export ARDUINO=../../PF-build-env-1.0.8/1.8.19-1.0.6-linux-64 `

I hope I was helpful and if you have any questions you can write me in the comments.
Thanks for the attention.

snurps, 2023-05-17:
Seams nobody read this even it's well explained and easy to fix ...
Could this issue also impact the compiled version of firmware we can download from prusa website ?(i can t select any other than english after update of the firmware)

LocutusOfHacker, 2023-05-22:
Hello, thank you for your comment on my issue.
I don't know if you upgrade your firmware by Octoprint (or by another print server).
But if I flash the firmware with Octotprint the multi-languages doesn't work.
Otherwise,  if I flash firmware by usb cable from the Prusa slicer, the multi-language works.
I hope that can help you.

snurps, 2023-05-23:
Indeed, i update it using octoprint. Will try with usb câble and if confirmed, i will report to the guy who manage thé octoprint module.Envoyé de mon mobile
-------- Message d'origine --------De : Locutus of borg ***@***.***> Date : 22/05/2023  21:30  (GMT+01:00) À : prusa3d/Prusa-Firmware ***@***.***> Cc : snurps ***@***.***>, Comment ***@***.***> Objet : Re: [prusa3d/Prusa-Firmware] [BUG REPORT]  Incorrect file path in lang/config.sh for those who want to make changes to the original firmaware (Issue #4185) 
Hello, thank you for your comment on my issue.
I don't know if you upgrade your firmware by Octoprint (or by another print server).
But if I flash the firmware with Octotprint the multi-languages doesn't work.
Otherwise,  if I flash firmware by usb cable from the Prusa slicer, the multi-language works.
I hope that can help you.

Prusa-Support, 2023-07-26:
Thanks for the suggestion, I'll make sure it reaches our developers.

I can confirm that the multi-language package of the firmware will only be installed correctly via Prusa Slicer at the moment.
As per https://github.com/prusa3d/Prusa-Firmware/issues/3144#issuecomment-835490667, tthis is a known issue.

Michele Moramarco
Prusa Researchh

3d-gussner, 2024-05-02:
Closing as not relevant anymore with cmake builds

closed 2024-05-02 by 3d-gussner