prusa3d/Prusa-Firmware #4186: [BUG] 3.12.x firmware - unload / load filament during pring PETG

TwadaCZ, 2023-05-03:
**Printer type** - [MK3S]
**Printer firmware version** -  [3.12, 3.12.1, 3.12.2]

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  When I print PETG on my MK3s, printer want unload and load filament during the print... randomly. There does not matter if it is 50 min. print or 12 hours print. It is seen during first few minutes... and before end of the print. But sometimes it need unload / load filament more times during print.

  When I print PLA, this issue was not seen (during 23 hours prints).

  If I downgrade firmware to 3.11.x, there are not problems with printing PETG. I try to print same file (gcode)... with any 3.12.x there is the problem... with 3.11.x are not any issue.

  I do not try to print another materials, because I have PLA (PLA+) and PETG only.

  I see this issue from 3.12 throught 3.12.1 to 3.12.2

  I try to send this issue to helpdesk via email. But I don't know if there is any solution coming.
  Today in any Facebook group I spoke with another guys, that have same issue after upgrading to 3.12.2... and people do not know about this issue.
  
**To Reproduce**
  Everytime when I try to print PETG with firmware 3.12.x
  
**Expected behavior**
  Print PETG as in 3.11.x firmware.
  
**G-code**

**Video**

Prusa-Support, 2023-05-06:
Thanks for your feedback.

This seems to be related to a combination of
_FW 3.12.x + MMU + filament un/loading operations_
hence a duplicate of issue https://github.com/prusa3d/Prusa-Firmware/issues/4110.

I hope you don't mind me closing this issue as a duplicate.

Michele Moramarco
Prusa Research

TwadaCZ, 2023-05-06:
Hi,
I not see on MK3s with MMU (because i did not print PETG on my second MK3s with MMU)...

This problem what I described is only on "pure" MK3s without MMU unit.

And I can reproduce repeately... when downgrade FW to 3.11.x, problem go out and printer work normally... When upgrade to 3.12.x, on the same file, same printer... problem is here...

TwadaCZ, 2023-05-06:
And after unload and load filament print will continue... Firmware not crashing... only stop the print and wait to press button to unload... and load filament again...

2 or more times during print...

with 3.12.1, 3.12.2 firmware...

I can repeat this behaviour every time when I print PETG and have 3.12.x version... from beginning of 3.12.x release...

closed 2023-11-14 by github-actions[bot]