prusa3d/Prusa-Firmware #4189: [BUG] Thermal model obstructs nozzle change procedure

staspika, 2023-05-04:
**Printer type** - MK3S+
**Printer firmware version** -  both 3.12.1 and 3.12.2
**Bug description**
  Thermal model protection almost inevitably kicks in while I am changing nozzle using metal tools, then printer emits a loud sound and restarts.
  
**To Reproduce**
  Follow normal [nozzle change procedure](https://help.prusa3d.com/article/changing-or-replacing-the-nozzle-mk2-5s-mk3s-mk3s_2069). While hotend is preheated (e.g. to 280 ˚C), use metal 17 mm spanner or adjustable wrench for holding heater block and 7 mm spanner/socket or pliers for tightening nozzle.
  
**Expected behavior**
  Thermal model protection should only be active when the hotend is **not** expected to loose heat to other objects. When changing nozzle, the hotend is by definition (according to procedure) expected to come in contact with metal tools. Thermal model should not interrupt the normal procedure. It should at least be possible to manually turn off thermal model protection from LCD screen (at least for a pre-defined period of time). Better yet, when the nozzle change procedure is activated from LCD screen, the user should be automatically presented with a choice to turn off thermal model protection for a pre-defined period of time (sufficient to perform nozzle change without being super-humanly effective).

jmd73, 2023-05-09:
You have to use the Nozzle change menu.
From the release notes for 3.12.0:
Nozzle change menu

We added the necessary Nozzle change menu option under Settings -> HW Setup which disables the temp model temporarily. Please read [Changing or replacing the nozzle (MK2.5S/MK3S/MK3S+)](https://prusa.io/nozzle-mk3s) before you start. Please also keep in mind that changing the nozzle will need the first layer adjusted, as you will never be able to tighten the nozzle at exactly the same height as the previous one. We made this change thanks to the feedback from our community testers (see https://github.com/prusa3d/Prusa-Firmware/issues/3719).

sarusani, 2023-05-10:
Use the Nozze change setting -> loosen up the nozzle -> turn off and cool down -> change what you want to change -> use the nozzle change setting -> tighten the nozzle -> turn off, cool down and tighten everything else up -> layer calibration -> done.

staspika, 2023-05-14:
Great hack, but an even better solution would be to be able to turn off the thermal model protection from LCD. How about an item in the menu "Temporary turn off thermal model protection for 5 minutes"?

Prusa-Support, 2023-07-12:
Thanks for your input.

You can loosen the nozzle and remove the plastic around your best while using the _[HW Setup - Nozzle Change](https://help.prusa3d.com/article/changing-or-replacing-the-nozzle-mk2-5s-mk3s-mk3s_2069)_ feature, and, in case of major hoted disassembly, it would be safer to take the whole hotend off and use proper tools and safety measures.

If the hotend metal parts are stuck (only if) you may want to take the hotend off and place it on a safe surface that doesn't burn (metal, stone, earthenware..). Hence use an external source of heat (heat-gun, hairdryer, oven...) and protection for your hands to dismount and clear the metal parts.

Now please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/3719.
Disabling the thermal model is not recommended and will not be allowed via the printer menu.
Advanced users can do that at their own risk via codes, especially when using third-party components - https://github.com/prusa3d/Prusa-Firmware/issues/4015#issuecomment-1481686416.

Michele Moramarco
Prusa Research

closed 2023-07-12 by Prusa-Support