prusa3d/Prusa-Firmware #4190: [BUG] Firmware crash error during MMU2S tool change with 3.12.2 5713 (1.06)

Rob70ar, 2023-05-04:
**Printer type** - [MK3S+]
**Printer firmware version** -  [3.12.2 5713]

**MMU upgrade** - [MMU2S]
**MMU upgrade firmware version** - [e.g. 1.0.6-372]

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  During a multicolor print FW crash randomly occur while tool change is in progress. Print cannot be resumed after FW crash. This error occurs only with 3.12.1 and 3.12.2 firmware. Printer works flawless when downgraded to 3.11.
  
**To Reproduce**
  Try to tune nozzle temperature during tool change. Eventually you might get the error.
  
**Expected behavior**
  No FW crash.
  
**G-code**
  It is not g-code dependant. Reproduced with several g-codes generated with latest prusa slicer version.

**Video**
  N/A

tim8315, 2023-05-15:
Same issue same hardware setting

[Slicer]  prusaslicer v2.6.0 alpha 6

FW crash during when MMU2S load filament to 1/3 of of the tube.
Not every print file occured this issue.(60% of the print files occured this issue)
But if the print file occured this issue in the first print, the FW will 100% crash everytime when the printer started to load filament to nozzle.

Can't figure out if it is the slicer gcode setting issue or MK3S+ FW bug issue or interaction.

sjbrouillard, 2023-05-16:
Recently had very similar issues, it seems to be related, so I'm adding my comment here.

I've sliced the same file with PrusaSlicer 2.5.2 as well as both Alpha-6 and Beta-1 of 2.6.0.

With firmware 3.12.1 and 3.12.2, when filament changes happen, either the MMU2s crashes with all lights flashing (no FW crash message on the printer) or a FW crash occurs. In all cases, the print can't be resumed and the filament is about 1/3 of the way down the PTFE tube. Tested about 15 prints and it consistently occurs during one of the first 5 filament changes.

Downgrade back to 3.11 and the filament changes become flawless, zero issues.

Rob70ar, 2023-06-12:
You will find Prusa's anwer on another thread. They are aware of the issue and hope to have it fixed on the next release. Meanwhile they made some suggestions to try.
In my case, I downgraded to 3.11 and print without issues. I will wait for a new release to be launched.
Nevertheless, altough my IR sensor was working fine and loading/unloading operations seemed to work most of the times, the other days trying to print Flex (TPU) I noticed using the sensor monitor option that the IR sensor was not giving consistent readings and had to calibrate it again. Perhaps it was just because of Flex material, but maybe there might have been inconsisten readings with PLA too, as other peopel relate this 3.12 errors with IR sensor calibrations.

bdaroz, 2023-07-23:
It's been months and I finally gave up on hoping for a fix - downgraded to 3.11.

A bit disappointed something this breaking has taken this long to address.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg