prusa3d/Prusa-Firmware #4198: [BUG]<Selftest Error>

SvenBrehm, 2023-05-09:
**Printer type** - [MK3S]
**Printer firmware version** -  [3.12.2]

**SD card**

**Describe the bug**
    i started calibration assistent - during selftest from fan, i get an Error about the hotend fan.
now i do an downgrade to a firmware that i know the assistent is running - the assistent works with 3.11.0 very fine.
after finishing the assistent and first layer a make and upgrade to firmware 3.12.2
after choose the language the printer started automaticly the assistent and the same selftest error is coming.
i think the printer is writing same informations about selftest information and after upgrade he take this and started the assistent again.

I think there is an bug with the assistent in firmware 3.12.2
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2023-06-29:
Thanks for reaching us.

Not sure if the description is sufficient in this case. A video may help.
Also please try to describe simple steps to reproduce the problem.

Can you confirm that an [_all data_ factory reset ](https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216)was attempted as a troubleshooting step?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-07-25:
Is this issue still relevant?
Please let us know and possibly provide a further description or even evidence, or consider closing the issue.

Michele Moramarco
Prusa Research

SvenBrehm, 2023-07-25:
Dear Michele,
yes, the problem still exists. I was supposed to do maintenance on the second printer, but I haven't done it yet because of the problem.

Mit freundlichen Grüssen / Best regards

Sven Brehm
Meister der Elektrotechnik

E-Shop für 3D-Drucke unter www.3d-print-brehm.de

M: +49 (160) 83 55 711
T: +49 (7822) 789 754 oder +49 (7822) 780 892
E: ***@***.******@***.***>

Schwarzwaldstrasse 16 | 77975 Ringsheim

Prusa-Support, 2023-08-17:
I'm sorry, we can't reproduce this issue and we haven't seen similar reports so far.
We can't confirm the presence of a bug so we need clear evidence about it.

I strongly recommend relating to our Customer Support about this as the problem may be clarified and resolved with a regular troubleshooting session.

Michele Moramarco
Prusa Research

SvenBrehm, 2023-08-17:
Dear michele,
I Order the Upgrade-Kurs for my both printers. And i hpope IT IS fixed after that.
When will you shipped IT?

Mit freundlichen Grüssen / Best regards

Sven Brehm

Meister der Elektrotechnik - für Elektro- & Gebäudetechnik

Mobil: +49 (160) 83 55 711

Festnetz: +49 (7822) 789 754

E-Mail: ***@***.***

Schwarzwaldstrasse 16 | 77975 Ringsheim

SvenBrehm, 2023-08-18:
https://www.dropbox.com/sh/1u6qdcd1orc2xbh/AAA-U6lUHhOkJTG6Zvs-LFK2a?dl=0
Hier sind Bilder und Videos – auch beim neuen Firmware 3.13.0 geht der Drucker gleich in den Selbsttest und behält diese Information im Speicher, dass der Selbsttest fehlerhaft war.
Wir kann diese Info mit dem fehlerhaften Selbsttest aus dem Speicher gelöscht werden?
Denn in den alten Firmware Versionen 3.10.0 hat er nach dem Firmware-Update keinen Selbsttest durchgeführt. Erst als ist das Firmware 3.12.2 aufgespielt hatte.

Kann der Printer auf Werkseinstellung komplett zurück gesetzt werden?

Gruß Sven Brehm

SvenBrehm, 2023-08-18:
https://www.dropbox.com/sh/1u6qdcd1orc2xbh/AAA-U6lUHhOkJTG6Zvs-LFK2a?dl=0

Hier der Link zu den Bildern und Video

Prusa-Support, 2023-09-07:
Thanks for the video.

The problem is that you were asked if _hotend fan_ (on the side) was spinning and you selected the **wrong answer**.

The _print fan_ (on the front) was spinning; the _hotend fan_ (on the side) was not spinning.
When you are asked about the _hotend fan_ (on the side) the correct answer is '**not spinning**' because the spinning one is the other.

The firmware is fine and your printer is fine 🙂 so please allow me to close the issue.
This simple misunderstanding could be solved with a video and the help of our [Customer Support](https://help.prusa3d.com/article/customer-support_2287), we have German tongue agents too.

Michele Moramarco
Prusa Research

closed 2023-09-07 by Prusa-Support