prusa3d/Prusa-Firmware #4205: Firmware crash detected while printing

0bsidianBlack, 2023-05-13:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.2

**MMU upgrade** - none
**MMU upgrade firmware version** - none

**SD card or USB/Octoprint**
  Octoprint, failed before while printing on SD-Card too though

**Describe the bug**
  While printing the printer beeps displaying the Message "Firmware Crash detected ... " Happens very sporadic, it feels more likely on longer prints. Happened twice in the past few days (thats why i reactivated my Octoprint and started to log serial).
  First crash in the past days was printed via SD Card, todays prints were printed via Octoprint.
  Today i printed the last two Parts of my Project with the same Settings, the first one finished perfectly, the second one crashed.
  
  Parts were Printed in PLA sliced with Prusaslicer 2.6 alpha3
  
  I had firmware crashes before while printing ABS but there were no crahes for a while so i thought nothing of it. 
  Those were on an older firmware and sliced with prusaslicer 2.4 so im not sure if its really the slicer or firmware.
  
**To Reproduce**
  seems random
  
**Expected behavior**
  Print not crashing
  
**G-code**
  I accidentally deleted the gcode from the first crash but here is the serial log
[print failure 1 serialLog , crash dump at 13h07min.log](https://github.com/prusa3d/Prusa-Firmware/files/11470545/print.failure.1.serialLog.crash.dump.at.13h07min.log)

Gcode from the first print today (didnt crash)
[GripLTop_0.2mm_PLA_MK3S_59m - Copy.txt](https://github.com/prusa3d/Prusa-Firmware/files/11470551/GripLTop_0.2mm_PLA_MK3S_59m.-.Copy.txt)
 
Gcode from the crashed print today
[Grip R Top_0.2mm_PLA_MK3S_1h1m - Copy.txt](https://github.com/prusa3d/Prusa-Firmware/files/11470553/Grip.R.Top_0.2mm_PLA_MK3S_1h1m.-.Copy.txt)

Serial Log of my last two prints with the crash dump at the end
[print failure 2 serialLog [crash dump at the End].log](https://github.com/prusa3d/Prusa-Firmware/files/11470554/print.failure.2.serialLog.crash.dump.at.the.End.log)

Github doesnt allow me to upload Gcode files so they are in TXT format.
The Original files are in this Zip folder:
[Gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/11470558/Gcode.zip)

gudnimg, 2023-05-14:
Looking at your log file it seems the crash happened when running `M45 Z`

[image]

Could try this again?:

1. Start a print like normal
2. Stop the print through Octoprint or whichever way you did before
3. Run M45 Z again (did you do it manually or did Octoprint do it?) Try to do it like before.

Does the crash repeat each time with the same steps?

gudnimg, 2023-05-14:
Another thought, Octoprint print sent M27 while running the calibration. I wonder if that's related. There is no reply.
[image]

0bsidianBlack, 2023-05-14:
My sincere apologies, but i forgot to add something to the previous post.
As stated the first print was printed via SD-Cardbut i didnt have my Pi installed at that time.
The serial log of the first crash is about 10 hours after the crash itself (which was in the middle of the night).
After discovering the crash at noon i setup my Pi again and dumped  the firmware crash. 
The commands in the Posts from you were send by me in octoprint to Resume the print after it failed like CNCKitchen described in [this article](https://www.cnckitchen.com/blog/guide-resuming-a-failed-3d-print) .
The Z-height was not at the right position then so i tried to offset it via M45 (which didnt work).
I think the nozzle did (or nearly) crashed into the print after so i did an manual reset via the X button at the printer display. 
At the and i didnt get resuming the print to work so i started printing it again with new Gcode (which worked).

staspika, 2023-05-14:
I also started getting sporadic "Firmware crash detected" after upgrading to 3.12.2. The printer is original stock MK3S+ (kit), no third-party modifications. In addition to sporadic, I reliably get "Firmware crash detected" every single time I select "PID calibration" from LCD menu. @0bsidianBlack, can you check if you also get the same? After downgrading to 3.12.1 the problem disappears, and reappears after upgrading to 3.12.2.

0bsidianBlack, 2023-05-14:
@staspika i had thermal anomaly issues for some time but that was just a bad hotend fan connection. i did the PID calibration a few times then and never had any issues with them.

NicholasMantovani, 2023-06-11:
I've had the same problem, my Y crash went from 0 to 62 in few prints, it happens very randomly i had prints with 0 crash and prints with 14. Everything seems fine mechanically.

0bsidianBlack, 2023-06-12:
@NicholasMantovani its not an issue with y crashes, its a firmware crash like this one 
[image]

Yesterday i successfully printed a 17h print without crashes, today it crashed on a 17 min print,
this one to be exact 
[VGA-Halterteil_0.2mm_ABS_MK3S_16m.txt](https://github.com/prusa3d/Prusa-Firmware/files/11721136/VGA-Halterteil_0.2mm_ABS_MK3S_16m.txt)

crash dump here:
[serial (1).log](https://github.com/prusa3d/Prusa-Firmware/files/11721137/serial.1.log)

Printed the same gcode again, woked fine, Rip filament

Prusa-Support, 2023-08-17:
Thanks for reporting and please keep on sending dump files.
However, a _FW Crash_ error may happen for many reasons and in completely different scenarios so we should possibly find a pattern to controllably reproduce the issue and confirm the bug.

The problem sometimes may appear seldom and may be easily fixable (temporary glitch or firmware corruption).

**Recommended actions.**
1. Downgrade and upgrade the firmware a few times to make sure the problem is only reproducible with FW 3.12.x.
2. Try a complete factory reset (for the [Einsy board](https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216) and for any other external connected device if there is any).
3. For debugging purposes, please also try other SD cards, disconnect any third-party parts (external boards e.g. Raspberry, cameras, UPS backup battery, LED stripes, other brand components e.g. motors, etc...), or even try printing with alternative methods (e.g. print some tests via Pronterface).

**If the issue persists**, possibly focus on the essential steps for reproducibility, take notes, and download the dump file.

1. Download the dump file (keep it on your computer, it may be useful later). The dump file (and warning message) will disappear after downloading. You can download it as per this article. https://help.prusa3d.com/article/crash-dump_364959
2. Possibly save a text file together with it and add relevant notes on it: FW version, date/time, how/when the FW crash occurred (last operation), personal notes...
3. If you can reproducible the problem, or it accidentally occurs multiple times, you will have multiple dump files and details for our developers to analyze. You can upload your files here.

Michele Moramarco
Prusa Research

closed 2023-11-21 by github-actions[bot]