prusa3d/Prusa-Firmware #4210: [BUG] MK3 --> MK3S+ upgrade: Greatly slowed mesh bed leveling

sp847242, 2023-05-15:
The printer is a factory-built MK3 that I upgraded to MK3S+.
Firmware is 3.12.2-5713. 
Printing from SD card. 

7x7 bed leveling, 5 samples per spot, magnet compensation turned on.

As a MK3, mesh bed leveling was decently fast: At each location, it did all five samples in very rapid succession. It would sometimes sample more slowly if there was a bit of plastic lifting the bed slightly at a sampling location.
MK3S+: Every sampling is slow, as if the entire bed is at a height the printer isn't expecting. It adds nearly a full minute to the leveling process. (Despite this, it does still print the first layer properly.)

All calibrations have been done and passed with no errors or warnings. (done twice now)

Two videos from [this Reddit thread](https://www.reddit.com/r/prusa3d/comments/12u7jl9/can_i_downgrade_a_mk3s_to_a_pinda_v2_from/) show exactly the behavior I'm seeing:

[Normal speed MBL.](https://youtu.be/-91n5ldMiqc)

[Slow MBL](https://youtu.be/_Ew0N_Z0ezI)

It does the bed-leveling process at the same slow speed whether it's started by the G28 Gcode command, or the Mesh Bed Leveling tool in the printer's Calibration menu. 

My guess is that the SuperPinda isn't _quite_ mounted at a height that the mesh bed leveling programming would be happy with, but the calibration routines disagree and say everything's fine.

leptun, 2023-05-16:
The slowdown is caused by the superpinda. It has a higher hysteresis to the measurement so it is harder to make those high speed probing moves without false measurements. That's why the XYZ calibration is also slower now. Without this slowdown, it would not be able to measure properly. This doesn't affect all superpindas, since my superpinda from a very early batch probes normally on most points and only a few need the slowdown.
As for how this could be fixed, I do not know. The only thing I noticed is that the superpinda needs to be very perpendicular to the bed. If it is even slightly angled, it will need the slowdowns much more often. Not sure how much this info helps you since for you it does the slowdown on all points.
There is a PR which might end up helping you with this issue. I am working on backporting the Print Area MBL feature from the MK4/XL so that only the area with the print is probed. This should significantly speed up the MBL for many people, and especially cases like yours since less points will be probed. See #4183 for more info.

sp847242, 2023-05-16:
On a hunch, I just tried something: I put a second heatbed on top of the first one, and did a mesh bed leveling. 

The process sped up, very nearly to normal. 55 seconds. A few measurement locations had maybe one or two samplings done slowly, but almost all of them were full speed.

leptun, 2023-05-16:
> Edit2: Rebooted the printer. Now MBL is slow again.

A reboot shouldn't change anything. Not sure how that would work.

> Can the MK3S+ firmware work with a downgrade to a regular PINDA, or does it need the SuperPINDA?

Yes, it can. The SuperPinda and PindaV2 are implemented to work on the same firmware file. So you can just swap to the other sensor, recalibrate and everything should work.

> Also: That's a neat bed-warp-visualizer there at that link.

That link is easy to use, but Octoprint has a much nicer plugin for this which works even better if configured properly.

sp847242, 2023-06-20:
Another datapoint maybe: 
I wanted to show someone the slowed MBL process on the MK3S+ printer (SuperPinda).
I turned on the cold (25°C) printer, went right to Calibration, and picked Mesh Bed Leveling.

It did the leveling in about a minute. Almost every data point was done rapidly. Only at a few spots did it slow for one of the five measurements. 

I heated it up and started a print.
It homed, and now MBL is slow like it's been - about 1m50s to do it.

Prusa-Support, 2023-08-31:
So far, after all, this doesn't seem to be a firmware bug but hardware-related.

As a recap, some aspects that may likely affect the probing speed are:
- Super PINDA is slightly more precise but possibly slower (this may vary) compared to PINDA v2,
- distance of the probe from bed (an angled probe affects the distance too),
- stability of the probe in the mount,
- Z movement accuracy,
- heatbed flatness and firmness,
- temperature of the bed (probably heat gradient too).

The firmware is not expected to improve this in the future as **the varying probing speed is a hardware requirement**.
The MBL and general calibration processes may generally improve but that's unlikely that the probing speed will intentionally be made faster at the current state of the art.
As of now, it is safe to lower the number of probings as long as the printer's geometry is precise and the first layer is accurate.

Feel free to swap PINDA probes between printers for testing but please notice one of these two printers has the trapezoidal nuts mounted upside down and this will probably offset the probing distance close to the calibration threshold or outside the accepted range.

Super PINDA and firmware both work as intended so I'd recommend closing this issue for now and this won't prevent you from keeping experimenting, reporting your findings, and sharing your thoughts.

Michele Moramarco
Prusa Research

closed 2023-11-22 by github-actions[bot]