prusa3d/Prusa-Firmware #4211: [QUESTION]<How is FW_COMMIT_NR defined>

easonjim, 2023-05-16:
Dear,
  How is FW_COMMIT_NR defined? Random characters? Or an incremental algorithm?

thank you.

leptun, 2023-05-16:
It is the number of commits in the tree of that git branch. It is automatically computed when building with cmake. Every commit that we make increments that number by one. A merge commit also increments it by one. Hope this helps.
@3d-gussner should know more details about this.

easonjim, 2023-05-16:
@leptun thanks, I found this file: ProjectVersion.cmake，inside there is build_number.

3d-gussner, 2023-05-16:
@easonjim At this moment the build number (which is based on git hub commit) isn't updated with cmake. We are working on that for the final firmware release. 
The official build server subtracts at this moment some commits. So a github commit 7313 results in 6816.

`PF-build.sh` one the other hand always uses the github commit number as FW_COMMIT_NR and this is also shown in the LCD Support.

easonjim, 2023-05-16:
@3d-gussner thanks，I got it.

closed 2023-05-16 by easonjim