prusa3d/Prusa-Firmware #4232: [FEATURE REQUEST] Add a filament drying mode for heat bed

amatulic, 2023-06-05:
[This YouTube video](https://www.youtube.com/watch?v=WC3jvuq-uq8) explains how to use the heat bed and a cardboard filament box as an effective, zero-cost filament dryer. It makes perfect sense; the heat bed can maintain a stable low temperature and the box with a few holes provides the enclosure and convection needed to pull the moisture away.

I just tried it with my MK3S. With a spool of moisture-contaminated Prusament PETG in the box on the build plate, I went into Settings, set the bed temperature to 50°C, and it worked fine until the "safety timer" shut it down.

In looking for a way to disable the safety timer, I came across issue #3778, which is closed, but unrelated to the use case of using the heat bed as a controlled-temperature filament dehydrator.

I would like to see either:
* An option to disable the safety timer **only** for the heat bed, OR
* A "filament dryer" mode available in the front panel menu, with presets for PLA, PETG, ABS, etc.

sarusani, 2023-06-05:
You can create a file with the gcode you want to run and run it from your card, no octoprint needed.

Create a new file, for example "disableSafetyTimer.gcode"
Edit the file and add M86 S0. Copy it to your SD Card and just run it like you would with a print.

cinmay, 2023-06-05:
I'm currently trying out the gcode from this model. I had to modify it a little but it seems to be working fine. 
https://www.printables.com/model/135985-diy-print-bed-filament-dryer

I agree in general that it would be cool to just have it built inn to the firmware thougt.

Prusa-Support, 2023-08-04:
Thanks for sharing your ideas.

Our developers will look into this.

However, I noticed the Prusament spool was placed on a 50°C source of heat.
Please be advised that the black Prusament spool's sides may separate from the core if placed on such a hot hatbed for some time. For more information please check the ['Drying filament' Help article](https://help.prusa3d.com/article/drying-filament_332086).

Michele Moramarco
Prusa Research

amatulic, 2023-11-15:
Issues become stale and auto-closed because Prusa is understaffed? It should remain an open issue until a PrusaSlicer developer deliberately closes it based on sound judgment or team discussion, not a bot-determined age. In all my previous jobs, issues could remain open for years if they were considered desirable to have but low priority.

There are also over 100 unresolved pull requests in the queue, some years old, and some of which have comments that suggest high community demand. Do they also get auto-closed due to being stale?

3d-gussner, 2023-11-15:
@amatulic Thanks for interact here and look at auto stale issues.
We introduced the stale bot as many tickets have not been updated for long time by owners or outdated.
  
This feature request may be something for some people but most will not use it. The SD card gcode is a good solution without using of additional resources in the firmware.

Disabling safety timers in the firmware are a sensitive topic and chances are slim to none that something like this will make it into the firmware.

Especially if there is a "controlled" and wanted way to achieve the result by a gcode file pointed out by https://github.com/prusa3d/Prusa-Firmware/issues/4232#issuecomment-1576512050

3d-gussner, 2023-11-15:
>There are also over 100 unresolved pull requests in the queue, some years old, and some of which have comments that suggest high community demand. Do they also get auto-closed due to being stale?

We don't stale bot any PRs but in the last months we started to look more into the community PRs. You can see that even very old community PRs have been merged in different releases.
There are [38 PRs without a milestone](https://github.com/prusa3d/Prusa-Firmware/pulls?q=is%3Aopen+is%3Apr+no%3Amilestone)  and we gonna look into these. Some of the PRs just not have been merged or reviewed due to resource shortages.
Lately we were able to reclaim some resources BUT still we fight for every byte used by PRs and some are higher in the priority queue than others.

Hope that explains a bit the current situation with the PRs.

amatulic, 2023-11-15:
>  may be something for some people but most will not use it.

I created this request because it was my understanding that Bambu Labs already has this feature built into their firmware. Whether people use it or not, from a business perspective it would make sense to meet or exceed the features available in competing products.

amatulic, 2023-11-15:
Thanks for your consideration and explanation of the issues involved. Based on that, and the fact that there is a workaround, I am closing this request.

3d-gussner, 2023-11-16:
@amatulic As you are the owner of the ticket, please consider to close it yourself. For me it "feels" better when WE don't close issue but the owner after discussing the topic.

amatulic, 2023-11-16:
In my last message, I selected "Close as not planned" but I didn't see that it closed. Trying again.

amatulic, 2023-11-16:
The "wontfix" tag got added, that's all. I don't see that this ticket is closed. Selecting "Closed as completed" this time.

closed 2023-11-16 by amatulic