prusa3d/Prusa-Firmware #4237: [BUG] 3.12.2 MK3S+ does not heat up before continuing a print after a crash detection

Flondythefirst, 2023-06-12:
**Printer type** MK3S+
**Printer firmware version** - 3.12.2

**MMU upgrade** - No

**SD card or USB/Octoprint**
  SD Card

**Describe the bug**
 The printer had detected a crash on the y axis while I was away for some time. When I returned I saw a prompt telling me, that a crash had been detected and whether or not it wanted to continue printing. Because everything looked fine I selected yes and confirmed.

The printer then immediately started to "print" but the hotend was cold. Therefore no plastic could extrude. After about a minute the hotend got hot enough again that plastic was able to come out of the nozzle but by then the printer had already moved up about 1 mm in the z direction and it was therefore trying to print in mid air. It then just started forming a ball of Spaghetti and I had to abort the print.   
  

  
**Expected behavior**
After a print is paused and resumed (crash detected, print pause clicked, inserting magnets or whatever) the printer should wait until the hotend reaches the correct temperature again before continuing the print.

t0mx, 2023-07-03:
Same problem here, printed from Octoprint.

gudnimg, 2023-07-15:
I am unable to reproduce this issue when using 3.13 and https://github.com/prusa3d/Prusa-Firmware/pull/4243 This PR resolves power panic behavior for paused prints and may have improved the situation for crashed prints.

For crashed prints, the firmware will raise the Z-axis by 10mm ,and even after a power panic (power outage) I am seeing the print recover in this situation when I turn the printer on again. 👍

Prusa-Support, 2023-10-13:
Thanks for reporting.
However, we are not able to reproduce the issue and suspect the recent re-work of related aspects may have addressed this anomaly.

This issue will be closed now but feel free to share further feedback.

Michele Moramarco
Prusa Research

closed 2023-10-13 by Prusa-Support