prusa3d/Prusa-Firmware #4239: "Good enough" start temptures

valkires1, 2023-06-17:
### Feature

Give the printer an option to start printing when it achieves a temperature close to the set temp start temp.
The time the bed leveling and the skirt are printed the temps would be achieved

For example the print sequence can start at 2c bed and 5c extruded below start temp (once both conditions are met/exceed).

### Justification for the feature:

Many time Ive seen the printer at temp but its a half a degree too hot or cold and on occasion ive actually seen it take up to a minute before it achieves that 0.5 degree difference and it starts.

While the bed leveling sequence is happening the nozzle temp usually can raise another 10c in that time and also if your printing on the higher end of the filament's temperature you have a large window of being able to print below listed temperature. 
And similar justification for the bed, most likely it wont deform much (if at all) with a difference from a few degrees so the change wont affect quality that much.

Plus with this being a customization option if its a concern you control when and how it happens. So you control how much of a risk of being off start temp is reasonable for your project.
In my observations just being within 2c would save probably 10-20 seconds every start of a print and those 2 degrees would be achieved the time the axises finish honing before the bed leveling sequence starts.

Prusa-Support, 2023-08-24:
Thanks for your suggestion, however, as of now, we don't recommend reducing the preheat time.

According to our observations a "long heating" is somewhat beneficial for the first layer preparation as it may help to stabilize the printer geometry and possibly slightly reduce minor temperature inconsistencies across the bed.

This is probably negligible so it would be safe for you to reduce the preheat time slightly - unless you are about to print a specifically complex and large first layer probably.

This can be done quite easily by setting a custom lower value in the start g-code for the following lines of code.
```
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

Michele Moramarco
Prusa Research

JustAnother1, 2023-08-28:
@valkires1 If the temperature over shoots (measured value above target value) or sits close to the target temperature without reaching the target temperature then your heater parameters (Kp,Ki,Kd) are not matching your printer and its environment.
How to fix that is described here: https://reprap.org/wiki/PID_Tuning

closed 2023-11-23 by github-actions[bot]