prusa3d/Prusa-Firmware #4242: [FEATURE REQUEST] Add feedrate percentage read / status support

ulope, 2023-06-22:
**Printer type** - MK*

**Is your feature request related to a problem? Please describe.**
Printerhosts like Octoprint are currently unable to show the currently set feedrate percentage since neither the various status commands (`M105`, `M155`) nor the "set" command `M220` supports getting the current value. 

This is especially true if the user changes the value on the printer directly.

  
**Describe the solution you'd like**
- Add a parameter to M220 to get and return the current value (e.g. `G`)
- Add periodic feedrate percentage reporting to M155 via one of the unused bits

3d-gussner, 2023-06-28:
@ulope Thanks, but `M220` reports back the current feedrate.

What is the purpose of this request? Is there any host system that reads the feedrate? And if so how does it read it?

```
[2023-06-28 17:06:31.004909] SEND M220
M220
[2023-06-28 17:06:31.006213] RECV 100%
[2023-06-28 17:06:31.006769] RECV ok
```

ulope, 2023-07-03:
@3d-gussner Interesting, that doesn't appear to be documented on the Prusa G-Code wiki (nor the reprap.org one).

Octoprint for example can set the feedrate but doesn't support reading it back (since it's apparently not a standard feature in any mainstream firmware).

The problem with this is that changing the feedrate therefore leads to wildly inaccurate completion estimations. 

This is additionally problematic when the user changes the value on the control knob.

Prusa-Support, 2023-09-09:
The printer's firmware recognizes the progress and remaining time thanks to the code `M73`, which is injected in the print file by Prusa Slicer every time the estimated print progress advances by -1m or +1%.
If the speed is modified via the tuning menu, the completion estimated time is altered resulting in a `?` symbol on the screen.

I'm not sure how Octoprint will process that information to calculate the completion estimation though. 

The code `M220` is listed on the dedicated Help page even though it doesn't explain its functionality went sent alone, without additional parameters.
https://help.prusa3d.com/article/prusa-specific-g-codes_112173

The documentation may use a general overhaul in the future, however, I hope you found the answers you were looking for.
May this issue be closed?

Michele Moramarco
Prusa Research

ulope, 2023-09-13:
Well partially, the current implementation would still require the host to poll the feedrate via periodic M220 commands which isn't ideal since a lot of work was done to not need polling anymore for the other relevant print parameters.

closed 2023-11-23 by github-actions[bot]