prusa3d/Prusa-Firmware #4248: Thermal Anomaly at same time in print

truginis, 2023-06-27:
**Printer type** MK3S
**Printer firmware version** -  3.12.2-5713

**Print using:** SD Card

**The issue:** Thermal anomaly error appears at the same time in the print. Other prints work fine with PETG at least. Will test other prints in PLA and report back since this is a long one. 

After searching for a while at first i thought maybe it was because of a breeze from a window or something, but now the windows are closed the temp is consistent but the thermal anomaly error appears about 45 mins in. Printing PLA generic profile 220C, 65c bed. 

Never had this problem with 3.11

truginis, 2023-06-27:
Forgot to mention. I did temp calibration and it was all good,

3d-gussner, 2023-06-28:
@truginis Thanks for the report. Please run the PID tuning and if possible try to serial log your print.
Can you share a picture of the print when you have TM error? Also a zipped gcode would help. But most important is the serial logging.
Please add `M155 S1` at the beginning to your gcode so we can see what happens to the temperatures during the print.

acompton, 2023-07-01:
I have seen a similar effect. I have a 7+ hour print that consistently runs into thermal anomalies at around the 80 minute mark. (I did try PID tuning, thermal model calibration, etc.), These persist intermittently for ~30 min, then stop and the print completes normally.  My theory is that it's a bad thermistor or wire that is affected within a specific z-range, but looking at a temperature plot shows readings all over the place but only generating thermal anomalies within that ~30 min window.. I'm going to try swapping the thermistor tomorrow to confirm.

In any case I have a couple of logs showing the behavior, which I will attach in case they are useful. Note anomalies start and stop at roughly the same times in both prints.

[Prusa_serial_logs.zip](https://github.com/prusa3d/Prusa-Firmware/files/11927107/Prusa_serial_logs.zip)

You can see the effect. Vertical red bars are thermal anomalies. Obviously the temperature is fluctuating a lot, but it isn't clear why only this section of the print is reporting anomalies.

[Figure - Temperature readings, full period](https://github.com/prusa3d/Prusa-Firmware/assets/730551/14f3e471-0fc2-4b1f-b474-8d187c24a43b)
[Figure - Temperature readings, thermal anomalies](https://github.com/prusa3d/Prusa-Firmware/assets/730551/24ba8c83-2c95-4638-be1e-2be3fca96cee)

Prusa-Support, 2023-09-01:
If the Thermal Anomaly occurs following a precise pattern such as a specific time during the print, I think the gcode may include some clues about the problem.
Is the Thermal Anomaly coincidentally occurring with the start of the print fan?

Michele Moramarco
Prusa Research

truginis, 2023-09-01:
I haven't tried printing that part again, but it also started happening on a different part. I have printed at least 8 copies of it before with no issue. No g code change, same filament, same bed and all of the sudden it started failing. I have not noticed if it starts with the fan starting but I will look for that next time I try printing. Planing on printing more this weekend so I will get back to you.

closed 2023-11-23 by github-actions[bot]