prusa3d/Prusa-Firmware #4252: [BUG]FANCHECK Cannot be disabled by simple firmware configuration (or the ostensible LCD menu setting)

PeterGribbin, 2023-06-28:
Printer type i3 MK3S+ Firmware affected 3.11, 3.12.2, 3.13 

**Problem:** When fancheck is disabled using the LCD menu settings, the PWM speed still pulses to 100% for 100ms every 5000ms, demonstrating that FANCHECK is still active. This behaviour is documented in [issue 1516.](https://github.com/prusa3d/Prusa-Firmware/issues/1516) This 100% speed bursts are a problem for me as it increases noise of by berd air pump, reduces motor life and causes inconsistent cooling. 

**What I tried, which failed:** 
1. uncommenting debug_disable_fancheck from 1_75mm_MK3S-EINSy10a-E3Dv6full.h. No effect on behaviour.
2. commenting out #define FANCHECK from 1_75mm_MK3S-EINSy10a-E3Dv6full.h. firmware fails to compile

**What I tried which worked:**
Change fan_check_period variable from 5000 to 86400000. problem solved, no 100% speed bursts every 5 seconds.

**What is needed:**
Fancheck should be fully disabled when the user sets it to disabled within the LCD menu. I was unable to figure this out.

sarusani, 2023-06-29:
Uncommenting #define DEBUG_DISABLE_FANCHECK only works if you also uncomment #define DEBUG_BUILD (it's a few lines above, at the start of the block).

I didn't test if that will successfully disable Fancheck, but it should work in theory.

Of course that still leaves the menu entry not behaving as expected.

c---, 2023-08-15:
No reason to enable all the debugging stuff. This is what I did for my MK3:
```diff
diff --git a/Firmware/variants/MK3.h b/Firmware/variants/MK3.h
index 78ee31d0..43d100bd 100644
--- a/Firmware/variants/MK3.h
+++ b/Firmware/variants/MK3.h
@@ -127,7 +127,7 @@
 #define UVLO_SUPPORT
 
 // Fan check
-#define FANCHECK
+#undef FANCHECK
 
 // Safety timer
 #define SAFETYTIMER
@@ -168,6 +168,7 @@
 //#define DEBUG_BUILD
 //#define DEBUG_SEC_LANG   //secondary language debug output at startup
 //#define DEBUG_XFLASH   //debug external spi flash
+#define DEBUG_DISABLE_FANCHECK     //disable fan check (no ISR INT7, check disabled)
 #ifdef DEBUG_BUILD
 //#define _NO_ASM
 #define DEBUG_DCODES //D codes
```

Notice where I put the `#define DEBUG_DISABLE_FANCHECK`, **above** the `#ifdef DEBUG_BUILD`.

Seems to work.

PeterGribbin, 2023-09-09:
Thanks C---, I tried your suggestion but the mk3s 3.13.1 firmware compile fails (built in Windows with cmake in vscode) due to missing fan check definitions.  

I am hesitant to enable debug build as per sarusani's suggestion as I don't want verbose serial writes potentially slowing down my printer. So, I am still using my hack of just setting the fan check timer to a large number to disable it.

c---, 2023-09-09:
@PeterGribbin Sorry about that. Yeah, they added the Alt Fan support without accounting for the fan check setting.

Regardless, this additional patch should make it work for your MK3S although I'm not sure if there could be other things related to the Alt Fan support that may not work correctly. A Prusa developer needs to fix/look at this stuff.
```diff
diff --git a/Firmware/fancheck.cpp b/Firmware/fancheck.cpp
index eedc5fe3..90409277 100755
--- a/Firmware/fancheck.cpp
+++ b/Firmware/fancheck.cpp
@@ -224,7 +224,9 @@ bool extruder_altfan_detect()
 
     EIMSK &= ~(1 << INT6);
 
+#if (defined(FANCHECK) && defined(TACH_0) && (TACH_0 > -1))
     countFanSpeed();
+#endif
 
     // restore fan state
     altfanStatus.isAltfan = fan_speed[0] > 100;
```

closed 2024-10-15 by github-actions[bot]