prusa3d/Prusa-Firmware #4259: Thermal runaway

triton-nereide, 2023-07-05:
I am running 2 Prusa Bears, with two Raspi 4b 8GB attached to them, every one features their own octoprint / octodash install with a Pimoroni4 SQ each. One of them now runs for about two years, the other one was completet two days ago.

Firmware 1.12.2

right after installation of the new firmware I get sometimes / not always sequences of loud beeps that indicate thermal runaway. The temperature of the printhead changes maybee 1° or 2° at a maximum, while the temp of the bed does not change @ all. The graph in Octoprint is showing the same fluctuations like before with other firmwares, all about 1-2° of temp change max. while printing.

I want to silence only the beep, not deactivate the thermal model, because I watch the screens / octoprint anway, until the thermal model works.

triton-nereide, 2023-07-05:
Correction: Firmware: 3.12.2

3d-gussner, 2023-07-06:
@triton-nereide Please run PID tuning and Thermal Model cal. ideally also enable the serial logging in Octoprint during the calibrations.

triton-nereide, 2023-07-06:
O.K., thanks. the termal model calibration was the first thing I did after upgrading to 3.12.2, but I missed out on PID tuning and did not enable serial logging...
So I will try one more time!

triton-nereide, 2023-07-10:
So now I´ve completed PID tuning and thermal calibration, here´s the log...
[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/12004370/serial.zip)

DRracer, 2023-08-01:
@triton-nereide would you remove batteries from your smoke detector? There is something electrically wrong with your printer and downgrading to a previous firmware won't solve it. But it's your choice of course.

triton-nereide, 2023-08-01:
@DRracer:

how do you come to the conclusion, that something is electrically wrong with my printer? Did the log file give you this information? What did it tell you is "electrically wrong" with my printer? 
Thanks for answering to my questions...

paulhayes, 2023-08-22:
I'm also getting a thermal runaway error during printing after upgrading the firmware. It seems intermittent, but happens about once every hour of printing. Wish I hadn't updated the firmware as it's been running well for a couple of years.

gouryellas, 2023-10-28:
> @triton-nereide would you remove batteries from your smoke detector? There is something electrically wrong with your printer and downgrading to a previous firmware won't solve it. But it's your choice of course.

lol. I downgraded to 13.11.x and it completely solved my thermal issues i had on 13.12.x and >

closed 2024-01-04 by github-actions[bot]