prusa3d/Prusa-Firmware #4266: [QUESTION] Can I change the Z-Offset via Gcode?

tdortton, 2023-07-17:
## Pretext
For my bachelor thesis Im working on automating a Prusa MK3S+. I have installed a MMU2S for material selection and a bed changing mechanism to enable continuous printing. The criteria is that I want to use both smooth and textured printing beds, depending on the selected material without having to manually change the Z-Offset. I’ve been looking into a way to set up different printer profiles for the corresponding print surface, with the goal to send the correct z-offset to the printer during the start Gcode.

## Questions
Is there a clean way or a workaround to set the z-offset via gcode?

## What I tried already
- I have been experimenting with the **M851** command in the Start-Gcode, to define the correct z-probe offset for each printing surface. During testing it did not change the current z-offset. I suspect it overwrites the manual set value from the EPROM. Here is the snippet from the Start-Gcode

~~~
<…>
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

;----------------------
;--- Enter Z-Offset ---
;----------------------
; Smooth Sheet
M852 Z-1.1; set Z-Probe-Offset
;----------------------
<…>
~~~

- Tried to manually offset the Z-Position of the printhead with G92 Z<x.xx> in the start Gcode, where I entered the difference between the two calibrated offsets from the sheets. I was expecting this to work, but also got no satisfactory result.

I appreciate all the tips I can get!
Thanks alot!

gudnimg, 2023-07-17:
`M851` is not supported on the MK3 printers and is only meant for Auto Bed Leveling. Auto Bed Leveling is not the same as Mesh Bed Leveling. We've been thinking about removing this code actually to avoid confusion like this as we don't intend to maintain this code.

In the official firmware released by Prusa Research, the gcode `M851` is actually not apart of the firmware (you need to enable it yourself and compile the firmware) see `#define ENABLE_AUTO_BED_LEVELING` in the code. I suspect the code doesn't compile anymore and I don't see it being used in the planner code.

If I were you I'd look into these functions:
`lcd_babystep_z` -- this is the LCD menu which allows users to tune the baby stepping, you can see how the EEPROM is written to
`babystep_load()` -- reads the value from EEPROM
`babystep_apply()` -- calls `babystep_load()` and uses the value from EEPROM
`babystep_undo()` -- reverts the action done by `babystep_apply()`

You could "patch" the firmware by rewriting the M851 gcode to do whatever you want :) 

There is some debugging code you could enable in `babystep_load()`

```cpp
#if 0
    SERIAL_ECHO("Z baby step: ");
    SERIAL_ECHO(babystepLoadZ);
    SERIAL_ECHO(", current Z: ");
    SERIAL_ECHO(current_position[Z_AXIS]);
    SERIAL_ECHO("correction: ");
    SERIAL_ECHO(float(babystepLoadZ) / float(axis_steps_per_unit[Z_AXIS]));
    SERIAL_ECHOLN("");
#endif
```

Hopefully this helps 👍

tdortton, 2023-07-18:
Thanks for the fast answer and the suggestion. I will look into it and see if this is a viable solution.

inovatorius, 2023-07-30:
I strongly believe that there should be an option to adjust the z-offset directly from the GCODE. The current options provided, such as "Live adjust Z" and "Z Offset" under Printer Settings -> General -> Size and coordinates, are not always effective.

For instance, I am using this guide (https://blog.prusa3d.com/how-to-print-on-t-shirts_55588/) to print on t-shirts. To achieve the desired z-offset, I initially reset it to +0.0000 from the printer menu, but then, before each print, I have to manually set it to around -0.5000. This repetitive process becomes cumbersome as I have to reset and adjust the z-offset each time, or add another -0.5000 to the existing value.

Unfortunately, the options currently available do not allow for automation. Therefore, we need alternative methods to set the z-offset according to individual preferences, rather than being restricted to Prusa's predefined options.

3d-gussner, 2023-11-23:
We have [implemented](https://github.com/prusa3d/Prusa-Firmware/pull/2201) `M850` to read and set Sheet names and the live Z offsets.

The Gcode is case sensitive so please  ensure that you don't use a serial terminal that UPPERCASE automatically, like Pronterface (PLEASE don't use it at all), OctoPrint. In OctoPrint you can add `M850` to `Terminal Auto Uppercase Blacklist` in Settings -> Features -> Features

Set sheet data 
- S[id] = Sheet ID  0-7
- Z[offset] 
- L[label] 
- B[bed_temp] 
- P[PINDA_TEMP] 
- A[IS_ACTIVE]

Examples:
Sending `M850 S0` to read
Results look like this 
```
Sheet 0 Z-0.8200 R-8 LSmooth1 B60 P25
A1
```
Sending `M850 S1 Z-0.800 B60 P25` will set the 2nd sheet to Z offset `-0.800` with Bed temp of `60`°C and PINDA temp of `25`°C

Sending `M850 S1 A1` will switch the active sheet to the 2nd sheet.

Be careful with setting the sheets via Gcode as this may cause issue.

Side note we will update the documentation and report in the next release.

3d-gussner, 2023-11-24:
@tdortton Please let us know if the question has been answered and consider to close this issue.

tdortton, 2023-11-26:
Thanks for implementing this in the code. I'm looking forward to testing it with the next release.
I appreciate the effort you and the team put into this.