prusa3d/Prusa-Firmware #4268: [QUESTION]<Enter comprehensive title>

Skinnedknuckles49, 2023-07-19:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**
With a dual head factory assembled Prusa XL running firmware 4.7.1, when I connect a Rasberry Pi running Octoprint I received a Safety Warning which refers me to https://community.octoprint.org/t/octoprint-tells-me-that-my-printers-firmware-lacks-mandatory-safety-features-what-does-this-mean/350.

Basically they are saying "it means that it detected a firmware running on your printer that is known to lack or have broken implementations of mandatory safety features such as thermal runaway protection."    It does mention that the firmware may have the protection and the implementation just be incompatible with their check.

Since I am confident that the XL implements thermal runaway protection, it would seem that Octoprint's test is not compatible with Prusa's firmware.  I can disable this check in Octoprint but that l doesn't "feel right".

I tried to search the issues database and did not find the issue mentioned.

gudnimg, 2023-07-19:
Hi 👋 XL and MK4 issues are located here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

This repository is only for the 8-bit printers (MK25/MK3/MK3S etc)

Skinnedknuckles49, 2023-07-19:
My apologies for the error.  I have re-posted in the correct area.

 

Thank you for your patience.

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

closed 2023-07-20 by gudnimg