prusa3d/Prusa-Firmware #4284: Please provide the source for the MMU2S firmware v3.0.0

nsheridan, 2023-07-27:
Per title, the MMU firmware binary is available for download from this repo but I don't see a source download.

I'm assuming the MMU 3.0 firmware is licensed under a GPL license, if this is the case can you please

1. clarify the license, as the release does not include one.
2. Provide a copy of the source code, or point me to the location of where it is published.

Thanks!

DRracer, 2023-07-28:
The source code is coming, stay tuned.

Itrel, 2023-08-15:
Where can I download MMU2S version 3.01 to use with the latest development build of MK3S? ... Or obtain the source so I can compile it myself :).

jfestrada, 2023-08-19:
@DRracer do you have any estimation of release date for the source code of the version 3.0.0?

gudnimg, 2023-08-20:
If you want to build the current MK3 branch but use 3.0.0 MMU FW, then you only need to do two changes:

Change:
`static constexpr char MSG_DESC_FW_UPDATE_NEEDED[] PROGMEM_I1 = ISTR("MMU FW version is incompatible with printer FW.Update to version 3.0.1."); ////MSG_DESC_FW_UPDATE_NEEDED c=20 r=8`
To this:
`static constexpr char MSG_DESC_FW_UPDATE_NEEDED[] PROGMEM_I1 = ISTR("MMU FW version is incompatible with printer FW.Update to version 3.0.0."); ////MSG_DESC_FW_UPDATE_NEEDED c=20 r=8`

Change:
`static constexpr uint8_t mmuVersionPatch = 1;`
To this:
`static constexpr uint8_t mmuVersionPatch = 0;`

It should work fine for testing purposes.

jfestrada, 2023-08-20:
In my case, I want to compile the 3.0.x for my mmu of 10 filaments.
[image]

xaionaro, 2023-09-13:
> The source code is coming, stay tuned.

Any estimates on "when"? :)

jfestrada, 2023-09-13:
I made my 10 colors mmu works on the 3.13.1 fw using the 2.6.X fw for the MMU which is located on the Prusa-Firmware-Buddy repository. For that I changed the version informed to 3.0.0 so the einsy firmware accept it. 

I do not understand why @josefprusa @DRracer or the person in charge to release the code didn't do it already. They are selling the MMU3 which is hardly based on the modifications/improvements developed by other people and released on printables.

nsheridan, 2023-09-13:
I'm not even sure I can request the source because license is unclear - it's not even clear that the MMU source as published even has a license, per https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3170 and that issue has gone totally ignored. 

I suspect it's still GPL but I haven't compared the pre-3.0 sources to the current to see if the code was reused or if 3.0 is a from-scratch rewrite. 

If the GPL license does apply to the 3.0 firmware then please release the full source as per the terms of the license. If it does not apply then please clarify the license under which the firmware is released. I don't understand the delay here.

schwabe, 2023-09-13:
If only Prusa has contributed to the firmware, they can also just decide not to release it under GPL anymore. If that is the case, they are in their full right not to release the software sources at all. I am not saying that is good or bad decision I am just saying it is in their right to relicense software that they have full ownership of.

Don't get me wrong. I would also love to tinker with the source code. I want  to see if I can hack voltage reporting of the MMU3 for those of us without PD buddy board into the firmware and octoprint.

jfestrada, 2023-09-14:
I don't really know it for sure, but the source code of the version 2.6.x is released on the Prusa-Firmware-Buddy which is licensed under the GPL. 

[image]

Anyway PrusaResearch could do whatever they want, but they are getting benefit from the people that published their modifications, I know that is legal and there is no problem, but if you as a company are getting those benefits the expected is that your company contribute as well.

arekm, 2023-09-14:
https://github.com/prusa3d/Prusa-Firmware-MMU

so issue can be closed.

closed 2023-09-15 by 3d-gussner