prusa3d/Prusa-Firmware #4285: MMU2S with MK3S+ does not work

Julasch17, 2023-07-27:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.0

MMU2S with MK3S+ does not work, MMU2s makes strange noises and MK3S+ says disable or turn off

gudnimg, 2023-07-27:
We need more info. 

What does the LCD show? It should show an error message.

And can you please explain more what you mean about _strange noise_?

gudnimg, 2023-07-27:
Only three error messages currently show the "Disable" button.

1. `FW_UPDATE_NEEDED` -- May happen if you update the printer first, fixed by updating the MMU and then restarting the printer
2. `MMU_NOT_RESPONDING` -- Could be a poorly seated cable
3. `COMMUNICATION_ERROR` -- Could be a poorly seated cable or potentially something else

Is the LCD showing one of these?

Julasch17, 2023-07-27:
[image]
[image]
[image]

Julasch17, 2023-07-27:
https://github.com/prusa3d/Prusa-Firmware/assets/140762107/fad45cc8-da9e-467a-a7a6-8779fc67874b

gudnimg, 2023-07-28:
@Julasch17 Oh I know what this is 👀 , you need to change the stallguard threshold value (or in other words sensitivity of the idler homing). Thankfully we can now do this via G-code on 3.13.0 :)  

The default sensitivity value is `7` it was chosen as the best value for MMU3 assembly specifically but some MMU2S units may need a lower value. **My MMU2S needed value `5` to work perfectly**.

So try see if either of these work better for you (The value after `X` as the new sensitivity value):

* `M708 A0x19 X5`
* `M708 A0x19 X6`
* `M708 A0x19 X7` (the default value)

Values above `7` or lower than `5` are not recommended but possible.

This value will stay persistent so you only have to change it once when you find the value which works best.

Hopefully this helps!

Julasch17, 2023-07-28:
ok, but where can I set or change this ?

gudnimg, 2023-07-28:
You need to connect a computer to the printer's USB connector and open the serial port with Baudrate 115200 and send the gcode command manually.

I usually find Arduino IDE to be easiest to use. But other software like https://www.pronterface.com/ or PuTTy or TeraTerm should also work. If you have Octoprint that can work also.

On Arduino 1.8.19 I just find the USB port in `Tools -> Port` and then click on this icon in the top right to open the serial port:
[image]
From there I just copy-paste the gcode command and hit Enter to send it.

Julasch17, 2023-07-28:
[image]

there I now enter M708 A0x19 X7 ?

gudnimg, 2023-07-28:
Exactly, Try first `M708 A0x19 X6` -- and restart the printer. If the homing is still sounding bad, try `M708 A0x19 X5`

If you want to revert the changes back to default then use `M708 A0x19 X7`

Julasch17, 2023-07-28:
M708 A0x19 X5

https://github.com/prusa3d/Prusa-Firmware/assets/140762107/76ad2edc-6ff2-43b2-87c0-1f0ec4f7199c

gudnimg, 2023-07-28:
Hmm `M708 A0x19 X5` only makes changes to the Idler 🤔 I have not seen this issue with the Selector before.

Does the Idler homing still fail like before?

Julasch17, 2023-07-28:
> Schlägt die Leerlaufsuche immer noch fehl wie zuvor?

yes

beardedbrewguy, 2023-07-28:
I too am seeing the same problem with homing. I had to revert my firmware to the previous version.

Julasch17, 2023-07-28:
> 
M708 A0x19 X76 ?

gudnimg, 2023-07-28:
No `M708 A0x19 X7` :)

gudnimg, 2023-07-28:
> I too am seeing the same problem with homing. I had to revert my firmware to the previous version.

Hi Usually tuning the default sensitivity value has worked for me.

Just to explain what the relevant value means:

The higher the value (which is by default 6), the less sensitive the Idler is to motor stalling. Higher sensitivity should make the homing more silent, but if the value is too low then the idler can't home due to being too sensitive.

[image]

It is usually enough to try one value below the default or one value above it. We added a new G-code in 3.13.0 to tune various parameters in the MMU (M708): https://reprap.org/wiki/G-code#M708:_Write_to_MMU_register

* `M708 A0x19 X7` (One **above** 6)
* `M708 A0x19 X5` (One **below** 6)

beardedbrewguy, 2023-07-28:
> Tightening of the two idler screws should be checked. I have a clip-on [idler body from printables](https://www.printables.com/model/124431-idler-body-for-mmu2s-with-some-extra-features) and at first (a few months ago when I installed a self compiled version) it made the same sounds and blocked. Changing the tightening resolved the issue for me.

In my case I removed the two screws entirely just to check if it would home with no pressure and I get the same sound. I'm about 40 minutes from completing a print and then I'm going to test changing the sensitivity values.

Julasch17, 2023-07-28:
start
echo: 3.13.0-6873
echo:MMU2:>S0*c6.
ResetRetryAttempts
SpoolJoin is On
echo: Last Updated: Jul 24 2023 12:44:59 | Author: (none, default config)
echo: Free Memory: 2517  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Hotend fan type: NOCTUA
CrashDetect DISABLED
Sending 0xFF
echo:SD card ok
// action:dump_available
echo:MMU2:<S0 A3*22.
echo:MMU2:>S1*ad.
echo:MMU2:<S1 A0*34.
echo:MMU2:>S2*10.
echo:MMU2:<S2 A0*4f.
echo:MMU2:>S3*7b.
echo:MMU2:<S3 A32e*17.
echo:MMU2:>Wb 5*33.
echo:MMU2:<Wb A*5c.
echo:MMU2:>W14 14*94.
echo:MMU2:<W14 A*52.
echo:MMU2:>f0*21.
echo:MMU2:<f0 A*c5.
echo:MMU2:>Q0*ea.
echo:MMU2:<X0 F0*87.
echo:MMU2:>R8*81.
echo:MMU2:<R8 A0*8d.
echo:MMU2:>R1b*9e.
echo:MMU2:<R1b Aff*bf.
echo:MMU2:>R1c*88.
echo:MMU2:<R1c Aff*60.
echo:MMU2:>R4*7b.
echo:MMU2:<R4 A32*b5.
echo:MMU2:>R1a*f5.
echo:MMU2:<R1a A0*41.
echo:MMU2:>Q0*ea.
echo:MMU2:<X0 F0*87.
echo:MMU2:>R8*81.
echo:MMU2:<R8 A0*8d.
echo:MMU2:>R1b*9e.
echo:MMU2:<R1b Aff*bf.
echo:MMU2:>R1c*88.
echo:MMU2:<R1c Aff*60.
echo:MMU2:>R4*7b.
echo:MMU2:<R4 A32*b5.
echo:MMU2:>R1a*f5.
echo:MMU2:<R1a A0*41.
echo:MMU2:>Q0*ea.
echo:MMU2:<X0 F0*87.
echo:MMU2:>R8*81.
echo:MMU2:<R8 A0*8d.
echo:MMU2:>R1b*9e.
echo:MMU2:<R1b Aff*bf.
echo:MMU2:>R1c*88.
echo:MMU2:<R1c Aff*60.
echo:MMU2:>R4*7b.
echo:MMU2:<R4 A32*b5.
echo:MMU2:>R1a*f5.
echo:MMU2:<R1a A0*41.
echo:MMU2:>Q0*ea.
echo:MMU2:<X0 E8087*6e.
echo:MMU2:>R8*81.
echo:MMU2:Command Error, last bytes: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58
echo:MMU2:SELECTOR CANNOT HOME

mmoraes80, 2023-07-28:
I tried X5, X6 and X7 and the problem persists... The idler still makes a very loud noise and keeps trying to home up to 3 times...

What was the original value in the FW 1.0.6?

beardedbrewguy, 2023-07-28:
I tried all three values as well and the problem is still there.

pOmelchenko, 2023-07-28:
I had same problem, but for me problem gone after factory reset

https://help.prusa3d.com/ja/article/factory-reset-mmu2s_1920

gudnimg, 2023-07-28:
@petrubecheru `M708 A0x19 X6` should revert the change. 6 is the default (I know I mistakenly said 7 before I looked at the code)

The value is stored in EEPROM so it will be persistent.

EJTH, 2023-07-28:
I cant help but notice that you have modded the unit, so that the idler has to collide with some number things,  I bet that the collision detects as the end of travel. Try to remove mods from the unit and get it to work stock before you add your own points of failure.

gudnimg, 2023-07-28:
@petrubecheru huh that is super weird. We'll look into it. It shouldn't have made any difference.

gudnimg, 2023-07-28:
@petrubecheru Just adding, you can read the currently used value with `M707 A0x19` and the MMU should report the value via Serial. https://reprap.org/wiki/G-code#M707:_Read_from_MMU_register

Julasch17, 2023-07-28:
> Verwenden Sie zufällig Pronterface? `

no, Arduino I use

ljford7, 2023-07-28:
I finally got it to work with M708 A0x19 X6, but now have another issue. 

The lights on the MMU are not working when switching to different filament slots via the buttons on the MMU. They are all off and should be highlighting which slot is selected.

gudnimg, 2023-07-28:
> I finally got it to work with M708 A0x19 X6, but now have another issue.
> 
> The lights on the MMU are not working when switching to different filament slots via the buttons on the MMU. They are all off and should be highlighting which slot is selected.

Could you please open a separate Github issue for that? It helps us track the various requests we may get these next days.

thomasloven, 2023-07-28:
I had the same problem.
Managed to get it working with `M708 A0x19 X6` through arduino IDE.
It seems octoprint has the same problem as pronterface and capitalizes the first X.

After that I also got the problem with selector homing, but a reset of the MMU2 fixed that one. Seems to be working well since then.

gudnimg, 2023-07-29:
I've corrected an error in my comments above where I said value 7 is default, changed the comment to say 6 instead to avoid confusion.

DesC21, 2023-07-29:
After changing to 5, the MMU is finally quiet... Great stuff (it's the little things ;-))

Rob70ar, 2023-07-29:
I have the same problem and could not fix it yes :-(

mlpriv, 2023-07-29:
also the same problem - sending gcode (M708 A0x19 X5 or other values) doesn't help. Flashed mutliple times back to 3.12.2 / 1.06 and factory resets and flashed again 3.13 / 3.00 without success. Back to 3.12.2 /106 and all works!
With 3.13 /3.00 nothing - always get "idler cannot home https://github.com/prusa3d/Prusa-Firmware/issues/4125"

tipirtti, 2023-07-29:
I had also same problem.
Arduino IDE & "M708 A0x19 X5" did the magic for me. 
Now it works.
Pronterface couldn't set the new value.

johnnyruz, 2023-07-30:
Testing on my MK2.5S with MMU2S - "M708 A0x19 X7" worked for mine

Mulle00, 2023-07-30:
> also the same problem - sending gcode (M708 A0x19 X5 or other values) doesn't help. Flashed mutliple times back to 3.12.2 / 1.06 and factory resets and flashed again 3.13 / 3.00 without success. Back to 3.12.2 /106 and all works! With 3.13 /3.00 nothing - always get "idler cannot home #4125"

I have the same problem, I can't manage to get the idler right. Even with open idler it does not succeed.

I did the following:
* Send the code with OctoPrint : M708 A0x19 X5
* Flash firmware to 1.0.6
* Factory reset
* Flash firmware to 3.00
* Same problem

With X7 also same problem.

When I send X5, I get back the following reply:
Recv: echo:MMU2:>R19*48.
Recv: echo:MMU2:<R19 A13*52.

Prusa-Support, 2023-07-30:
Hello, thank you for the reports. While I see that the fix with GCode M707 worked for some of you, I see that it wasn't resolved for everyone, the unresolved cases can point to a possible mechanical cause. 

I'd recommend checking the aspects in this article https://help.prusa3d.com/article/idler-cannot-home-04125-mmu_395865, and if anyone still has issues after these checks, of course feel free to send more feedback here in this thread, but also please let me know via email info@prusa3d.com, you can mention my name and this GitHub thread in the email and we can more directly check case by case in more detail, that way we can make sure that all the cases are solved. 

Alessandro Pantaleo
Prusa Research

Mulle00, 2023-07-30:
Hello,

It still doesn't work for me:

The idler can move freely with the hand up to the stops as I can feel.
The idler is also nicely aligned with the motor and shaft. I disconnected the motor to check this.
There is nothing to block the idler.

See video with what I mean:
[https://photos.app.goo.gl/hXnZRiiSC1FgsEzN9](url)

mlpriv, 2023-07-30:
At my side "M708 A0x19 X4" does works perfectly - no rattling, no idler home failures.
X4 - no rattling - idler homing always success
X5 - little more rattling bot works
X6 - more rattling and idler home fail
X7 - most rattling and idler home fail

Mulle00, 2023-07-30:
The problem was solved with the command M708 A0x19 X5.

**ATTENTION : sending via OctoPrint does not work !!!**

With the Arduino IDE it does work

sarusani, 2023-07-30:
> The problem was solved with the command M708 A0x19 X5.
> 
> **ATTENTION : sending via OctoPrint does not work !!!**
> 
> With the Arduino IDE it does work

Just as an info, you can use Octoprint if you add M708 to the "Terminal Auto Uppercase Blacklist" in Setting->Features.

jayree, 2023-07-30:
@Prusa-Support @gudnimg my findings and observations so far: 

original state (right after the firmware update): buzzes 3 times very loudly

`M708 A0x19 X5` looks like a good value
`M708 A0×19 X6` buzzes gently while reaching the end stops
`M708 A0x19 X7` sounds like the original state. Which suggests that this value is set somewhere in the firmware.

other abnormalities:
- there is no difference between the MMU `stealth` and `normal` mode (it sounds like `normal` is always used
- LEDs are off

To successfully connect to the printer using the Arduino IDE, PrusaLink must be deactivated.

GitPetri, 2023-07-31:
Just to be sure I

- Downgraded both MK2.5S and MMU2S firmware to previous versions
- Factory-resetted the MMU
- Verified that the MMU works (well at least to the quality level that it has worked since beginning i.e. glitching)
- Upgraded MK2.5S to 1.13
- Waited for the printer to complain about incompatible MMU firmware
- Disabled MMU
- Upgraded MMU to 3.0
- Listened the horrible sound when the idler attemps to self-suicide
- Enabled MMU from printer menu
- Listened the horrible sound when the ider attemps to self-suicide
- Gave up and downgraded the firmwares

This seems to be the comms data on the MMU idler failure:
`Recv: echo:MMU2:>Q0*ea.
Recv: echo:MMU2:<X0 E8107*df.
Recv: echo:MMU2:>R8*81.
Recv: echo:MMU2:Command Error, last bytes: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58
Recv: echo:MMU2:<R8 A0*8d.
Recv: echo:MMU2:>R1b*9e.
Recv: echo:MMU2:<R1b Aff*bf.
Recv: echo:MMU2:>R1c*88.
Recv: echo:MMU2:<R1c Aff*60.
Recv: echo:MMU2:>R4*7b.
Recv: echo:MMU2:<R4 A0*66.
Recv: echo:MMU2:>R1a*f5.
Recv: echo:MMU2:<R1a A0*41.`

mmoraes80, 2023-07-31:
> Just to be sure I:
>
> * Downgraded both MK2.5S and MMU2S firmware to previous versions
> * Factory-resetted the MMU
> * Verified that the MMU works (well at least to the quality level that it has worked [...]

After several attempts, I was finally able to make it work:
1. If your MMU2S was working with the FW 1.0.6, you DO NOT need to check for any assembly issues (at least, if you assembled it right at the first time)
2. Using Prusa Slicer, flash both printer and MMU2S with respective firmware. Once you complete this step, most likely your MMU2S will start making some very loud and weird noises. Leave it be, it will stop after 3-4 attempts of homing the idler.
3. If you have PrusaLink, disable the RPi port in the printer Settings menu
4. Download and run Arduino IDE. I was using **Pronterface it doesn't save the data to the EEPROM**, so it was like nothing works.
5. Connect the Arduino IDE to the printer (COM3 / Mega or Mega2560 board).
6. Open the Serial Monitor and send the gcode command: M708 Ax019 X5
7. After this, you see the MMU2S home normally and very quietly.
8. Restart the printer
9. Remember to re-enable the RPi port in the settings menu.

Rob70ar, 2023-07-31:
> Just as an info, you can use Octoprint if you add M708 to the "Terminal Auto Uppercase Blacklist" in Settings->Features.

MK3S+ with MMU2S and Octoprint (Raspberry Pi Zero W)

These steps helped me solve the homing error:

1) Enable MMU on the MK3 menu
2) Octoprint: add M708 command to the "Terminal Auto Uppercase Blacklist" in Settings->Features
3) Octoprint terminal: send command M708 Ax019 X5

NOTE: If you fail to perform step 2) then command sent via Octoprint won't succed and nothing will change.

Julasch17, 2023-07-31:
> > Nur als Info, Sie können Octoprint verwenden, wenn Sie M708 zur "Terminal Auto Uppercase Blacklist" in Einstellungen->Funktionen hinzufügen.
> 
> MK3S+ mit MMU2S und Octoprint (Raspberry Pi Zero W)
> 
> Diese Schritte haben mir geholfen, den Homing-Fehler zu beheben:
> 
> 1. Aktivieren Sie MMU im MK3-Menü
> 2. Octoprint: M708-Befehl zur "Terminal Auto Uppercase Blacklist" in Einstellungen->Funktionen hinzugefügt
> 3. Octoprint-Terminal: Sendebefehl M708 Ax019 X5
> 
> HINWEIS: Wenn Sie Schritt 2) nicht ausführen, ist der über Octoprint gesendete Befehl nicht erfolgreich und es ändert sich nichts.

do i have to enter this here ?

[image]

danvandermeir, 2023-08-02:
Hi all. Thank you for the excellent work done on the project over the years. Especially @Rob70ar for his described fix which worked for me:
> MK3S+ with MMU2S and Octoprint (Raspberry Pi Zero W)
>
> These steps helped me solve the homing error:
>
>     1. Enable MMU on the MK3 menu
>
>     2. Octoprint: add M708 command to the "Terminal Auto [...]

@Prusa-Support I understand sometimes these things happen, however this has been an incredibly frustrating issue for me. I spent well over 12 hours with Prusa Support personnel through LiveChat on prusa3d.com who should have easily been able to find this thread no less diagnose what the real issues are and deal with my support request more properly. Especially the one individual who claimed to have been part of the QA process for this firmware release for the last year..... Very disappointing that this setting set got missed for a full on release as it caused me well over a day's worth of hours searching for a fix until I stumbled here myself (I'm sure many other people as well). Through the 5 people I dealt with through prusa3d.com literally all of them failed. Typically, I am used to a level of support from Prusa that is beyond excellent, so I'm curious what's going on over there that the experience has flipped to the polar opposite? I will follow up via email as well, because the **constant** pressure from Prusa support personnel to insist that somehow this is a user error seems very hostile. For instance, when the new firmware directs people to check for mechanical issues as a first step... don't you think perhaps people have checked for this issue already? You know, given that is something that was _already suggested to the users as a first step to take_, maybe it would instead make sense that support personnel **ask what the current state of diagnosis and steps taken have been as to reduce further wasted time** in a global sense?

mmoraes80, 2023-08-02:
> Hi all. Thank you for the excellent work done on the project over the years. Especially @Rob70ar for his described fix which worked for me:
>
>
> @Prusa-Support I understand sometimes these things [...]

Since MMU3 and MMU2S have different values, all they need to do is to release a new hex file specifically for the MMU2S with the value changed to 5 and be done with it.

gudnimg, 2023-08-02:
It's stored in EEPROM. When the firmware is booting up it checks if the EEPROM value is 0xFF which means the EEPROM is in the default state and has never been written to, then the firmware automatically writes the default value.

We may add some sort of a LCD menu to tune this value but nothing set in stone yet... I will try implementing it and see how much flash memory it would need

mmoraes80, 2023-08-02:
> > Since MMU3 and MMU2S have different values, all they need to do is to release a new hex file specifically for the MMU2S with the value changed to 5 and be done with it.
> 
> @mmoraes80 , are you sure the value is from the hex file? or is it in the EEPROM ? The source code for 3.0.0 is not released yet, as pointed out here #4284

Once you use the M708 command, it replaces the value in the FW.. SO might as well just generate a FW version with the correct  value as the MMU2S and MMU3 use the same firmware, but not the same parameters

Elcar903, 2023-08-03:
I am having the same issue with the idler not homing message. 
I have tires to reset my MMU2s after the update and that did not fix it.

I have tried changing M708 A0x19 X6 to M708 A0x19 X5 and M708 A0x19 X7 values 
and that did not help either.

I tried resetting my MMU2s to 1.06 and factory resetting it with 1.06 firmware and then updating it top the MMU 3.0.0 firmware
but this didn't not fix it either

I noticed that when it tries to home the Idler motor seems to turn 3 times faster than it did with the 1.06 firmware.

I have no issues with the 3.12.2 ( 1.06 ) firmware so I am not sure what the issue is with the new one.

Rob70ar, 2023-08-03:
> I am having the same issue with the idler not homing message. I have tires to reset my MMU2s after the update and that did not fix it.
>
> I have tried changing M708 A0x19 X6 to M708 A0x19 X5 and M708 [...]

Yes, new firmware makes movements much faster and less noisy, a pleasure when it works.

Are you really sure the command is reaching the printer in the wright way? If not, the printer might respond to the command moving the idler and repeating the idler homing error message.

At least that was my own experience sending the command through Octoprint: I sent the command and the idler moved but problem persisted…. Until I configured Octoprint not to convert to uppercase this command. What I mean is that printer reacted very similar when it received the bad command, so the only noticeable difference was that with the wright command the issue was solved…

Elcar903, 2023-08-03:
I sent "M707 A0x19" to the printer after I changes the settings to make sure it was changing them correctly 
and it showed to be updating correctly for each setting.

mmoraes80, 2023-08-03:
> I sent "M707 A0x19" to the printer after I changes the settings to make sure it was changing them correctly and it showed to be updating correctly for each setting.

Some people are having success with X4 instead of X5...

kinoX51, 2023-08-04:
Same setup, same issue here - After the firmware upgrade started with the (new) factory reset prcedure (https://help.prusa3d.com/article/factory-reset-mmu_510075) which does not solve the issue. I then connected the printer via usb to my laptop and changed to value to "M708 A0x19 X5" and restarted the printer. The MMU thenhomed successfully.

I plan to buy the upgrade kit to mmu3 when it is available. Since it also get new printed parts I assume that I need to change the value back to it's default, right?

Elcar903, 2023-08-04:
I tried X4 as well and still no luck.

I also tried re flashing the MMU and printer to latest version then factory resetting both devices, then I tried X4, and X5 and still no luck with it.

I went back to 3.12.2 ( 1.06 ) and everything works  fine

mmoraes80, 2023-08-04:
> I tried X4 as well and still no luck.
> 
> I also tried re flashing the MMU and printer to latest version then factory resetting both devices, then I tried X4, and X5 and still no luck with it.
> 
> I went back to 3.12.2 ( 1.06 ) and everything works fine

Are you sure the printer is actually saving the information in the EEPROM? When I tried for the first time using Pronterface, the command was being sent but the printer was not acknowledging it so nothing seemed to work. After I switched to Arduino IDE, it took 3 seconds to fix it at first attempt, using X5.

Rob70ar, 2023-08-04:
Sorry to hear that. Once it works, the difference between 3.0 and the
previous one is huge!

Elcar903, 2023-08-05:
> Are you sure the printer is actually saving the information in the EEPROM? When I tried for the first time using Pronterface, the command was being sent but the printer was not acknowledging it so [...]

yes I checked,  its receiving and saving the settings.

I dont have an Arduino so I am going to just wait for a patch for this bug.

evaldasguogis, 2023-08-05:
X4 worked for me

[118 more replies]