prusa3d/Prusa-Firmware #4288: [BUG]Thermal calibration still not fixed

Plyn1x, 2023-07-28:
**Printer type** - MK3S+ with revo 
**Printer firmware version** -  3.13.0

**Describe the bug**
Thermal Calibration Crashes
  
**Expected behavior**
  Normal thermal calibratiom
   
I collected the log in a txt file

[fail thermal calibration.txt](https://github.com/prusa3d/Prusa-Firmware/files/12195095/fail.thermal.calibration.txt)

trfunk, 2023-07-28:
You might have to do step 5-16 in #4105 (or alternatively factory reset) to get it working. From looking at the log the thermal model parameters are still set to the v6 version:

`echo:  M310 P38.00 U0.0000 V1.00 C12.10 D0.0650 L2160 S1 B1 E1.74 W1.20 T-7.00`

I assume you could also try to just set the revo values (step 13):
`M310 P40 U-0.0014 V1.05 D0.15 L270`
and then run the thermal calibration wizard again. The 3.13.0 firmware seems to be using the same values as the comment in #4105: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/thermal_model/e3d_REVO.h

ggppjj, 2023-07-28:
Just to be certain @Plyn1x, the 3.13 FW files have a [specific build for Revo](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.0/e3d_REVO_fw_MK3_MK3S_MK3S+_3_13_0_6873.zip), is that what you used?

ggppjj, 2023-07-28:
The last few lines of the diagnostics info do say "all with revo fw", nevermind!

lawitzer84, 2023-07-29:
I had the same issue and really needed to do the **factory reset** to get the thermal calibration working.

Plyn1x, 2023-07-29:
Now only one printer fails
`TM: initial C est.
E0:0 RPM PRN1:0 RPM E0@:0 PRN1@:0
E0:0 RPM PRN1:0 RPM E0@:0 PRN1@:0
E0:0 RPM PRN1:0 RPM E0@:255 PRN1@:0
E0:0 RPM PRN1:0 RPM E0@:255 PRN1@:0
TM: error |1.289891|>1.200000
LCD status changed
TM: error |1.394564|>1.200000
TM: error |1.509255|>1.200000
TM: error |1.622618|>1.200000
E0:0 RPM PRN1:0 RPM E0@:255 PRN1@:0
TM: error |1.708699|>1.200000
TM: calibr. failed!
echo:Temperature Model settings:
echo:  M310 I0 R20.50
echo:  M310 I1 R18.40
echo:  M310 I2 R16.70
echo:  M310 I3 R15.20
echo:  M310 I4 R14.10
echo:  M310 I5 R13.30
echo:  M310 I6 R12.70
echo:  M310 I7 R12.10
echo:  M310 I8 R11.70
echo:  M310 I9 R11.30
echo:  M310 I10 R11.00
echo:  M310 I11 R10.80
echo:  M310 I12 R10.60
echo:  M310 I13 R10.40
echo:  M310 I14 R10.20
echo:  M310 I15 R10.10
echo:  M310 P40.00 U-0.0014 V1.05 C12.10 D0.1500 L270 S1 B1 E1.74 W1.20 T-7.00
TM: error |1.775167|>1.200000`

Wilts01, 2023-07-29:
I had the same problem.  Despite having loaded the Revo 3.13.0 variant of the firmware, every time I tried to run the Thermal Cal Wizard it errored.  I found that as others had mentioned, M310 reported the V6 hotend values.  By typing 
`M310 P40 U-0.0014 V1.05 D0.15 L270`
I was able to rerun the Thermal Cal Wizard and this time it passed.  
I could have reset the factory defaults but I didn't fancy typing in the sheet parameters again.  Hope this helps

nebulorum, 2023-07-30:
I had the same behavior of failing TM calibration. Decided to do a full factory reset. Smooth first time wizards, and printing seem to be working well. I wanted to avoid the reset but it worked well for me.

ggppjj, 2023-07-30:
I was on a self-compiled 3.13.0 alpha that had the Revo fixes applied and had done a successful Thermal Calibration, and did get the same issue on updating to the released 3.13.0. A factory reset didn't help, and the TM values were correct as compared to the steps shown in #4105, but I was still immediately getting a shutdown when attempting a Thermal Calibration. I finally fixed it by doing the PID calibration with Thermal Modelling disabled. Is this possibly the issue? Would adding PID tuning before a thermal calibration be possible? Is there some way that the printer would report the correct TM calibration values without actually using those values on TM calibration unless they're first manually saved, or saved via a factory reset?

nebulorum, 2023-07-30:
> 

> A factory reset didn't help, and the TM values were correct as compared to the steps shown in #4105, but I was still immediately getting a shutdown when attempting a Thermal Calibration.

I tried PID and TM calibration and both failed. I did see some level or thermal calibration happened in the wizard. I did not try anything beyond that.

Plyn1x, 2023-07-30:
> I had the same problem.  Despite having loaded the Revo 3.13.0 variant of the firmware, every time I tried to run the Thermal Cal Wizard it errored.  I found that as others had mentioned, M310 [...]

Tried it but didn't work did. give it this but nothing helped I even did m500 but I will try again. Factory reset is not an option wanna keep my printer stats

AndyPC54, 2023-08-01:
I have also had problems with 3.13.0 Revo.
Had to do factory reset all data. PID tune. Manually initiated Thermal Model tune (as per #4105 ) and another PID tune to get a stable system.
When printing I have seen some significant temperature glitches as shown in the attached file. It looks like the TM is detecting false temperature anomaly's and is reducing power. No TM Anomaly warning was triggered.
I tried another TM calibration today from the front panel and it failed with a TM Failed message. File attached.
I will revert to a previous version so that I can check that there is no hardware fault causing the temperature glitches.

[PrusaTMrerun.txt](https://github.com/prusa3d/Prusa-Firmware/files/12232353/PrusaTMrerun.txt)
[Prusa3.13.0Revotempglitch.odt](https://github.com/prusa3d/Prusa-Firmware/files/12232390/Prusa3.13.0Revotempglitch.odt)

Andy

JWartenberg, 2023-08-03:
I had the same issue. Neither PID tuning nor Thermal Model tuning worked after upgrade to the new firmware, REVO version.
I did a factory reset and after that PID and Thermal Model could be tuned without any problems.
There are a lot of posts in several forums describing the same issue. From my point of view it is not acceptable that a factory reset is needed after a firmware upgrade.

D-an-W, 2023-08-03:
I updated to 3.13.0 Revo and carried out a factory reset, I currently can't get past the first few seconds of the TM calibration.

Would it be possible to disable is with M310 S0, run there calibration and then enable it again?

JWartenberg, 2023-08-03:
> I updated to 3.13.0 Revo and carried out a factory reset, I currently can't get past the first few seconds of the TM calibration.
> 
> Would it be possible to disable is with M310 S0, run there calibration and then enable it again?

Did you the PID Tuning first? Maybe it helps doing it at first.

nebulorum, 2023-08-03:
Yesterday I decided to run both TM and PID calibration. And they all ran fine. Have been printing a but and it's seems fine.

D-an-W, 2023-08-03:
> > I updated to 3.13.0 Revo and carried out a factory reset, I currently can't get past the first few seconds of the TM calibration.
> > Would it be possible to disable is with M310 S0, run there calibration and then enable it again?
> 
> Did you the PID Tuning first? Maybe it helps doing it at first.

I will try now, thanks.

It seems strange how the same printers give such varied results.

jokohoko, 2023-08-05:
I am stuck with this to (MK3S with revo installed and revo firmware, did a factory reset), and tried following adjustments:
M310 W2.00 E2.00
M310 S0

But i still get a print aborted upon trying to run thermal calibration. 
Also did a PID tuning that ran without an issue, but still thermal calibration fails:
echo:  M310 P40.00 U-0.0014 V1.05 C8.77 D0.1500 L270 S1 B0 E1.75 W1.75 T-7.00
ok
TM: error |1.818301|>1.750000
TM: error triggered!

This error shows seconds after starting the calibration and the value i have seems way higher that any others have above. So trying to raise the W and E values fails (they seem limited to 1.75)

At the moment i cannot use the printer since even disabling the thermal model still gives an error as soon as i heat up the hotend.
In the old firmwares (without thermal model) i never had any issues).

nebulorum, 2023-08-06:
I noticed that there are 4 version for the Firmware, one for MK3 and one for MK3S/S+, and then 60W or normal. I had to double check my printer, but I wonder if some people are not using the incorrect one.

Wilts01, 2023-08-06:
> I noticed that there are 4 version for the Firmware, one for MK3 and one for MK3S/S+, and then 60W or normal. I had to double check my printer, but I wonder if some people are not using the incorrect one. It felt a bit more complicated for me, and I wonder how this will be moving forward.

I certainly was using the correct one but I agreed that a ReadMe file explaining the differences wouldn’t go amiss

genechuah, 2023-08-07:
Having this issue too. On an Mk3.
Used the correct version of the firmware, E3DREVO_fw_MK3_3_13_0_6873.hex, in zipped folder named e3d_REVO_fw_MK3_MK3S_MK3S+_3_13_0_6873.
Can't do Thermal Model Calibration. 
Initial C Est. > Stopped  ... then gave a loud short beep with "Print aborted".

alienryes, 2023-08-10:
Exactly the same issue as @genechuah

CelsoSantos, 2023-08-12:
I'm going to be complaining about the same issue...

MK3S+, fw 3.13.0, did a factory reset after flashing but it fails to pass the thermal tests.. I have the standard Revo 6 (not High Flow) and have used the Revo firmware file from the official release :(

nagyben, 2023-08-20:
I am also experiencing this issue. I go to `Settings -> Calibration -> Thermal model calibration`, the printer starts the cal, then within 5 seconds it beeps and the LCD says "STOPPED"

Octoprint logs:

```
Recv: LCD status changed
Recv: TM: initial C est.
Recv: echo:busy: processing
Recv: T:47.7 /230.0 B:28.0 /0.0 T0:47.7 /230.0 @:127 B@:0 P:25.3 A:30.7
Recv: echo:busy: processing
Recv: TM: error |1.336440|>1.200000
Recv: T:56.6 /230.0 B:27.8 /0.0 T0:56.6 /230.0 @:127 B@:0 P:25.7 A:31.8
Recv: TM: error |1.522053|>1.200000
Recv: TM: error |1.718790|>1.200000
Recv: LCD status changed
Recv: TM: calibr. failed!
Recv: echo:Temperature Model settings:
Recv: echo:  M310 I0 R20.50
Recv: echo:  M310 I1 R18.40
Recv: echo:  M310 I2 R16.70
Recv: echo:  M310 I3 R15.20
Recv: echo:  M310 I4 R14.10
Recv: echo:  M310 I5 R13.30
Recv: echo:  M310 I6 R12.70
Recv: echo:  M310 I7 R12.10
Recv: echo:  M310 I8 R11.70
Recv: echo:  M310 I9 R11.30
Recv: echo:  M310 I10 R11.00
Recv: echo:  M310 I11 R10.80
Recv: echo:  M310 I12 R10.60
Recv: echo:  M310 I13 R10.40
Recv: echo:  M310 I14 R10.20
Recv: echo:  M310 I15 R10.10
Recv: echo:  M310 P38.00 U0.0000 V1.00 C12.10 D0.0650 L2160 S1 B0 E1.74 W1.20 T-7.00
Recv: TM: error |1.862032|>1.200000
Recv: TM: error triggered!
Recv: // action:cancel
Cancelling on request of the printer...
Recv: echo:Printer stopped due to errors. Supervision required.
```

PaulandChrissy, 2023-08-22:
I'm using the V3.11 firmware and it works fine apart from the irritating prompt to upgrade the firmware every time I print. Obviously can't upgrade because it won't work with the Revo. Logical solution would be for E3D to deal with this problem pronto. Can the firmware upgrade prompt be turned off in the interim.

3d-gussner, 2023-08-22:
@PaulandChrissy Please try Following steps 5-16 in https://github.com/prusa3d/Prusa-Firmware/issues/4105 worked for me:

> Can the firmware upgrade prompt be turned off in the interim.

Yes you can disable it in Settings -> HW Setup -> Checks -> Firmware
[Warn] is default
[None] will ignore it , but be careful with that as there might be a reason at some point that you have to update the firmware when using a new PrusaSlicer or gcode

PaulandChrissy, 2023-08-22:
Sending all these commands to the printer is just beyond me. I will turn off the prompt for FW upgrades. I understand what you mean about updates that are important, but if I am ignoring it anyway, it makes no difference. I will keep an eye open for any updates that may be important or solve the Thermal Model problem. While the printer is working beautifully I will just turn off the irritation in the interim.
Thanks for the help. Cheers PaulandChrissy

werz99, 2023-08-28:
Same problems on my MK3S. After some seconds of calibration "print aborted".
Went back to 3.11 because I want a reliable printer.
Hope this will be fixed and a "normal" update will be possible in the future. - Otherwise the REVO upgrade is not recommendable and was a waste of money!

WolfgangZenner, 2023-09-03:
I have 3.13 too and the temperature is as in the picture from octoprint - should I go back to 3.11 ?
[image]

3d-gussner, 2023-09-04:
@WolfgangZenner Please provide bit more information about your printer. Which firmware file have you installed? 

This temperature graph is "wild". Do you you have thermal model active?

WolfgangZenner, 2023-09-04:
Hello, I have 3.13.0 installed. Because I don't know about thermal model, it should be default. The behavior is since I installed the recommended firmware

3d-gussner, 2023-09-04:
Which printer type?
Which Revo? 
Which heater? 
Which thermistor? 

Looking at the shared picture I guess you have
A different thermistor 
A different heater 
A broken cable. 

Did you disable the thermal model? What did you do while taking the recording?

WolfgangZenner, 2023-09-04:
I bought the "printer kit" Prusa i3 MK3S+ in August 2022 and assembled ist. All parts are original.
The firmware is 3.13.0-6873.
How to disable/enable/verify thermal model
What is REVO?
I wonder why the cable should be broken, the moment I installed the new firmware.
Before, i had installed prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex.
Should I reinstall this firmware, to check whether the cable is broken or not? Which cable could be broken?
I don't speak much english - at the "Saarland" we learned french as foreign language.

3d-gussner, 2023-09-04:
@WolfgangZenner Hallo, normalerweise Antworten wir hier nur auf Englisch, aber mache hier mal eine Ausnahme.
Du hast in diesem E3D REVO Ticket reagiert, darum war ich davon ausgegangen das Du einen modifizierten Drucker hast.

Bitte wende Dich an den Prusa Chat mit diesem Problem.

Die Grafik zeigt nach der Aufwärmphase wenn der Extruder beginnt sich zu bewegen sehr große Abweichungen und das ist häufig verursacht durch ein Kabelbruch.

Frage: Wie reagiert Dein Drucker bei diesen Spitzen? Bekommst Du Thermal Error auf dem LCD angezeigt?

Hier eine Anleitung wie Du eventuelle Kabelbrüche untersuchen kannst:

1. Schaue nach ob der Thermistor im Heatblock fest sitzt.
2. Schaue ob das Kabel am Thermistor in Ordnung ist. Wir haben gesehen dass zum Beispiel diese brechen wenn man mit einer Drahtbürste zu "heftig" sauber macht.
3. Überprüfe dass die Kable und Verbinungen RICHTIG im EINSY sitzen
4. Starte Octoprint und heize den Drucker vor
5. Warte bist die Temperatur stabil ist.
6. Bewege in Octoprint Kontrolle die X achse mehrmals nach rechts und links, schaue ob die Kurve stabil ist oder die Spitzen bekommt.
8. Bringe den Extruder in X=125  Z=100 und bewege den Kablebaum an verschiedenen Stellen in alle "normalen" Richtungen. Schaue in OctoPrint an welchen Stellen oder Bewegungen dies passiert.

Ich hoffe das Hilft etwas weiter und wie oben erwähnt, wendest Du Dich am besten and den Chat Support.

WolfgangZenner, 2023-09-04:
Vielen Dank für die Antwort. Ich werde das alles heute Abend testen. Die Firmware spiele ich immer über PrusaSlicer auf, weil Prusa das so angibt.
Wenn ich nichts finde, versuche ich mal den Chat - hoffe die machen das auch auf deutsch.

3d-gussner, 2023-09-04:
@WolfgangZenner Gerne und Prusa Chat ist auch in Deutsch. Problem mit OctoPrint Firmware flasher ist das nicht alles coorect überschrieben wird. Der Xflash wo die Sprachen sind wird derzeit überhaupt nicht geschrieben von Firmware OctoPrint Flasher.

Viel Glück :four_leaf_clover:

WolfgangZenner, 2023-09-05:
Ich habe den Drucker vorgeheizt und den Kopf nach rechts und links bewegt - da hat sich keine Temperaturänderung ergeben.
Ich habe dann den Kabelstrang am Druckkopf/Extruder etwas hin und her bewegt, dabei hat sich die Temperatur auch nicht geändert, dann startete er nach jedem Druckstart neu, zuletzt kam jetzt "M112 cancelled - emergency stop" und ein lauter Piepton. Jetzt weiß ich gar nicht mehr weiter. Ich probiere mal den Chat.

gbzh01, 2023-09-11:
PID tuning didn't work after upgrade to the new firmware, REVO version.
I did a factory reset , maybe I didn't do the PID then because I still had failure. However, I use the printers for work and there are many other things that don't work (not the place for those issues here) so I am hoping for firmware improvements. This is not the time to fiddle on 6 printers to get in run.

ggppjj, 2023-09-11:
Note: Step 10 may be omitted if it doesn't work, it's a debug code that may not be enabled in the released firmware IIRC.

ggppjj, 2023-09-11:
Agreed, it is a bit of a hassle right now. I believe Prusa is working with E3D on a better way forward, but I'm just someone commenting on a public issue so I don't actually know what Prusa has going on behind the curtain. Based on my understanding of how this works though, you should not need to do this again. You may need to run another calibration in the menu if you change to a different nozzle type with different thermal properties (maybe CHT or ObXidian), but that's expected.

D-an-W, 2023-09-11:
> Agreed, it is a bit of a hassle right now. I believe Prusa is working with E3D on a better way forward, but I'm just someone commenting on a public issue so I don't actually know what Prusa has [...]

Do you know how to correctly use the W and E values?

I assume the higher the W value makes it less sensitive but unsure how the E value works with it?

ggppjj, 2023-09-11:
I don't actually know what you're referring to here.

Here's the G-code page on the RepRap wiki with more info on what M310 is doing:

https://reprap.org/wiki/G-code#M310:_Temperature_model_settings

D-an-W, 2023-09-11:
> I don't actually know what you're referring to here.
> 
> Here's the G-code page on the RepRap wiki with more info on what M310 is doing:
> 
> https://reprap.org/wiki/G-code#M310:_Temperature_model_settings

Thanks, I will have a read. 

It was here I was referring to…

https://github.com/prusa3d/Prusa-Firmware/issues/4105#issuecomment-1606964249

ggppjj, 2023-09-11:
Ah, gotcha. Yes, the page linked explains.

gbzh01, 2023-09-11:
OK got it. I may give it a try. Thanks so much to youFreundliche Grüsse/Best regardsGregor Bührer

D-an-W, 2023-09-11:
> Ah, gotcha. Yes, the page linked explains.

```
Ennn error threshold (K/s; default in variant)
Wnnn' warning threshold (K/s; default in variant)
```

I have been trying to figure out what unit K/s and if I should try E higher or lower, would you have any idea please?

trfunk, 2023-09-11:
K/s should be Kelvin per second.

quitmeyer, 2023-09-19:
> 12\. M500

This helped fix my problems it seems, thanks a lot

paalmiso, 2023-09-22:
worked also for me, before I did factory reset and calibration via menu, but printer was beeping every minute...

thank you

Using revo6 and 3.13.0

> @gbzh01 The steps above should get you past this issue. You'll need a serial console, so Pronterface or Octoprint's terminal or even straight PuTTY if you're up to it.
> 
> For convenience: Steps 5-16 from #4105:
> 
> 1. Send `M503` to show current settings.
> 2. Send `M310 S0` to temporary disable TM for PID cal
> 3. Got to LCD -> Calibration -> PID cal.
> 4. Send `M310 S1` to enable TM.
> 5. Wait until the printer is cold 🥶 before you continue.
> 6. Send `G28 W` to home all without mesh bed level.
> 7. Send `G1 X125 Y105 Z1` to move the hotend to the middle of the bed.
> 8. Send `M155 S1 C3` to auto report temps and fans.
> 9. Send `M310 P40 U-0.0014 V1.05 D0.15 L270` to set REVO values.
> 10. Send `D70 S1` to enable tml debug code.
> 11. Send `M310 A F0` to run Thermal model cal. without checking existing TM settings.
> 12. Send `M500` to store TM values.

nebulorum, 2023-10-09:
Anyone has tried the new Firmware 3.13.1?

malhovic, 2023-11-16:
> Anyone has tried the new Firmware 3.13.1?

I have tried 3.13.0, .1 and .2, all 3 of them give me issues with thermal anomalies and eventually thermal errors. I've done the steps above, i've factory reset, manually PID tune, etc. and nothing but thermal errors (and lots of beeping). 

I went back to 3.11.0, factory reset, performed calibration/PID tuning, and am back to printing again. Hopefully it continues to run well. I have stuff to get printed for christmas!

nebulorum, 2023-11-19:
> I have tried 3.13.0, .1 and .2, all 3 of them give me issues with thermal anomalies and eventually thermal errors. I've done the steps above, i've factory reset, manually PID tune, etc. and nothing but thermal errors (and lots of beeping).

I decided to give 3.13.2 a try and it is working perfectly. Not a lot of prints, but no issue at all. Did you use the correct version?

In my case it was `E3DREVO_fw_MK3S_MK3S+_3_13_2_7080.hex`. The default download will not work. You need to download from Github and use the correct version (including heater power).

JWartenberg, 2023-11-20:
I also use 3.13.2, downloaded from GitHub, same file as mentioned by nebulorum.

After the firmware update I initially had no problems, then suddenly printing was no longer possible due to constant thermal anomalies.

It's been working rather good since yesterday.

What is the difference?

I print PETG in an enclosure. There are no/less thermal errors when the door is closed (temperature in the enclosure > 33°) or half open (temperature in the enclosure > 27°). When the door is open and the temperature in the enclosure is low (approx. 20°), errors occur, e.g. when the first print is started after a long pause. Room temperature is between 18° and 20°.

A workaround may be to increase the ambient temperature.

But: no chance of printing anything with ASA. Thermal anomaly already at the Purge Line. Preheating the closed enclosure to almost 40° didn't help.

That's why I decided to turn off the feature.

malhovic, 2023-11-20:
I tried 3.13.2 with no enclosure and I got thermal anomalies which turned
to an error/stop print. I’m curious what’s going on as even on 3.12 my
temperature reporting is way skewed. It seems fine to start but I can watch
the temp reporting go from 215 to 231 down to 211 in three reporting
cycles, so it’s like the printer is reading the temp wrong or the temp
sensor in the revo head is bad. I did use the proper firmware from GitHub
as well.

I also don’t believe the temp sensor is bad because the bed temps are
wildly reporting as well. Not as huge a variance in temp but still the bed
doesn’t heat up and cool down by ~10C in less than 10 seconds.

mhouse1, 2023-12-02:
> That's why I decided to turn off the feature.

also getting thermal anomally , how did you turn off this feature? I'd like to keep using the other features of 3.13.2

mhouse1, 2023-12-02:
looks like you can disable using [gcode here ](https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1380243362) but its not very practical since our print farm has double digit number of machines, with lots of files on sd card we'd have to reslice. 

Also the upcoming reprint feature on fw 3.14 will be a huge plus for print farms with hundreds of files, so sticking with 3.11 is not really ideal.

snafu1282, 2023-12-11:
MK3S+ with Revo 40W 0.4 obXidian nozzle.

I have managed to flash `E3DREVO_fw_MK3S_MK3S+_3_13_2_7080.hex` and fine tune `M310` so the printer no longer generates thermal anomaly errors. 

I did it by issuing several `M310 Exx Wxx` commands, increasing the values of `E` and `W` by 5 after each thermal anomaly. The default was `E4.0 W3.5`. The errors finally stopped occurring after I sent `M310 E7.5 W7.0`.

This seems to me to be a long way from the default settings. Maybe it's OK, but I can't help but wonder if it is.

D-an-W, 2023-12-11:
I found the settings below stopped all errors…

`M310 W2.00 E2.00`

3d-gussner, 2023-12-11:
> The default was `E4.0 W3.5`. The errors finally stopped occurring after I sent `M310 E7.5 W7.0`.

This are really high values you set `E7.5` and `W7.0` as the defaults are [E1.23 W0.8](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/thermal_model/e3d_REVO.h#L15-L16)

3d-gussner, 2023-12-11:
> I found the settings below stopped all errors…
> 
> `M310 W2.00 E2.00`

Please  never use same values for `W` and `E` the `E` should be always larger.

D-an-W, 2023-12-11:
> > I found the settings below stopped all errors…
> > `M310 W2.00 E2.00`
> 
> Please never use same values for `W` and `E` the `E` should be always larger.

Thanks, to be honest I never quite understood them! Would E2.01 be ok?

[14 more replies]

closed 2024-03-13 by 3d-gussner