prusa3d/Prusa-Firmware #4289: MMU2S with MK3S+ Light Not Working Switching Filament Slots

ljford7, 2023-07-28:
**Printer type:** MK3S+
**Printer firmware version:** 3.13.0
**MMU firmware version:** 3.0.0

**Describe the bug:** After upgrading the MMU2S to firmware 3.0.0, the lights don't light up when switching filament slots with the buttons on the MMU. 

**To Reproduce:** Manually switch slots on the MMU with the buttons on the MMU
 
**Expected behavior:** The lights should light up highlighting which slot is currently selected on the MMU

https://github.com/prusa3d/Prusa-Firmware/assets/2449331/70161835-0406-4f0c-bfc5-233a58846967

StopJustStop, 2023-07-29:
I am also having this issue after upgrading the firmware.  I also tried a factory reset.

DRracer, 2023-08-01:
Please see: https://help.prusa3d.com/article/status-led-explained-mmu3_510059

And no, LEDs do not reflect the position of the selector when moved.

ljford7, 2023-08-01:
> Please see: https://help.prusa3d.com/article/status-led-explained-mmu3_510059
> 
> And no, LEDs do not reflect the position of the selector when moved.

This doesn't make sense. Why would a very useful function be removed from the new firmware?

gudnimg, 2023-08-01:
Removed bug label for now because this behavior is by design in MMU FW 3.0.0.

StopJustStop, 2023-08-01:
> Please see: https://help.prusa3d.com/article/status-led-explained-mmu3_510059
> 
> And no, LEDs do not reflect the position of the selector when moved.

@DRracer @gudnimg    The MMU3 has a printed label on the front that shows the position of the selector.  If I had a MMU3, I would be fine with using the label to determine which filament was selected.  Please keep in mind that the MMU2S doesn't have the numbered slots printed on the front like the MMU3.  We can't look at a label and tell what position the selector is in.  For MMU2S owners, the ONLY indicator of selector position is the indicator LED.

The LEDs USED TO reflect the position of the selector.  That functionality has been taken away which just made it very painful for MMU2S users.

Prusa-Support, 2023-08-02:
Thank you for your feedback.

I'd like to ensure you that the meaning of LED lights is very different now.
This is intentional and the relevant [Help article](https://help.prusa3d.com/article/status-led-explained-mmu3_510059) was updated accordingly.
Feel free to go through it if you are curious to learn more about the meaning of the LED lights and please notice the lights no longer simply track the position of the "virtual MMU-extruder" or un/loading operations but provide further analytic information.

The user would not necessarily need to study the exact meaning of all light combos (information close to a machine language) but follow the indications on the printer's LCD displayed as messages (at a human language degree).

This is one of the biggest advantages of the new firmware.

The position of the selector is quite explicit and most importantly pretty much all operations can be performed via the LCD menu, now with an improved user interaction that doesn't simply rely on a combination of lights.

Michele Moramarco
Prusa Research

clintonefisher, 2023-08-06:
That's neat that the new firmware uses the lights at bootup to show error status.  On my MMU2S they used to also show the selector location.  Some may not have liked the LEDs but I liked them and they were useful.  Now, this USEFUL functionality has been taken away.   Sure, I could abandon the wonderful new features of the new firmware, if I wanted the lights, but the point is WHY would Prusa remove a useful feature from long-standing customers?  

Feature removal is generally annoying to customers and you never know what action is the final straw that causes a long-time Prusa fan to jump ship for a competitor.

Why not make the selector LEDs an OPTION?   Put the option in a register accessible by GCODE.  Then EVERYONE is happy!  Old time MMU2S owners get the LED's.   MMU3 users who don't need them, or folks who don't like the LEDs, can turn them off.

Options are great ways to avoid annoying long-time customers and to keep them satisfied.

halofx, 2023-08-24:
> I'd like to ensure you that the meaning of LED lights is very different now. This is intentional and the relevant [Help article](https://help.prusa3d.com/article/status-led-explained-mmu3_510059) [...]

This doesn't justify a poor decision.
I agree with the annoyed folks, removing a useful function was a bad call. It is fine that the LEDs serve other functions while the MMU is going through other states, but why can't they still serve a function in the IDLE state? This seems very short sighted in my opinion.

halofx, 2023-08-30:
> It works for me _when printing_. MMU 3.0.0 fw. Here is a video: https://youtu.be/z9jfaW7bJyQ

Yip, that is by design as well.

I gave up, and rolled my firmware back today. The MMU had become completely unreliable in loading filament. Rather disappointed.

Prusa-Support, 2023-09-05:
Basically, the lights now have quite specific meanings during the bootup (for analytical purposes) and only match the selector position when loaded with filament to print.

The selector's position is generally not relevant as long as there is no filament loaded so, as of now, the light won't match the position of an empty selector.
I totally understand that some users are already used to the old fashion position lights however the firmware was totally rewritten and improved on all aspects, and this is not a bug but intentional.

Michele Moramarco
Prusa Research

closed 2023-11-23 by github-actions[bot]