prusa3d/Prusa-Firmware #4290: [BUG]MK3S+ Firm 3.10 filament Sensor not working

FrostyDogs2, 2023-07-29:
Hi

Have a Prusa MK3S+ with REVO 24volt Prusa MK3 Edition works great with the 3.10 Firm installed. It works great but would like the filament sensor to work on 3.10 Iam not a firmware expert editor.  If I run 3.12 from Prusa the sensor works fine so we know it can work.  Is there a workaround for now on 3.10 and when sensor runs out of filament it stops.  Tried 3.13 Firmware and it not working yet iam not a firmware expert editor.

Thanks

FrostyDogs2, 2023-07-29:
It wont pass PID on anything other than 3.10

sarusani, 2023-07-29:
Try installing 3.13 again. But after install do a factory reset.

FrostyDogs2, 2023-07-29:
How is factory reset done ?

FrostyDogs2, 2023-07-29:
Installed 3.13 and flashed it then when it was done pressed the X button under wheel to restart. Tried PID again and it just starts to go up on 210 deg but then goes to zero and then after waiting a long time it says Finished but it not work.

FrostyDogs2, 2023-07-29:
Ok Thanks found the factory reset and reset ALL data it doing the PID settings this time.

sarusani, 2023-07-29:
Great! Glad it worked out 👍

FrostyDogs2, 2023-07-29:
Well made it throu the PID and setup but it still wont stop printing when it runs out of FILIAMENT.  Have 3.13 and done factory reset ALL DATA and did the full wizard setup and the sensor settings are turned on.  It sure would be nice to have this back. Either something happend or Firm dont know. 

Sarusani was thinking we had it but will require more looking.  Thank you that did help with the factory rest it got the PID done.

sarusani, 2023-07-29:
@FrostyDogs2 
I just tested this on my MK3S+, my print stops when I run out of filament. So it's not missing in the firmware.

What does "Fil. sensor v." show when you go to the LCD Support menu? (Mine is "0.4 or newer")
Does the state of the sensor in "Support->Sensor Info" change correctly when you insert/remove filament? ("1" when inserted /  "0" when removed)

FrostyDogs2, 2023-07-29:
It says 0 and have filament in machine going to take hot end apart and check all wires

FrostyDogs2, 2023-07-29:
Think we might have it the sensor wire is lose in control box. Installed a new spare sensor. Now it states 1 when loaded and 0 when removed. 

That wire must of got bummped when doing the wire hookups.

Well going to put everything back together now and test.

FrostyDogs2, 2023-07-29:
@sarusani  You get GOLD star thanks think we got it now passed.  

1 Revo installed, 
2 3.13 Firm flashed, 
3 Factory ALL DATA reset, 
4 New filament sensor installed and wires hooked up.
5 Did 4 test prints and ran out of filament and printer stopped 

At last now maybe can get back to some printing

Think this case is fixed

sarusani, 2023-07-29:
@FrostyDogs2 
You’re very welcome.
Enjoy the rest of the weekend! 😄

closed 2023-07-31 by leptun