prusa3d/Prusa-Firmware #4291: [BUG]MK3S+ and MMU2S updated to 3.13.0 and 3.0.0. MMU2 not working

victorbjo, 2023-07-29:
Using a MK3S+ and a MMU2S, I have updated to the newest firmware (Successfully according to prusaslicer). Ever since the MMU2 will just try and home? the idler. After doing this none of the LEDS light up and the MK3 does not detect the MMU.

The MMU, while not working fine before the update, did work and could connect to the printer. I was using it all day yesterday, so I am certain this problem arose after the update.

https://github.com/prusa3d/Prusa-Firmware/assets/47253754/7a6a8f62-69b0-4b80-b638-a53e0a9482a6

mlpriv, 2023-07-29:
also here, exactly same error...

Rob70ar, 2023-07-29:
same here, stuck at idler connot home #04125... idler can move freely, screws checked...

victorbjo, 2023-07-29:
I was in contact with the support team for around 2 hours. We did not fix the issue, but the support worker told me he had created a ticket.
Additionally, I got everything working by downgrading the FW

Cajda123, 2023-07-29:
i have discovered, that MMU is disabled by default in menu.

victorbjo, 2023-07-29:
@Cajda123 can you enable it again?

Cajda123, 2023-07-29:
Yes, in the menu is MMU -> OFF/ON. In release notes is this described.

victorbjo, 2023-07-29:
@Cajda123 Jesus....
Thank you. I do not know how I missed this. But neither of the 2 PRUSA support workers I talked with knew it either.
I guess I should close the bug then?

GitPetri, 2023-07-31:
For me enabling the MMU doesn't help, I'm still stuck with #4125 (MK2.5S&MMU2S).  @victorbjo did you get rid of the errormessage in your setup by enabling MMU (meaning that I need to create a new ticket for the issue)?

victorbjo, 2023-07-31:
I did not have an error message on the LCD. It just didn't show the MMU. But yes, enabling MMU solved all my issues

gudnimg, 2023-08-01:
Closing because issue was resolved.

closed 2023-08-01 by gudnimg

IllicLanthresh, 2023-10-22:
Hi @gudnimg, the issue is NOT resolved in the latest firmware provided in the downloads section. I just tried to upgrade my mk3s and mmu2s firmware using the latest firmware supplied at https://help.prusa3d.com/downloads/mmu2/firmware and I still have the same issue as OP describes, that sound is terrific and I'm scared that any of the parts got damaged because of the torque being applied to the idler.

IllicLanthresh, 2023-10-22:
if you guys know that the firmware is faulty, you should remove it from the downloads page to prevent these issues...

gudnimg, 2023-10-22:
The firmware is not “faulty” and is running by design.

For the terrible homing sound, the firmware can’t tune the homing for you unfortunately. Have you tried some of things here: https://github.com/prusa3d/Prusa-Firmware/issues/4285 e.g. Send g-code to change the homing sensitivity?

Its very unlikely to damage anything. But if you see damage then it best to reach out to Prusa’s Technical support. 

In 3.13.2 we are adding a new LCD menu to tune the sensitivity so sending a g-code is not needed. You probably need to lower it from 6 to 5 or lower. 

In case the  MMU is “N/A” in the LCD “Support” menu, then you need to enable the MMU in the settings first. (This is a new setting which is set to “Disabled” by default”)

tyl3r413, 2023-10-29:
> The firmware is not “faulty” and is running by design.
>
> For the terrible homing sound, the firmware can’t tune the homing for you unfortunately. Have you tried some of things here: #4285 e.g. Send [...]

The firmware appears to be faulty as mentioned by every other person on this post. I have 6 MK3s printers with MMU2S. I updated all of them two days ago and all six of them became potatoes. I have enabled MMU2S in the menus and tried all suggested fixes. Rolling back to a STABLE firmware version is the next step. Prusa screwed this up big time. If Prusa calls this "working as intended" then I'm the freaking pope. This has been an absolute nightmare.

rumbledethumps, 2023-10-29:
Same issue here. Upgrading MMU firmware to 3.0.0 causes various errors. We tracked it down to noise in the cables. I was told by support to buy an upgrade to MMU3 for the new electronics. Obviously, this answer is unacceptable, the printer worked fine before the firmware upgrade.

Prusa-Support, 2023-11-07:
This issue was solved by enabling the MMU via the printer settings and closed as such.
Possibly please only provide information that is relevant to the issue.

Please consider reaching out to our [official support](https://help.prusa3d.com/article/customer-support_2287) for a proper troubleshooting session before posting further below in this issue issue which is archived as solved, or try to find other existing issues that better represent your specific case.

Michele Moramarco
Prusa Research

tyl3r413, 2023-11-07:
> This issue was solved by enabling the MMU via the printer settings and closed as such. Possibly please only provide information that is relevant to the issue.
>
> Please consider reaching out to our [...]

I don't think you guys understand. Enabling the MMU did NOT fix the problem. The firmware is trash and we had to roll back in order to keep our printers working. We have been through this multiple times....... And you keep posting this like it's an all around solution but it does NOT fix the problem. Please read the whole thread prior to replying with an irrelevant response.

tyl3r413, 2023-11-07:
This thread was prematurely closed by Prusa support prior to a solution being found. Talking with support on the phone has been just as useless. The only working solution we have found is to roll the printer firmware back to the previous working version. They refuse to admit there are bugs in the code and the firmware is corrupt.

Prusa-Support, 2023-11-23:
I'm sorry, I'm afraid you are looking at the wrong thread.
The problem is solved for the author, hence closed.

All in all, this issue is not about a bug and should be closed anyway.

This GitHub repository is theoretically only meant for firmware bug tracking purposes.
Please only post confirmed bugs but first check if some other user reported _the very same bug_ already.

GitHub is not meant for individual support so please contact our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

tyl3r413, 2023-11-23:
......THIS IS A FIRMWARE BUG...... Prusa support is full of idiots.

tyl3r413, 2023-11-23:
I wish an intelligent Prusa support professional was on this thread and not
the clown who is too arrogant to acknowledge this bug. Crazy how a dozen
people tell them what the problem is and they ignore everyone and tell us
it's not a firmware bug. Prusa needs to hire smart people. It's
embarrassing how bad their support has gotten over the past couple years.
We still haven't resolved this problem and phone support has been even
worse than their arrogant programmers. Bunch of clowns.