prusa3d/Prusa-Firmware #4292: [QUESTION]<MMU2S with firmware 3.0.0 service menu>

DesC21, 2023-07-29:
I just updated my MMU2s to the new 3.0 firmware. I am running the MMU outside of my enclosure, so the PTFE from the MMU2 to the printer is longer than the stock one. The old firmware had the service menu (https://help.prusa3d.com/guide/service-menu-individual-filament-calibration_86376), where I could set my custom length. But I can't enter the service menu on the newest version. Is there a change on how to set the length? 

thanks in advance

3d-gussner, 2023-08-02:
@DesC21 Does the MMU3 in your setup work? Can you share a picture and add more details how long your PTFE tube is.

DesC21, 2023-08-06:
> @DesC21 Does the MMU3 in your setup work? Can you share a picture and add more details how long your PTFE tube is.

Thanks for your answer. Yes, it works fine but filament changes take a lot longer than they could/should, which adds up quickly.
My PTFE tube is ~70cm long. 

1. loading:
On a load it "fast" feeds for about 30cm I guess (I haven't measured it). Then it slowly continues to feed filament until it reaches the extruder and the sensor on the extruder triggers. Works fine but the loading takes really long.

2. unloading: 
it unloads fast until about 15cm of filament is left in the ptfe tube. Then it tries to move the idler, doesn't work and continues to fast unload until the finda is not triggered anymore. 

This is how my setup looks, so yes works fine apart from wasting a lot of time on load :).
[image]

From what I see in the firmware it appears these load/unload values are hardcoded now and can't be set on the MMU/printer anymore, right?

ainuke, 2023-08-07:
I have the same issue/question, FWIW.

3d-gussner, 2023-08-10:
FYI we are working on a solution to be able to change the bowden length. Please don't ask WHEN we gonna release that version!

3d-gussner, 2023-08-11:
Thanks for reporting.
See https://github.com/prusa3d/Prusa-Firmware/pull/4322

Please consider to close this request.

DesC21, 2023-08-11:
> Thanks for reporting. See #4322
> 
> Please consider to close this request.

awesome, thank you very much! Looking forward to test it :-)

closed 2023-08-11 by DesC21