prusa3d/Prusa-Firmware #4295: REVO thermal error while calibrating or heating up[BUG]<Enter comprehensive title>

ibonneau, 2023-07-30:
**Printer type** MK3S
**Printer firmware version** -  3.13.0

**MMU upgrade** - NO
**MMU upgrade firmware version** - N\A

**SD card or USB/Octoprint** SD Card
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
After installing the Revo as per E3D's instructions. It worked with no issue on 3.11.X.
Upgraded to 3.13.0 because it supports the Revo but I get Thermal anomaly when PID tuning or when attempting to Calibrate the Thermal model.
Will downgrade to 3.11.X for now.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

jokohoko, 2023-07-30:
I am having the same issue. The calibration or heating reaches 50 °C in about 2 seconds and throws the error!

thomasloven, 2023-07-30:
Duplicate of #4288?

EdwardGusman, 2023-07-31:
Same problem as all the above. Will downgrade to 3.11 for now.

PaulandChrissy, 2023-08-05:
Just fitted Revo to my MK3s with exactly the same thermal anomaly on V3.13. Flashed to V3.11 and problem disappeared. Hopefully future firmware flashes will have this problem solved, otherwise we are relegated to V3.11 for eternity. Thought Revo was just a drop in with no firmware concerns. Just plug and play. What a drag.

PaulandChrissy, 2023-08-05:
My issue was while using SD card. Reconnecting the original hot end solved issue As soon as Revo connected, problem returned. V3.11 working OK.

gulligin, 2023-08-14:
This isn't the awesome Prusa I know.  Prusa is known to be the best of the best, and this ain't cuttin it.

PaulandChrissy, 2023-08-14:
Surely E3D should be imputing to resolve this v3.13 firmware issue. Come on E3D, you sell this well thought out upgrade only to fall at the first hurdle. It works fine on 3.11 but every time we make a print, the printer prompts us to upgrade the firmware which is not possible while we have REVO installed.

PaulandChrissy, 2023-11-15:
Still only works with firmware version 3.11. This is irritating as the printer keeps reminding me every print, to upgrade to V3.13 but this will not work with the Revo. 🙁

3d-gussner, 2023-11-15:
@PaulandChrissy Have you read the pinned issue https://github.com/prusa3d/Prusa-Firmware/issues/4451 ?

closed 2024-01-22 by github-actions[bot]