prusa3d/Prusa-Firmware #4297: [FEATURE REQUEST] MMU fw 3.0: Tube length in register

mdjurfeldt, 2023-07-31:
The 3.0 MMU firmware doesn't work well if your bowden tube between MMU and extruder is of non-standard length.

Juan Puchol has shown how to patch the source code for a different length: [https://forum.prusa3d.com/forum/postid/666326/](https://forum.prusa3d.com/forum/postid/666326/) 

It would be better to have an MMU register for the length, gettable and settable with M707/M708.

DRracer, 2023-08-01:
@mdjurfeldt we are currently thinking about this option for modded printers.
The reason why we didn't include such register in the default firmware is because increasing the PTFE tube renders the MMU buffer useless and will cause even more issues - the filament loop won't fit inside and you'll have filaments randomly ejecting or falling out of the MMU.

mdjurfeldt, 2023-08-01:
@DRracer Thanks for your information. I can see the point about the buffer shipped with MMU3. But then, there are all kinds of buffer designs out there...

Can you explain further what you refer to with "more issues - the filament loop won't fit inside and you'll have filaments randomly ejecting or falling out of the MMU"? Is this still with regard to the buffer? If not, I don't see how increasing the length can lead to other kinds of problems. It works well with the old firmware.

DRracer, 2023-08-01:
@mdjurfeldt Yes, I was referring to the buffer - the original MMU2S one and even the new greatly improved MMU3 one. 

Of course, there are other 3rd party solutions which may not suffer from the increased length of the PTFE tube.

mdjurfeldt, 2023-08-01:
Actually, looking at the MMU3 buffer (which seems nice! :), it's probably not that hard to mod it to accomodate a slightly longer loop.

erikosvaldsson, 2023-08-08:
FWIW I have the same issue since upgrading to the MMU 3.0 firmware!

Coolgrey5, 2023-08-09:
Hi, I know nothing about programming so I don't know how easy this would be.. but i did notice that in the new 3.0 firmware there is now a loading test option, could this be renamed to a loading calibration? (maybe even moved to the calibration menu?) so it would run the test as it does now, loading each filament to the extruder in turn, only this time it would store the value for the tubes length itself, if it was different to currently stored/default value?
While the old 1.06 method made a huge difference to the reliability of my MMU2s it was a pain to calibrate each tube by carefully moving the filament to they same spot on the gear each time.......

3d-gussner, 2023-08-11:
Thanks for reporting.
See https://github.com/prusa3d/Prusa-Firmware/pull/4322

Please consider to close this request.

mdjurfeldt, 2023-08-11:
The feature request has been addressed in #4322.

closed 2023-08-11 by mdjurfeldt

awigen, 2023-09-13:
Thank you for implementing this. What is the register address?

awigen, 2023-09-15:
Thanks for publishing the MMU3 source code I could now find the address myself:
https://github.com/prusa3d/Prusa-Firmware-MMU/blob/e2f396e2a269e0d36433d0fc149e88f419b8be04/src/registers.cpp#L168