prusa3d/Prusa-Firmware #4301: M850 impossible to use with MMU2/3 enabled

wabbitguy, 2023-08-03:
**MK3S**
**3.13.0**

**MMU2S** 
**MMU3 3.0.0**

 Repetier Server Pro/Pronterface

**Describe the bug**
Using either the console in Repetier Server Pro OR connecting via USB to use the M850 command the terminal fills with so many "echo" messages from the connected MMU2S it's impossible to see any changes or even ask for a value because the list never stops filling and scrolling.

A work around is to disable the MMU2S in Settings menu. Then enable it when changes are done. But it gets chatty again.
  
**To Reproduce**
 Connect any server to the MK3S with an MMU2S enabled and the MMU2S echo messages never stop.
  
**Expected behavior**
Unless there is a purpose for the echoed messages, there shouldn't be any from the MMU2S. Or at least there should be a way to disable/enable the messages.
  
**echo:MMU2:<R1b Aff*bf.
echo:MMU2:>R1c*88.
echo:MMU2:<R1c A5*f6.
echo:MMU2:>R4*7b.
echo:MMU2:<R4 A0*66.
echo:MMU2:>R1a*f5.
echo:MMU2:<R1a A0*41.**

gudnimg, 2023-08-03:
We know it get's quite "spammy", we've discussed about improving this but no decision has been made about the implementation just yet. Printer senda a `Q0` (Query) command to the MMU every second and it's all logged currently.

3d-gussner, 2023-08-03:
@wabbitguy I don't know Repetier server but can't you create filters to "ignore" MMU messages? In OctoPrint I added a simple filter `echo:MMU2`.

wabbitguy, 2023-08-03:
@gudnimg thanks for the update, I wasn't sure it'd been noticed so that's why I flagged it. Spammy is a good word for too. As I said the work around is to disable the MMU2S in the menu to use M850 (and admittedly it's a high level option command that most users wouldn't be doing anyway). It just caught me by surprise. I will say the new printer and MMU2S  firmware is awesome. I'm astounded by how quiet and efficient it is.

@3d-gussner I tried Octoprint once, but Octoprint won't run a single delta let alone three deltas at the same time plus four cartesians like Repetier Server Pro will all via a USB hub from a single Rasp Pi. But thanks for the idea on that, I'll see if there's any way to do it and talk to the developer if not, he's very open and helpful.

JPTa, 2023-11-01:
In the variants files there is this:
#define MMU_DEBUG //print communication between MMU and printer on serial
But it is not used anywhere (anymore?).

gudnimg, 2023-11-01:
@JPTa you're right, it's probably dead code from the old MMU code from FW 3.12 and older.

I made a PR which filters the MMU logs a bit https://github.com/prusa3d/Prusa-Firmware/pull/4455 but it's unlikely to be merged at the moment.

3d-gussner, 2023-12-28:
As PrusaLink and also OctoPrint MMU plugin depend on the serial output we will not change it for the moment.
Closing this issue.

closed 2023-12-28 by 3d-gussner