prusa3d/Prusa-Firmware #4303: [ENHANCEMENT] Add "Thermal Modelling Debug" option to "Experimental Menu" in the printer.

ggppjj, 2023-08-03:
**Printer type** - All with Thermal Modelling
**MMU Upgrade** - N/A

**Is your enhancement related to a problem? Please describe.**
  Currently, Thermal Modelling is arguably difficult to disable for people that have non-stock hotends, and even more arguably not quite ready after a few revisions for the non-stock hotend that Prusa is targeting (Revo). It's not like it's crazy difficult or anything, but it does feel like the experience is lacking polish. 
  
**Describe the solution you'd like**
  There is an "[Experimental Menu](https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213)" in the printer that is "capable of adjusting the defining parameters of some components", which I believe should include being able to debug and/or disable Thermal Modelling. Ideally, the setting would exist to disable or quieten the thermal modelling beeping, add some additional debug tools for TM that might help support debug issues with TM (logging to SD or similar if that is possible, a standard routine that could be run and the results of that saved for devs to inspect or similar), and/or allow the user to disable TM from this advanced menu. Preferably, an option to disable TM until the next firmware version would be available here.

mix579, 2023-08-03:
+1 as I strongly believe this TM was at best released too soon, or worse, is fundamentally flawed.

Patrick-Schmitt, 2023-08-04:
+1 - it works yet with my Revo, but would appreciate more control over TM.

gbzh01, 2023-09-11:
I am having 6 printers with Revos and none of them work for the TM

ggppjj, 2023-09-11:
@gbzh01 please comment in issue #4288.

closed 2023-11-23 by github-actions[bot]