prusa3d/Prusa-Firmware #4304: Full Setup Wizard XYZ calibration error

jeepac, 2023-08-03:
**Printer type** - MK3S
**Printer firmware version** - 3.13.0

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A
**SD card or USB/Octoprint** -print via SD card

**Describe the bug**
 XYZ calibration error
  
**To Reproduce**
  when run full setup wizard, however at the end of XYZ calibration the whole Z assembly rams in to the bed
  
**Expected behavior**
 perform 9 point z calibration
  
**G-code**
  n/A
**Video**

https://forum.prusa3d.com/wp-content/uploads/2023/08/XYZ-calibration-failure.mp4

D-an-W, 2023-08-03:
I have also had this error.

At the point where the calibration with the paper finishes the Z axis should raise 150mm, if it doesn't and you continue then you get the result as in your video.

It was briefly discussed here [https://github.com/prusa3d/Prusa-Firmware/issues/4105](url) as two of us testing the firmware had the problem but it then started working ok for me.

sdh2, 2023-08-04:
I had that issue too. The only workaround I found is to shut off the machine during the end of XYZ calibration, then move the Z up and run the Z calibration separately. Even then it only sometimes works, other times it fails immediately.

sarusani, 2023-08-04:
When you move the z axis up and down manually (by long press of the knob). Does it move freely or does it also sometimes stop when it shouldn't?

If so, it's indication that the resistance for the motor is too high (binding) and the motor thinks it reached the top/bottom. That would explain why it's not moving up the full 15cm...

I looked at the code and when it shows the "place steel sheet" message it definitely issued the command to raise 15cm. The question is what are possible reasons for this command to fail.

@gudnimg 
As I understand raise_z should return the actual traveled distance. So maybe something like this could prevent this?
```
//make space
if (raise_z(150) == 150) {
  lcd_show_fullscreen_message_and_wait_P(_T(MSG_PLACE_STEEL_SHEET));
}
else {
  return false;
}
```

gudnimg, 2023-08-04:
@sarusani that's correct. But I suspect something set `planner_aborted` to `true` because it looks like the planned move is just ignored completely.

I added the following log into 3.13.0, can anyone check and see if it appears in the logs?
[image]

gudnimg, 2023-08-04:
Another note about `raise_z` is it should be a blocking call, so the firmware should stay in `raise_z` and wait until the 150mm move is done (which takes a few seconds on my printer). This is what `st_synchronize()` is for. If the printer returns immediately then something is wrong.

D-an-W, 2023-08-04:
> @sarusani that's correct. But I suspect something set `planner_aborted` to `true` because it looks like the planned move is just ignored completely.
> 
> I added the following log into 3.13.0, can anyone check and see if it appears in the logs? [image]
> 
> I would check but I don't get this problem on my MK3S+ for some reason

I can check if it happens tonight, do I need to run anything on terminal beforehand?

gudnimg, 2023-08-04:
> do I need to run anything on terminal beforehand?

No just make sure to use FW 3.13.0

gudnimg, 2023-08-04:
> If so, it's indication that the resistance for the motor is too high (binding) and the motor thinks it reached the top/bottom. That would explain why it's not moving up the full 15cm...

@sarusani its an interesting idea. I understand you mean motor stall? We could detect the event and log it 🤔This is what crash detection is for. I dont see the Z axis being handled though.

Patrick-Schmitt, 2023-08-04:
I had the same issue on one of my printers (MK3S+) at work. What seemed to help was to downgrade to 3.11 - do a whole factory reset - and then reflash 3.13 again.

Strangely, I had no issues with my private MK3S+.

gudnimg, 2023-08-04:
I found that `VERBOSE_CHECK_HIT_ENDSTOPS` exists in the firmware which can enable logs for when the firmware thinks it is hitting endstops. Maybe that can help debug this issue, the code doesn't compile but I can fix it probably optimise it a bit.

D-an-W, 2023-08-04:
> @sarusani that's correct. But I suspect something set `planner_aborted` to `true` because it looks like the planned move is just ignored completely.
>
> I added the following log into 3.13.0, can [...]

I did another factory reset and ran the wizard (After sending M310 W1.00 E1.74 so I could complete TM calibration) and it completed without error!

sarusani, 2023-08-04:
> I found that `VERBOSE_CHECK_HIT_ENDSTOPS` exists in the firmware which can enable logs for when the firmware thinks it is hitting endstops. Maybe that can help debug this issue, the code doesn't compile but I can fix it probably optimise it a bit.

I saw your PR. Unfortunately I can't reproduce the error on my MK3S. (I tried for the last 2 hours...) So I can't really help out.

bigsmalloverall, 2023-08-05:
Hi, 
It's my printer in the video. This error only happens when you run XYZ calibration with the wizard. 
Downgrading firmware from 3.13.0 to 3.12.3 fixes the problem.

naldarn, 2023-08-20:
I have experienced this error as well.  I did not factory reset after flashing the new 3.13.0 firmware and after following the direction to place the steel sheet the printhead was driven into the bed.  

I have since done a factory reset and still crashed into the bed.  I had to push the X to reset and it did not save the calibration to that point.  

I tried to re-flash the 3.13.0 firmware, and still had same issue and did not save calibration.  

I then did a factory reset, and this was the first time I heard the loud beep, but still had the same issue.

I purchased my printer November of 2020, if the electronics themselves have anything to do with the issue.

Is there any more information I can provide that could help troubleshoot this issue?

gudnimg, 2023-08-23:
I wonder if this is another issue only affecting MK3S printers or MK3S upgraded to MK3S+. 

Has anyone seen the issue on factory stock MK3S+?

D-an-W, 2023-08-23:
Mine was a kit.

gudnimg, 2023-08-23:
@D-an-W ok thanks for confirming. I’m sorry to say a fix for this won’t be in 3.13.1 :(

I’m hoping to look more into this issue.

D-an-W, 2023-08-23:
Thanks, now I know to watch for it I can deal with it accordingly. This has to be a coincidence but the last two times I updated the firmware it took 5 attempts at calibration before the Z axis raised up the expected 150mm at the point where you are asked to insert a steel sheet.

D-an-W, 2023-08-23:
Not at all no.

Patrick-Schmitt, 2023-08-23:
I just add some more info on my two MK3S+ printers:

Printer at which the issue appeared (work printer):
Is a black originaly assembled by PRUSA MK3S+. Never modified anything on it. No SW mods. No custom GCODE send via octoprint.

Bevor REVO upgrade printer was on latest 3.12.x firmware and working normally.

Did the REVO upgrade. Did a factory reset (to clear the wrong TM calibration data which is not cleared if just an update is done)
And than updated to the special REVO version of the 3.13.0 firmware.

--> printer crashes nozzle into bed during calibration because head is not set to correct height.

My own printer at home: 
Is a black sold as kit MK3S+ printer which I really enjoyed building ;)
I had the REVO kit since firmware 3.11.x installed. I previously updated to 3.12.x and DISABLED TM via two GCODE commands. (one to disable TM and another one to save)

Than firmware 3.13.0 was released. I updated the firmware without any factory reset. But the TM calibration failed instantly.
So I did a factory reset of the 3.13.0 firmware and tried the calibration again.

Without any issues. No crash.

This is all out of my memories. I hope that maybe there is some kind of pattern which guides you too the root of the cause.

If you need further help --> I have currently 2 additional MK3S+ printers at work which needs the REVO upgrade.
I can try to monitor the upgrade in more depth if you need anything? Video of the calibration? Detailed steps? Log file?

I am happy to help!

naldarn, 2023-08-23:
> @D-an-W <https://github.com/D-an-W> just to rule out one more thing: do
> you modify any of the stepper motor settings? In other words modify the
> code.

No, pure stock settings.

naldarn, 2023-08-23:
Mine was a MKS3+ kit directly from Prusa.

jwall87, 2023-08-26:
Having the exact same issue.

gudnimg, 2023-08-27:
I have an idea, there is a possibility the motors are all disabled when the bug appears.

When you see the message `"Please please steel sheet on heatbed."` and the Z-axis is not at the very top of the axis, try moving X, Y, and Z axis by hand.

gudnimg, 2023-08-27:
I can confirm the stepper inactivity timer does expire around the instant when the bug takes place. Though in my test run this didn't cause any issue for me. I can try disabling the MMU to see if that makes any difference.

Is anyone here using an MMU?

[image]

gudnimg, 2023-08-27:
Hmm I only once got the stepper inactivity timeout message. Even if I disable the motor manually, it does get enabled again for the move somehow.

`axis_known_position[Z_AXIS]` is `false` for the entire XYZ calibration (even after running Auto Home and Z calibration) this means `raise_z()` is using the TMC2130 stallguard to stop the Z-axis motor.

gudnimg, 2023-09-01:
This issue will be resolved in 3.13.1: https://github.com/prusa3d/Prusa-Firmware/pull/4357

3d-gussner, 2023-09-04:
Please install https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.1 and report back if the issue have been solved also on your printers.

naldarn, 2023-09-05:
The issue has been resolved, but seems to have spawned a new issue.  I have
never had Y axis crashes unless I have caused them by messing with the
printer in action.  I am now getting many Y axis crashes, 14 of them in the
first print after update, and 23 in the next before I turned off crash
detection.

gudnimg, 2023-09-12:
Bug fix was released today with https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1

Please consider closing this issue.

sdbarker, 2023-09-16:
> The issue has been resolved, but seems to have spawned a new issue. I have never had Y axis crashes unless I have caused them by messing with the printer in action. I am now getting many Y axis [...]

I am seeing this same behavior.

naldarn, 2023-09-16:
The first print after the update to 3.13.1, I had 23y crashes and since
then I have had a total of 47 y-crashes and 2 x-crashes.  I suspect that
the stall guard has been set to a higher sensitivity than previous firmware
releases.  The crashes have decreased in frequency, but before this release
I had encountered 0.

sdbarker, 2023-09-16:
> The first print after the update to 3.13.1, I had 23y crashes and since then I have had a total of 47 y-crashes and 2 x-crashes. I suspect that the stall guard has been set to a higher sensitivity [...]

I just reverted to 3.12.2, and I'll see if that helps anything.

gudnimg, 2023-09-16:
@naldarn @sdbarker FYI we haven't changed any of the motor settings for the past 2 years at least, especially stallguard values.

If these are crashes when _printing_, then they are unrelated to this Github issue.

I had some Y-axis issues myself a few weeks ago and it turned out to be a mechanical problem where the Y-axis idler wheel stopped spinning and so significantly increased friction on the belts. It was just too tight screw :) If that's the case problem should also appear on 3.11 and 3.12 (not just 3.13)

over 20 Y axis crashes is a lot! Mechanical issue is more likely than firmware issue (otherwise I'd expect we'd be flooded with bug reports).

sdbarker, 2023-09-18:
3.12.2 and 3.12.0 also had phantom Y crashes.  3.11.0 printed my first benchy without one.  

I'm going to stick with 3.11 or now since I've got some parts that I need to run off.  I'll try and give the upgrade a shot after a while and see if I can find a reliable repro.

orbitinstasis, 2023-10-15:
Can't get through a single pre print height calibration without getting "some problem encountered, z leveling enforced". Close to moving back to 3.11.

Prusa-Support, 2023-11-08:
Please notice that <_problem encountered, z leveling enforced_> errors and axis collisions are not relevant to this specific issue.

The problem described by the author of this issue (main post at the top) has been addressed by the [FW release 3.13.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2), and therefore, this issue will be closed as solved.

We thank @sarusani for contributing to solving this problem.

Michele Moramarco
Prusa Research

closed 2023-11-08 by Prusa-Support