prusa3d/Prusa-Firmware #4317: [BUG]<G2 and G3 arcs ignore the Z parameter on the MK4>

MichaelJLew, 2023-08-05:
**Printer type** - [MK4]
**Printer firmware version** -  [4.7.2]

**Describe the bug**
 G2 and G3 controlled arc commands ignore the Z parameter. That means that arcs cannot be used to make helices where the Z level at the end is different from the start.

Arcs where the Z level is supposed to change print as flat arcs. A vase-style cylinder constructed with arcs prints all on one level and ends up as a flat donut of excess extrusion.

I have a couple of threads on the forum exploring this problem, and one user showed that the problem does not happen on his Prusa Mini, so it seems to be MK4 specific. 
See here: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/g2-and-g3-arcs-dont-work-properly-on-mk4/
and here: https://forum.prusa3d.com/forum/general-software-discussion/what-makes-gcode-file-printable-on-prusa/paged/2/#post-663323

  
**To Reproduce**

The attached model is a hollow cylinder made of arcs. The lower 5mm has layers and the arcs print correctly (even if the Z parameters are ignored). The upper 5mm is a helix that should give a vase-style cylinder. 

In the file the header and footer portions are taken from a PrusaSlicer cylinder print file and the model portion is hand coded (with custom software).
  
**Expected behavior**
  The file should yield a straight sided cylinder 10mm tall but on my MK4 only the lower portion prints correctly. The upper half is a mushy donut because Z height stalls at 5mm even as the rest of the print tries to continue.
  
**G-code**

[Arc test cylinder c.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/12265246/Arc.test.cylinder.c.gcode.zip)

MichaelJLew, 2023-08-05:
The issue has been now confirmed on the forum by @ssmith. The test file yields a 10mm cylinder on their Mini, but not on their MK4.

gudnimg, 2023-08-05:
Hi MK4 and XL issues should be reported here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

This repository is only for the 8-bit firmware (MK2.5/S, MK3/S/S+)

MichaelJLew, 2023-08-05:
OK, I was misled by the "Prusa-Firmware" title of this repository. I'll make a new bug report in the right place.

closed 2023-08-05 by MichaelJLew