prusa3d/Prusa-Firmware #4320: [BUG] The Action of new folder inside Prusalink doesn't work

justoeu, 2023-08-08:
**Printer type** 
MK4

**Printer firmware version** 
4.7.2

**SD card or USB/Octoprint**
USB CARD

**Describe the bug and Reproduce**
This Bug is simple to reproduce, Inside the PrusaLink, access the storage click in add new folder, input new name and waiting... This process its not work.

[image]

3d-gussner, 2023-08-08:
@justoeu This repository is for the 8-bit MK3/S/+ and MK2.5/S printers. Please close your issue here and open a new one in https://github.com/prusa3d/Prusa-Firmware-Buddy which is for the 32-bit printers like Mini, MK4 and XL.

justoeu, 2023-08-09:
This Bug is incorrect to this firmware

closed 2023-08-09 by justoeu