prusa3d/Prusa-Firmware #4323: MK3S/+ MMU2S start gcode issue with T[initial_tool] command

mmoraes80, 2023-08-12:
**Printer type** - [MK3S Stock]
**Printer firmware version** -  [3.13.0]

**MMU upgrade** - [MMU2S]
**MMU upgrade firmware version** - [3.0.0]

**SD card or USB/Octoprint** - [PrusaLink 0.7.0]

**Describe the bug**
 When sliced with the MMU2S profile, the print job starts and the printer runs the start gcode normally. However, right until the bed leveling is complete and the print head moves to the origin (G1 Y-3 F1000, G1 Z0.4 F1000), the T[initial_tool] command makes the print head to quickly lift and move all the way to right as in a filament change procedure, then it loads the filament to the nozzle, purges the filament, asks the user to confirm the color change and moves back to origin before starting a new mesh bed leveling instead of print the purge line. So the issue is that no filament change  nor the second bed leveling is supposed to happen at this time.
  
**To Reproduce**
Slice the model with the Stock MK3s MMU2S profile.
  
**Expected behavior**
 Correct filament to load to nozzle after the bed leveling and start printing the purge line.
  
**G-code**bed leveling
  M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.12.2 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

; Send the filament type to the MMU2.0 unit.
; E stands for extruder number, F stands for filament type (0: default; 1:flex; 2: PVA)
M403 E0 F{"" + ((filament_type[0]=="FLEX") ? 1 : ((filament_type[0]=="PVA") ? 2 : 0))}
M403 E1 F{"" + ((filament_type[1]=="FLEX") ? 1 : ((filament_type[1]=="PVA") ? 2 : 0))}
M403 E2 F{"" + ((filament_type[2]=="FLEX") ? 1 : ((filament_type[2]=="PVA") ? 2 : 0))}
M403 E3 F{"" + ((filament_type[3]=="FLEX") ? 1 : ((filament_type[3]=="PVA") ? 2 : 0))}
M403 E4 F{"" + ((filament_type[4]=="FLEX") ? 1 : ((filament_type[4]=="PVA") ? 2 : 0))}

{if not has_single_extruder_multi_material_priming}
;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
; select extruder
T[initial_tool]
; initial load
G1 X55 E29 F1073
G1 X5 E29 F1800
G1 X55 E8 F2000
G1 Z0.3 F1000
G92 E0
G1 X240 E25 F2200
G1 Y-2 F1000
G1 X55 E25 F1400
G1 Z0.2 F1000
G1 X5 E4 F1000
G92 E0
{endif}

M221 S{if layer_height<0.075}100{else}95{endif}
G92 E0

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|SOLUBLE).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}

**Video**
  Please attach a video. It usually helps to solve the problem.

gudnimg, 2023-08-12:
Is this behavior repeatable (does it happen everytime?)

Could you share a video?

> command makes the print head to quickly lift and move all the way to right as in a filament change procedure, it then loads the filament to the nozzle, purges the filament, asks the user to confirm the color change and moves back

This sounds like M600 (filament runout event) but it should be impossible when executing T codes as we are intentionally disabling it.

mmoraes80, 2023-08-12:
> Is this behavior repeatable (does it happen everytime?)
>
> Could you share a video?
>
>
> This sounds like M600 (filament runout event) but it should be impossible when executing T codes as we are [...]

Video has been uploaded. I did not let the printer purge filament and do the second bed leveling.

gudnimg, 2023-08-12:
From the video it definately looks like filament runout event is triggered when it shouldn’t.

if it’s not sensitive, could I get a copy of the gcode? I can try running it on my MK3S+.

mmoraes80, 2023-08-13:
> From the video it definately looks like filament runout event is triggered when it shouldn’t.
> 
> if it’s not sensitive, could I get a copy of the gcode? I can try running it on my MK3S+.

[Shape-Box_0.4n_0.2mm_PETG_MK3SMMU2S_39m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/12329710/Shape-Box_0.4n_0.2mm_PETG_MK3SMMU2S_39m.gcode.zip)

gudnimg, 2023-08-13:
Thanks for the file. I could not reproduce the issue on my end. But one difference is I'm using the SD card to print. I can't rule out that USB printing is behaving differently just yet.

I enabled CMDBUFFER_DEBUG to see what the command queue looks like. So somewhere between `G80` and `T0`, filament runout might be possible according to your video 🤔 

[image]

gudnimg, 2023-08-13:
@mmoraes80 are you familiar with how to extract the logs from the printer? It could tell me whether this is the FINDA sensor complaining or the extruder filament sensor.

gudnimg, 2023-08-13:
I think it's very unlikely this is the extruder filament sensor. Since we see `F?` on your LCD, the condition where `F?` is rendered is also used to block the filament sensor from triggering a runout event (MMU setup only). FINDA runout should also not be possible here because we only allow FINDA to trigger runout event if the E-motor is moving 🤔 On my end, FINDA runout isn't even checked during T0 execution.

gudnimg, 2023-08-13:
@mmoraes80 just an idea... I've seen reports that PrusaLink "could" corrupt g-code files. I haven't seen it myself when using MK4 but have you double checked that? I think there is a possibility for FINDA runout event if `G1 X55 E29 F1073` is run before `T0`.

I'm currently out of ideas to explain this

[image]

mmoraes80, 2023-08-13:
> @mmoraes80 just an idea... I've seen a report somewhere on Github about PrusaLink "could" corrupt g-code files. I haven't seen it myself when using MK4 but have you double checked that? I think there is a possibility for FINDA runout event if `G1 X55 E29 F1073` is run before `T0`.
> 
> I'm currently out of ideas to explain this
> 
> [image]
> 
> I can confirm the g-code above will produce _exactly_ the same result as in your video.

I am not familiar with extracting logs form the printer but I will be happy to do so if you point me to the directions.

The gcode is the standard start gcode from Prusa for the MMU2S profile. Absolutely nothing was changed.

The highlighted G1 code is par the of the purge line, which goes back and forth before priming the nozzle all the way (line #186). Nothing in there justifies the M600 behavior.

gudnimg, 2023-08-13:
> The gcode is the standard start gcode from Prusa for the MMU2S profile. Absolutely nothing was changed.

I think the problem is not in the Slicer but rather communication between PrusaLink (USB) and the printer. Another user in a separate thread @sruggier has described the exact same problem when using Octoprint. So hopefully we can track the issue down. 

> If you follow the video and the code line by line, you can see that the T0 is where the problem 

We need logs to confirm that to be 100% sure.

> All sensors are OK, since the single profile works just fine when the input selection in the screen is done

I suspect it works fine because the toolchange command is not coming from PrusaLink/Octoprint. SD printing also doesnt have this issue. It seems to be related to using USB printing somehow. 🤔

mmoraes80, 2023-08-13:
> I think the problem is not in the Slicer but rather communication between PrusaLink (USB) and the printer. Another user in a separate thread @sruggier has described the exact same problem when using [...]

I juts downloaded the file to a SD card and ran the print from the LCD SD card reader... same issue right after the bed leveling...

gudnimg, 2023-08-13:
Was it exactly the same file as you shared with me? https://github.com/prusa3d/Prusa-Firmware/issues/4323#issuecomment-1676375449 it worked fine for me. I wonder, does the issue disappear if you disable the Raspberry Pi?

mmoraes80, 2023-08-13:
> Was it exactly the same file as you shared with me? [#4323 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/4323#issuecomment-1676375449) it worked fine for me. I wonder, does the issue disappear if you disable the Raspberry Pi?

Exactly same file... I downloaded the zip file I shared with you and sent it to the printer via SD card... I wonder if this is something with the FW... Maybe a reset would do it?

sruggier, 2023-08-13:
Just to be completely sure, I retried a print from an SD card, after a cold boot of the printer, without Octoprint connected, and the problem also happens in that scenario.

gudnimg, 2023-08-13:
Hmm ok so my theory about it being only USB related is not correct. 🤔 

Smells like a firmware issue, I wonder if it appears when Rasperry Pi is connected to the hardware. I have a raspberry pi zero 2 W, I can try find the correct pins to connect it and see if that changes anything.

mmoraes80, 2023-08-13:
If I turn the RPi port off and run the file from the SD card, it works.

sruggier, 2023-08-13:
Interesting, maybe we have different problems.

DRracer, 2023-08-14:
:thinking: the whole issue looks to me like an ordinary filament runout, not running g-codes out of order.

@mmoraes80 May be a stupid question - do you have your MMU enabled in the Settings menu?

I tried starting a G-code with RPi port [on] but my MMU still grabbed a filament at the start and fed it into the extruder instead of performing a non-MMU filament change.

mmoraes80, 2023-08-14:
> 🤔 the whole issue looks to me like an ordinary filament runout, not running g-codes out of order.
>
> @mmoraes80 May be a stupid question - do you have your MMU enabled in the Settings menu?
>
> I tried [...]

Yes. the MMU2S port is enabled. This issue was also happening in 3.12.0 which did not have this option, so the MMU was always enabled.

In summary:

- Print job with MMU2S - Single profile, running from SD card - NO ISSUE
- Print job with MMU2S - Single profile, running from PrusaLink - NO ISSUE
- Print job with MMU2S profile with RPi port turned OFF, running from SD card  - NO ISSUE
- Print job with MMU2S profile with RPi port turned OFF, running from PrusaLink - NO ISSUE
- Print job with MMU2S profile with RPi port turned ON, running from SD card  - ISSUE
- Print job with MMU2S profile with RPi port turned ON, running from PrusaLink - SSUE

I factory reset the printer last to check if the issue could be related to the FW flashing but the issue persists.

After some further investigation, I am starting to think that this issue is actually related to PrusaLink as I do recall that this issue was intermittent with 0.7.0 RC which, at the time, I thought it would a filament sensor related, ruled out when I turned off the RPi port. and, according to my posts in the community, a simple printer reset would suffice 8/10 times.

sruggier, 2023-08-14:
An update from me: I reproduced this again, but with serial logging enabled in Octoprint. This should be the relevant part needed to understand what's happening, and at what point in the print:
```
2023-08-14 19:16:00,313 - Send: N20 G80*35
... [lots of noise from the MMU2, with temperature/power updates every two seconds]
2023-08-14 19:17:19,076 - Recv: LCD status changed
2023-08-14 19:17:19,078 - Recv: FINDA filament runout!
2023-08-14 19:17:19,079 - Recv: ok
2023-08-14 19:17:19,082 - Send: N21 M403 E0 F0*20
2023-08-14 19:17:19,083 - Recv: Resend: 20
2023-08-14 19:17:19,090 - Recv: ok
2023-08-14 19:17:19,092 - Recv: LCD status changed
2023-08-14 19:17:19,093 - Send: N20 G80*35
2023-08-14 19:17:19,095 - Recv: echo:Enqueing to the front: "M600"
2023-08-14 19:17:19,098 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 19
2023-08-14 19:17:19,099 - Recv: Resend: 20
2023-08-14 19:17:19,104 - Recv: ok
2023-08-14 19:17:19,107 - Send: N21 M403 E0 F0*20
...
2023-08-14 19:17:21,084 - Recv: echo:busy: processing
...
2023-08-14 19:17:23,082 - Recv: echo:busy: processing
...
2023-08-14 19:17:25,085 - Recv: echo:busy: processing
...
2023-08-14 19:17:27,084 - Recv: echo:busy: paused for user
```

I'm assuming this is partly a hardware issue, or it would be easier to reproduce on other printers, but it's hard to speculate beyond that without more detailed information from the firmware.

sruggier, 2023-08-15:
Earlier, you said this:
> FINDA runout should also not be possible here because we only allow FINDA to trigger runout event if the E-motor is moving

This is happening before the initial load, so at the point in the print, it's expected that the FINDA sensor would be false. In particular, on [this line](https://github.com/prusa3d/Prusa-Firmware/blob/3766aadd9943608a87c46634861331a53a718000/Firmware/mmu2.cpp#L205), it's not a bug for the FINDA sensor to be false, but [`check_fsensor`](https://github.com/prusa3d/Prusa-Firmware/blob/3766aadd9943608a87c46634861331a53a718000/Firmware/Marlin_main.cpp#L523C1-L528) shouldn't be returning true, and yet it is. Looking at the definition, it won't return true unless it thinks the e-axis is active, so the problem, in my case, seems to come from [e_active](https://github.com/prusa3d/Prusa-Firmware/blob/3766aadd9943608a87c46634861331a53a718000/Firmware/planner.cpp#L514-L529) returning true, for whatever reason, while the printer is finishing up mesh bed leveling (G80).

sruggier, 2023-08-15:
I haven't replicated your issue, though: my Pi port should be disabled, assuming it wasn't unintentionally re-enabled. I'll doubke-check next time I'm at my printer.

gudnimg, 2023-08-15:
Maybe there is a retraction move somewhere in G80 code 🤔 that could explain it.

sruggier, 2023-08-15:
In that case, I should see a runout event on every execution of G80, right? It would also be a problem for every owner of an MK3S+/MMU2S running the latest firmware. Is that possible?

Do you have any other troubleshooting ideas, or will I need to flash a custom firmware build that logs more information via serial about what's in the block buffer when e_active executes?

As an aside, I successfully worked around the problem this morning by preselecting the correct lane in the MMU, then pushing the filament in until the sensor is triggered 🙂

gudnimg, 2023-08-15:
I found this code in G80, I wonder if it's related somehow. Just a guess.
```
if (((int)degHotend(active_extruder) > extrude_min_temp) && eeprom_read_byte((unsigned char *)EEPROM_TEMP_CAL_ACTIVE) && calibration_status_pinda() && (target_temperature_bed >= 50)) {
        current_position[E_AXIS] += default_retraction;
        plan_buffer_line_curposXYZE(400);
    }
```

On stock MK3S+ there shouldn't be any PINDA calibration because SuperPINDA doesn't have a thermistor. But I suppose it can be possible if this was an upgraded printer. This code is compiled into the MK3S+ firmware.

@sruggier Could you add a log there and see if this `if` statement triggers? I can also send you a firmware hex if you want (assuming you have spare time).

gudnimg, 2023-08-15:
Just a random thought, does the issue go away if you set the Bed Target Temperature to 40°C? That should make sure the above `if` statement is not executed.

sruggier, 2023-08-15:
Thanks! That looks very promising. I tested running just G80 with a cold bed, and the issue did not reproduce. I can't test today, because a print is ongoing, but tomorrow, I should be able to reproduce it outside a print by setting bed and extruder temperatures.

I'm guessing that the best way to solve this would be to avoid retracting if the extruder IR sensor isn't active, with a comment added to explain that the e-axis movement there will trigger runout detection.

This would be my first time building firmware, but the build process looks simple enough. Flashing from Octoprint is also very convenient (no need to move the printer or a computer), so I should be able to find time for this tomorrow, or sometime this week. Hopefully I can also implement and test a fix, but we'll see.

DDuck007, 2023-08-16:
I had exactly the same problem and did a small change in the firmware and informed Prusa-support about it - see change below:

[image]

mmoraes80, 2023-08-16:
> I found this code in G80, I wonder if it's related somehow. Just a guess.
> 
> ```c++
> if (((int)degHotend(active_extruder) > extrude_min_temp) && eeprom_read_byte((unsigned char *)EEPROM_TEMP_CAL_ACTIVE) && calibration_status_pinda() && (target_temperature_bed >= 50)) {
>         current_position[E_AXIS] += default_retraction;
>         plan_buffer_line_curposXYZE(400);
>     }
> ```
> 
> On stock MK3S+ there shouldn't be any PINDA calibration because SuperPINDA doesn't have a thermistor. But I suppose it can be possible if this was an upgraded printer. This code is compiled into the MK3S+ firmware.
> 
> @sruggier Could you add a log there and see if this `if` statement triggers? I can also send you a firmware hex if you want (assuming you have spare time).
> 
> EDIT:
> 
> This E-axis move is a bit suspicious it plans a 1mm extrusion at 400mm/s! Maybe this is fast enough so FINDA runout does not always happen. FINDA runout is definately not sampled at 400Hz :)

That's interesting... My Printer is a MK3S stock, no upgrades, besides the bed bearing holders.

gudnimg, 2023-08-16:
@mmoraes80 I’m not saying this is the issue. It’s just one possibility. Does the issue go away on your end if the bed temperature target is less than 50C? For example 40C or lower

mmoraes80, 2023-08-16:
> @mmoraes80 I’m not saying this is the issue. It’s just one possibility. Does the issue go away on your end if the bed temperature target is less than 50C? For example 40C or lower

I just tried to print a cube with the bed at 40C and the issue did not happen... The printer head moved to the origin after the bed leveling procedure and the correct filament was loaded to the nozzle.

gudnimg, 2023-08-16:
If @sruggier also sees the same result then I can create a pull request to fix it this weekend.

gudnimg, 2023-08-16:
Thinking aloud some more…

If I understand correctly the PINDA on MK3S has a thermistor. But the PINDA on MK3S+ does not. This might further suggest the G80 is the issue. 

This suggests the issue mainly affects stock MK3S users and not MK3S+ (unless the MK3S+ is an upgraded MK3S and still retains old PINDA temperature calibratiom)

sruggier, 2023-08-16:
That would apply to mine, I didn't wipe the memory when I upgraded it to MK3S+.

gudnimg, 2023-08-16:
> That would apply to mine, I didn't wipe the memory when I upgraded it to MK3S+.

Nice, I’m hopeful we have found the problem then 💪

gudnimg, 2023-08-16:
Found this old issue which sounds like the same issue. https://github.com/prusa3d/Prusa-Firmware/issues/1948

Braddg, 2023-08-16:
I'm having the same issue with filament runout after the bed leveling  (Upgraded MK3S+ with MMU2S and Octoprint).  Just tried printing at 40C bed temp.  After bed leveling, the correct filament loaded as expected.  I tried it again with only changing the bed temp to 50C, Octoprint Terminal reported "Recv: FINDA filament runout!" after bed leveling.  Changed back to 40C bed temp, just to be sure I can turn it on and off consistently, and the filament loads as expected.   Thank you guys for your efforts on this!  Its been driving me nuts.

gudnimg, 2023-08-16:
@Braddg Thanks for your report. I’ll prepare a fix for this long-standing issue 🤞Hopefully it can be squeezed into 3.13.1.

sruggier, 2023-08-16:
Yeah, I wouldn't be surprised to see more instances of this issue - it has been a long-standing problem for me, but what changed recently is that with the latest firmware, it now always happens instead of sometimes. In hindsight, I should have raised it for discussion earlier.

sruggier, 2023-08-16:
I can also confirm that if I preheat the bed and run a mesh bed levelling, the problem magically occurs, as we'd expect. I didn't distinguish whether it's the set temperature that matters, or the actual temperature, but I guess that's not really necessary. From the code, it looks like it's the set point for the bed, and the actual temperature for the nozzle.

They heat up too fast to test what happens when the set point is hot enough, while the actual temperature is too cold. It looks possible to test the reverse situation for at least the bed. I just won't do it unless it's going to be useful.

@Braddg try my workaround while you're waiting: preload the first filament (the one your G-code will use during the initial load), push it until the FINDA triggers, and then start the print like that. The MMU3 firmware is smart enough to recover from that situation during the tool change, it'll proceed normally without any intervention.

Braddg, 2023-08-16:
@sruggier I just started a print with your workaround, at 60C bed temp.  After bed leveling, the selected filament loaded and the print started.  Thank you!

mmoraes80, 2023-08-16:
> I can also confirm that if I preheat the bed and run a mesh bed levelling, the problem magically occurs, as we'd expect. I didn't distinguish whether it's the set temperature that matters, or the [...]

I tried the same work around with a 245C / 85C with the filament preinserted in the MMU2S until ensuring the FINDA was triggered. The print head did not move for a filament change this time, but the screen kept showing the new Retry message and the print job did not start. I pressed the knob and the LCD went back to the main screen with the message "Engaging idler..." but nothing happened.

MK3S 3.13.0 and MMU2S 3.0.0 - both stock

closed 2023-08-17 by DRracer

sruggier, 2023-08-25:
Pushing the filament in before every print gets old pretty fast, so I switched to an easier workaround: in PrusaSlicer, click on Printer Settings, Custom G-code, and Start G-code. Find the G80 instruction that looks like this:
```
G80 ; mesh bed leveling
```
and replace it with this:
```
M140 S49 ; work around issue Prusa-Firmware#4323
G80 ; mesh bed leveling
M190 S[first_layer_bed_temperature] ; revert workaround for #4323
```

After upgrading to a firmware version that fixes this issue, one can revert the print settings change. There might be a risk of producing inaccurate mesh bed levelling data, because the bed will be cooling off a bit during bed levelling, but I think it should be fine in practice.