prusa3d/Prusa-Firmware #4326: [ENHANCEMENT] Add a thermal safety check to the printer using a third thermistor

RWL-69, 2023-08-13:
**Enter what type of printer or upgrade the enhancement applies to.**

MK3S+ and possibly others

**Is your enhancement related to a problem? Please describe.**

I ran into an issue where my printer was showing the correct temperature in the display but was way over temperature by 50 degrees as measured with an external type "K" thermocouple.  Filament was coming out like water and the temperature was high enough to damage the hotend housing. 

  **A clear and concise description of what the problem is.**

As the printer was showing what it thought was the correct temperature, there was no warning or shutdown during the print process.    If this was a long print, it could have caused a fire, and as many long prints can be unattended, this is a problem.

In my case, it looks like the thermistor is not working as expected and not linear as the display was showing the correct temperature.
  
[More details with photos.](https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/thermal-issue-over-temperature-but-display-is-normal/) on the Prusa English forum.

**Describe the solution you'd like**

Looking at the Einsy/Rambo board, there is a location for a third thermistor.  (TP15, TP20 and TP21) Enhancement would allow an individual to put a second thermistor to the heater block area to measure and test for a high temperature.  This thermistor would have a maximum temperature limit which it would shut down the printer, just as the one in the heater block.  

This third thermistor could also be used as a comparison between the heater block temperature by comparing the difference between the two readings.  If the the difference is too large, then the print process would alarm and be shut down and the printer cool down cycle started.  If the temperature doesn't drop in a short while, shut the whole process down.

If the third thermistor is not attached to the block, the ambient temperature will be below the heater block.  If this temperature is higher, then there is a problem with the temperature not being regulated properly.

Software would have to take into account the heating and cooling cycled of the printer and check to see if these are in progress.  Maybe only test once the temperature gets within the range on the heat cycle and check for temperature to be dropping during the cooling cycle.

There are ways of monitoring for over heating issues using OctoPrint or other tools but this would be part of the printer and could be incorporated in the design for future products.   An extra safety feature is never a problem and may be a selling feature.

3d-gussner, 2023-11-15:
@RWL-69 The latest firmware has Thermal Model which should catch these kind of issues. Which firmware do you have installed?

RWL-69, 2023-11-17:
I have been using one version down on my software with the Thermal Model.

As an tech and a person that has had a printer go into thermal runaway, this is an issue that I feel is a safety issue.  In my case, the thermistor stated that the temperature was normal yet, when measured with a K type thermocouple, it was >50 degrees over what the display was stating.  It was hot enough to start melting the fan shroud printed out of ASA and damage my SuperPINDA probe.  PETG came out of the nozzle like water.

As for a thermal model, if the thermistor is damaged, will the program detect that issue?  I have not looked closely at the thermal model code, but I do have my concerns about a single point of failure.  If the thermistor is out of calibration, will it be able to detect that it is reading the wrong temperature?  The thermistor may be looking normal but outside the operating range it is supposed to be.   I understand that the thermal model is supposed to be looking at the heating curve for the print head.  You have to calibrate with the "suspect" thermistor in place.  If you calibrate with a bad thermistor, will it be able to catch that issue since the software is using the same thermistor to say that the bad one is bad?

In temperature.cpp, there is a section for AMBIENT_THERMISTOR so, it shouldn't be hard to implement.  It may just be a flag to get a reading.   And some way of setting an ambient offset value, which would depend on the placement of the ambient thermistor.

I have just started to use Octoprint and there is a plugin for checking other temperature locations and that is an option but I still feel that an extra thermistor can provide a level of safety above what the software can discover. 

In hindsight, I wish I hadn't opened up my thermistor so I could do some tests and compare it to a known good one or a bead thermistor.

3d-gussner, 2024-05-02:
@RWL-69 I am closing this request as it would need a hardware change in the heater block to allow two thermistors being placed.
BTW we made few blocks like this and tested the Thermal model but with the 2nd one connected to a different device.

closed 2024-05-02 by 3d-gussner