prusa3d/Prusa-Firmware #4329: [BUG]3.13.0 Revo rapid change Thermal Anomaly

redzero36, 2023-08-16:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0-6873
SD card

**Describe the bug**
  Trying to perform first layer calibration but getting thermal anomaly. 
  
**To Reproduce**
  Installed Revo rapid change with .8mm nozzle. Updated firmware to MK3S+ Revo 3.13.0-6873. Perform first layer calibration
  
**Expected behavior**
  Execute first layer calibration
  
**Video**
  Please attach a video. It usually helps to solve the problem.

https://github.com/prusa3d/Prusa-Firmware/assets/5809820/728af3f7-7623-43d2-b1c5-61119843cb1c

D-an-W, 2023-08-16:
Can you try sending the command below before running the calibration (it works ok here)…

redzero36, 2023-08-16:
> 

How does one send commands to the printer? Specifically on linux.

D-an-W, 2023-08-16:
I use Terminal in Octoprint (Forgot to mention that sorry!).

SNDR0, 2023-08-16:
Have you done a thermal calibration after the hardware changes?

redzero36, 2023-08-16:
Is thermal calibration “Thermal Model cal.” If so the yes. It stops before completes the calibration. 

Uploading 23-08-16 11-53-22 10562.mp4…

punkyman, 2023-08-18:
Same issue here after upgrading to firmware 3.13.0 for mk3s with revo upgrade. Also tried a full factory reset without any better results.

The printer is an MK3 upgraded to MK3S. Just checked the revo model I have, it’s from the upgrade kit, I suppose it’s the 24V 40W heater block since the 60W seems to be sold for high flow purposes; the firmware has been flashed accordingly.

Having no filament loaded makes the TM test fail in seconds, while with filament loaded, the test does heat up to 230c, cool down, moves the extruder to a corner, and then fails with the « aborted » message.

For now, the printer is running with model disabled.

werkstrom, 2023-08-18:
Can confirm, The issue with Revos still persist. I cannot perform an initial pid cal, cannot heat the nozzle. As soon as temp reaches 50-65 deg C any action is aborted and temp set to 0. Have installed 3.13.0 REVO version of FW. Appears to work fine on downgraded to 3.11 (so does not appear to be HW related)

werkstrom, 2023-08-19:
@redzero36 You did use the Revo specific FW for your setup (there are four different ones in the revo.zip), correct?

redzero36, 2023-08-19:
@werkstrom Yes. Installed the MK3S+ revo version. Not the HF 60W one. Used the second of the four files.

werkstrom, 2023-08-19:
FYI, I'm now back on 3.13.0.6873 REVO. Having disabled  the temperature model functions (Using M310 S0 followed by M500 to save to EPROM) no issues. So a lot points to this being related to the temperature model functions and thus them not functioning correctly... still.... And since I'm under the assumption there are no such features in 3.11 and before. There is no added "fire hazard risk" disabling this compared to reverting to 3.11 unless some other "legacy" safety feature has been removed in 3.12 and replaced by the temp mod stuff. Perhaps someone else knows and could clearify 🙂

tacomaboy, 2023-08-22:
I also have this issue with the new REVO I purchased.  I was able to roll back to 3.11 and it works now.

PaulandChrissy, 2023-08-22:
I will stay on 3.11 until it's resolved on 3.13 or above

redzero36, 2023-08-23:
@werkstrom yup. I see an R after 3.13 so I guess thats the revo version. I just setup an mini pc for octoprint. Disabled TM. Still on 3.13. Things are up and running and just printed a first layer calibration.

antimix, 2023-08-26:
**IT SEEMS THE MK3S+ WORKS WELL WITH FW REVO 3.13.0**, but _you have to do an additional step._

I have upgraded my MK3S to MK3S+ and in this occasion I also installed the REVO.
I installed the REVO 3.13.0 MK3S FW, and performed the wizard setup to recalibrate the printer. Unfortunately nothing worked.

1.  As soon as the printer tried to raise the hot-end temperature I always got strange THERMAL ANOMALY errors, or  the printer just stuck, or reset the target temp to 0°.
2.  The PID test failed after few seconds and could not be set or completed. 

I spent hours trying to make it work, until in order to solve another issue I had on the printer, I decided to FACTORY RESET & DELETE ALL DATA from the printer. 

After the reset, the printer restarted, and the Configuration Wizard started. This time all the wizard tests were successfully passed, all the hot-end access worked, the PID tuning config either worked perfectly, and the successful message come from the printer on the LCD.

Now I have done also a couple of print, and all worked fine. And also the other issue I had automagically was gone.

So, the Prusa Firmware for some functions make really mess when it use values in conflict with the EEPROM values stored, so just clear all the EEPROM data, and restart, and then all should work fine.

antimix, 2023-08-27:
> Same issue here after upgrading to firmware 3.13.0 for mk3s with revo upgrade. Also tried a full factory reset without any better results.
>
> The printer is an MK3 upgraded to MK3S. Just checked the [...]

I am able to print with the 3.13.0. 
At the beginning it had your same behaviour, but then I got the idea to erase the EEPROM data that were probably confusing the FW, so I performed a FACTORY RESET (All Data) option, and then the printer restarted from scratch. Then I performed the Wizard, realigning the axes, and performing the PID thermal test, and all finished successfully, and I was able to print.
I think that even if you have the REVO and try to perform the TM setup, the FW insist to use some old stored values for E3DV6.

3d-gussner, 2023-08-28:
@antimix 

> the FW insist to use some old stored values for E3DV6.

Yes flashing the FW3.12.0 will write E3D v6 Tjermal model and PID values to the EEPROm which aren't changed by flashing the `E3DREVO_fw_*.hex` file.
A Factory reset All data will delete the eeprom and write the default values (E3DREVO_*hex has the PID and TM values for the REVO).

E3D and we are working on a solution.

@everyone Please ensure that you flash the correct hex file. Make sure that you flash the `E3DREVO_fw_*.hex` file and flash it only with PrusaSlicer as OctoPrint Firmware updater doesn't flash the xflash correctly.

3d-gussner, 2023-08-28:
@redzero36 Just a side note when you change the nozzle size please don't forget to adjust the Settings -> HW Setup -> Nozzle.
This will ensure that you get warnings if a 0.4 nozzle gcode is used while you have a different nozzle size installed. Also the first layer calibration will be adjusted for the different nozzle size.

tacomaboy, 2023-08-28:
Resetting my EEPROM worked, I am now able to run with the Revo 3.13.0 firmware :)

OlmiVanguard, 2023-09-04:
full factory reset didn't solve for me. using revo high flow hotside. had to upgrade fw to use mmu3. besides thermal anomaly, getting weird "filament unexpectedly detected" error.

mmcglumphy, 2023-09-10:
I'm still getting Thermal Anomaly beeps with REVO 3.13.1 firmware. I've performed factory reset, pid tune, and successful thermal calibration on two MK3s+-R's. Both will occasionally beep with anomalies.

Some days they go without a sound. Last night one was beeping once every few minutes. After about 20 min of intermittent beeps and warnings it was fine and had no further warnings for another 5 hours when the print finished.

The only pattern that I've seen is that it generally occurs when the nozzle is at a lower Z height. Could the heated bed ambient temps affect nozzle temp monitoring?  I'm printing exclusively in PETG. 235/85 - 240/90.  

I'm backing down to 3.11, because these are farm machines with lots to do.

D-an-W, 2023-09-10:
I too still get the occasional Thermal Anomaly or just the beeps with no message, never during a print however.

To be honest I don’t know how the E command in “M310 W1.00 E1.74” affects it and if it could help in fine tuning. 

Ideally for me an option to turn off / disable checking during loading and unloading of filament would fix things. 

Hope you manage to get it sorted to keep the farm running ok!

mmcglumphy, 2023-09-10:
Thanks. Luckily, I've only added REVO to two machines. :) I also tried that M310 command from that start with no change in behavior. 

Another observation I just made is that when on 3.13.1 the nozzle temp is somewhat erratic. It bounces +/- 2C around the set temp. Now that I have them downgraded to 3.11, the temp stays solid on the desired value.

mmcglumphy, 2023-09-11:
Well, hell. It's possible this is a problem with the REVO extension cable connectors for both of these printers. One just refused to heat at all, even on 3.11. So, I messed with the wiring, reseated all connectors, and all is well again. So, everything above may be a false alarm. When I am able, I'll reupdate to 3.13.1 and see what happens.

mmcglumphy, 2023-09-15:
I've found that the cable connector is not the cause of the anomalies. With my two REVO MK3S+ printers, I have one running 3.11 and one with 3.13.1. Both have been printing non-stop for the last 4 days. Just now the 3.13.1 machine beeped with an anomaly. Nothing has changed in those 3 days, not even the filament type. The print head is 2.3mm above the bed. So, there is still an issue with 3.13.1 tolerances.

RcrossF, 2023-09-23:
I also have incessant beeping during prints, seems to be during long extrusion moves. Newly installed revo on FW 1.13.1 MK3S with an 0.6mm obxidian nozzle, nozzle size set in FW and PID, TM calibration both run.

klsc18, 2023-10-06:
Today I installed my new REVO update, flashed the E3DREVO_fw_MK3S_MK3S+_3_13_1_6876.hex, and get the same errors. Even a factory reset didn't bring any improvement. Every time the temperature reaches approx. 60°, it stops. Even the setup wizard stopped at this point.

gudnimg, 2023-10-06:
> I'm not sure whether it will also work with the MMU3 with the old firmware 3.11

@klsc18 It won't work. The MMU3 firmware can't communicate with printers on FW 3.12 and older because the communication protocol is completely different. (Not backwards compatible)

klsc18, 2023-10-06:
The hint from @werkstrom seams to work
(Using M310 S0 followed by M500 to save to EPROM)
Now I can heat up

klsc18, 2023-10-07:
> > Today I installed my new REVO update, flashed the E3DREVO_fw_MK3S_MK3S+_3_13_1_6876.hex, and get the same errors. Even a factory reset didn't bring any improvement. Every time the temperature reaches approx. 60°, it stops. Even the setup wizard stopped at this point.
> > I'm using a new MMU3 and I'm not sure whether it will also work with the MMU3 with the old firmware 3.11, which seems to work with the Revo. Does anyone have any experience with this?
> > The hint from @werkstrom seams to work
> > (Using M310 S0 followed by M500 to save to EPROM)
> 
> I use `M310 W2.00 E2.00` as I was getting similar errors, had none since, the REVO warms up too quick for the stock settings and triggers warnings.

Feedback: your "M310 W2.00 E2.00" also seems to work. I have now printed PLA and PETG several times.

klsc18, 2023-10-07:
But since the "Thermal model calibration" aborts EVERY TIME, you cannot successfully complete the setup wizard. This means that every time the printer is switched on or reset, a message appears indicating that the setup has not yet been completed. I hope a bug-fixed firmware image will appear soon.

D-an-W, 2023-10-07:
> But since the "Thermal model calibration" aborts EVERY TIME, you cannot successfully complete the setup wizard. This means that every time the printer is switched on or reset, a message appears indicating that the setup has not yet been completed. I hope a bug-fixed firmware image will appear soon.

I add that line before running  the Wizard.

MartinMajewski, 2023-10-21:
The new TM is shi**y and does not work with REVO reliably. I don't think it will ever do, as it is too sensitive to everything that is not stock Prusa. With their own fast removable nozzle system, there is no incentive for them to make it better. Both Revo-modded printers of mine worked fine for two months now, and yesterday (right when I needed them most), I went into this Thermal Anomaly limbo.

I've deactivated the TM now, and they work fine again.

OlmiVanguard, 2023-10-22:
> The new TM is shi**y and does not work with REVO reliably. I don't think it will ever do, as it is too sensitive to everything that is not stock Prusa. With their own fast removable nozzle system, [...]

What is the point of touting open source as a virtue if a "drop in" replacement upgrade from your closest third party collaborator doesn't even work with your printers?

While trying to compete with Bambu's superior machines, Prusa decided to drop the 10 discount on assembled i3 printers. Now they've decided to retrospectively make third party hardware less compatible with the MK3... Seems like worst of both worlds. You get the joy of assembling the printer and wondering which screw is causing vibrations and rattling, with all the disadvantages of being locked fenced into a de facto ecosystem.

MartinMajewski, 2023-10-22:
> > The new TM is shi**y and does not work with REVO reliably. I don't think it will ever do, as it is too sensitive to everything that is not stock Prusa. With their own fast removable nozzle system, there is no incentive for them to make it better. Both Revo-modded printers of mine worked fine for two months now, and yesterday (right when I needed them most), I went into this Thermal Anomaly limbo.
> > I've deactivated the TM now, and they work fine again.
> 
> What is the point of touting open source as a virtue if a "drop in" replacement upgrade from your closest third party collaborator doesn't even work with your printers?
> 
> While trying to compete with Bambu's superior machines, Prusa decided to drop the 10 discount on assembled i3 printers. Now they've decided to retroactively make third party hardware less compatible with the MK3... Seems like worst of both worlds. You get the joy of assembling the printer and wondering which screw is causing vibrations and rattling, with all the disadvantages of being fenced in by a de facto closed ecosystem.

I don’t think that this is the right place to start a discussion about Prusa‘s business strategies but I'm completely with you. This "rattling screw issue" is a perfect picture. 👍

JPTa, 2023-10-25:
> Can you try sending the command below before running the calibration (it works ok here)…
> 
> M310 (to check current values) M310 W1.00 E1.74

This helped me. My setup is kinda unique though as the Revo in it is one of the beta versions and I replaced the thermistor in that as well.

beelsebob, 2023-10-29:
Given that this seems to happen only instantaneously, and then immediately go back to normal, it seems like this could be easily fixed by simply requiring a few more anomalous samples before triggering the alarm.  Super annoying though.

ndemarco, 2024-01-04:
I'm on v3.13.2 and cannot get past this in any reasonable way.

ndemarco, 2024-01-07:
@3d-gussner 🙏 step 4 in [#4451](https://github.com/prusa3d/Prusa-Firmware/issues/4451) solved the issue.

ndemarco, 2024-01-08:
> @3d-gussner 🙏 step 4 in [#4451](https://github.com/prusa3d/Prusa-Firmware/issues/4451) solved the issue.

Ok, I was too quick. The BEEP-BEEP...BEEP `THERMAL ANOMALY` reappeared on the next print. I'll comment on [#4451](https://github.com/prusa3d/Prusa-Firmware/issues/4451)

j0serx, 2024-01-15:
My MK3S+ with revo was working fine on 3.13.2 but randomly started with the beeps. I just went back to version 3.11.0. I am also having problems with my prusa mini with revo micro, on version 5.1.0 it seemed to work but after 1 hour it stopped drastically with lots of beeps.

Prusa-Support, 2024-02-24:
Please make sure you have installed the Revo-specific firmware as per the instructions and follow the main post of issue https://github.com/prusa3d/Prusa-Firmware/issues/4451.
If that won't help, please reach out to E3D support at https://e3d-online.com/ because something is potentially wrong with your hotend.
Our Revo-specific firmware was approved by E3D so E3D can help you solve the problem or collect relevant data for us to further tune the Thermal Model, if needed.

Michele Moramarco
Prusa Research

beelsebob, 2024-02-24:
While that might be perfectly reasonable advice, and I can understand wanting the debug data, for some of us, there’s currently:

1. No way to get the printer to stop harassing us about calibration, and requiring manual local intervention on all prints (including those started remotely).
2. No way to get any information about what is out of calibration.
3. No way to correctly calibrate the printer.

please please please provide a way of just telling the printer to shut up about calibration.  At very least please provide a way to remotely say “I know, it’s fine”.

RcrossF, 2024-02-24:
> While that might be perfectly reasonable advice, and I can understand wanting the debug data, for some of us, there’s currently:
>
>     1. No way to get the printer to stop harassing us about [...]

You can disable the TM with 
```M310 S0; M500```

I completed PID tuning & calibrated my model then ran this. It has been working well so far

Prusa-Support, 2024-03-13:
Actually, this and much more can be found in the complete documentation at #3552.
However, in the case of Revo hotends, a most effective user guide is provided at #4451, _including the step where persistent thermal anomalies on Revo are **not ignored** but reported to E3D as a potentially serious hardware concern_.

Michele Moramarco
Prusa Research

ndemarco, 2024-03-14:
E3D has asked me to do a fairly intense amount of work to help troubleshoot
this issue. I'm sharing it here, as it should help E3D (and me!) if
multiple people report their conditions and experiences.

From E3D support:
So that we can investigate this issue further, please can you review the
troubleshooting steps found below and provide the requested information.

1 - Please provide an image of the HeaterCore Serial Number label. Please
see the below example for its location.

2 - Please check that all the connectors or wire splicing from the heater
cores thermistor wire on the HotEnd side to the control board input are
fully seated and stable.

3 - Please can you provide an image of the resistance reading from the
thermistor connector on the heater core itself with the multi-meter set to
200kohms on a manual setting and the in addition to also confirming the
ambient temperature when this measurement was taken?

4 - Please can you provide an image of the resistance reading from the
Heater connector on the heater core itself with the multi-meter set to 200
ohms on a manual setting and the in addition to also confirming the ambient
temperature when this measurement was taken?

*Please note that both the thermistor and heater resistances will vary
based on the ambient temperature so your reading may vary slightly.*

5 - Please can you check and provide images of the crimps inside the Molex
connector on the heater core side If the connectors have not been provided
strain relief the pins may have opened the crimps.

6 - If you Gently pull on the Individual wires of the thermistor or heater
connector does it feel loose or appear to have detached from inside the
connector?

7 - Please can you share some pictures of your setup - in particular,
images of the wiring loom at the Molex connection point of the HeaterCore
itself?

3d-gussner, 2024-03-14:
@ndemarco Thanks for sharing this and maybe that is a hint what for users can look for.

But everyone PLEASE DON'T share anything here!  Go to the E3D support to provide these things.

closed 2024-05-21 by github-actions[bot]