prusa3d/Prusa-Firmware #4333: [BUG] Preheat / Load / Unload nozzle temp threshold & behaviour has changed

linzwatt, 2023-08-17:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0

**SD card or USB/Octoprint**
Repetier Server - not relevant for this issue

**Describe the bug**
I published this as a bug because I didn't see any notes in the release about this behavior changing.
Since updating to 3.13.0 the nozzle temperature threshold for preheat, load and unload seems to have been reduced to 0.

In previous versions and for as long as I can remember with Prusa printers, if you preheat to 230deg for example, the printer would allow you to start loading within +/- 10deg. So as soon as the nozzle hit 220deg it would allow you to load the filament - saving time, since by the time the filament hits the nozzle, the nozzle would be within a couple degrees of the set temp.
Now with 3.13.0 the printer must reach exactly 230deg, which is frustrating because it can hover around 228-229deg for several seconds before finally hitting 230deg.

Additionally, during a filament auto-load, if the temp is sitting at 229 for example, the autoload won't happen as soon as you insert the filament. The printer will switch to preheat mode, wait for it to hit exactly 230, then ask to press the button to load filament. This is also frustrating as most times I've inserted filament expecting an autoload to happen, I have to wait several seconds, then manually press the button.
  
**To Reproduce**
Preheat issue:
Preheat printer, watch as it needs to hit the exact set temperature before allowing you to load.

Autoload issue:
Preheat printer and leave it to heat completely (wait a few minutes), then get ready to do a regular auto-load but wait for it to drop 1deg below the set temperature. Insert the filament, it will go into the preheat screen and wait until it hits the set temp exactly, then will ask you to press the button to load.
  
**Expected behavior**
Before 3.13.0 the printer would allow you to load several degrees before the set-temp, additionally it would auto-load filament as long as nozzle is within a few degrees of the set-temp.

Unsure what motivated this un-documented change of behaviour, it may have been an unintended side-effect of the changes made for the "Heat bed on preheat" feature. Personally, I would love to see a configurable threshold for nozzle temp, that can be set from 0-10 degrees defaulting to 5deg. As the auto-load feature has been ruined in most cases (in my experience over the last week or so) by this behaviour change.

Additionally, with the "Heat bed on preheat" feature, it would be great if this could be toggled on/off inside the preheat screen. It is a hassle to dig through the settings to toggle it each time depending what the user is trying to do. For example sometimes I just need to unload filament and immediately cool down again - in this case I don't need the heat bed. But other times I want to preheat to load and start a print in which case I want the heat bed. If there could be a toggle for the heatbed above the cancel option on the preheat screen, that would make this feature a lot more useful and usable.

gudnimg, 2023-08-17:
Just a note but his is one change I did for 3.13.0: https://github.com/prusa3d/Prusa-Firmware/pull/4033 it was mainly to make things consistant in the firmware to avoid UI bugs.

linzwatt, 2023-08-17:
> Just a note but this is one change I did for 3.13.0: #4033 it was mainly to make things consistant in the firmware to avoid UI bugs.

Thanks for the note 🙂 it seems the temp threshold is much lower than 5deg in your changes, it seems to be at 0 now?

gudnimg, 2023-08-17:
Pinging @3d-gussner @wavexx

wavexx, 2023-08-17:
I don't remember any intentional change in this direction, allowing to load/unload close to the target temp was always a nice feature (and something I wish to see on the buddy fw too TBH)...

PR #4338 (2023-08-21): Fix TEMP_HYSTERESIS calculation
This fixes #4333.
closed 2023-08-21 by DRracer