prusa3d/Prusa-Firmware #4335: [BUG] MMU2S filament speed

Gabri988, 2023-08-17:
**Printer type** - MK3S+
**Printer firmware version** - 3.13.0-6873

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.0

**SD card or USB/Octoprint** - PrusaLink 0.7.0

**Describe the bug**
  The filament slows down too early when loading into the extruder before reaching the printer chimney.
  
**To Reproduce**
  This happens in any circumstance, both during a normal loading made from the LCD and during a single or multicolor print.
  
**Expected behavior**
  The filament should reach the chimney above the bondtech gears faster, and slow down its descent just before activating the sensor.
  Before I updated the firmware of the printer and mmu2s I remember the filament not slowing down to that height in the ptfe tube.
  
**Video**
https://streamable.com/a16fmf

SNDR0, 2023-08-18:
It's not a bug, it's a smooooooth approaching... Looking your video ... You are printing a massive purge tower with a tiny little cute multicolor piece!!!

Gabri988, 2023-08-18:
> It's not a bug, it's a smooooooth approaching... Looking your video ... You are printing a massive purge tower with a tiny little cute multicolor piece!!!

Ok maybe it's incorrect to call this problem bug. But I don't understand how the purge tower correlates with the problem I explained. Regardless of the size of the multicolored object, I note that the advancement of the filament is the same as that seen in the video.

SNDR0, 2023-08-18:
I'm joking... That you had noticed about the approaching of the filament is a consequence of a new firmware written by scratch... So what you are supposed to do this version don't do anymore... For now.. The old firmare on my mmu2s often grind the filament because the approaching was too quick and the filament hit the pulley of the extruder.. with the new firmware it never happened again..

Gabri988, 2023-08-18:
> I'm joking... That you had noticed about the approaching of the filament is a consequence of a new firmware written by scratch... So what you are supposed to do this version don't do anymore... For now.. The old firmware on my mmu2s often grind the filament because the approaching was too quick and the filament hit the pulley of the extruder.. with the new firmware it never happened again.. 

Ok 😄  , anyway it's not a problem. I just thought I was the only one who noticed this behavior with the new firmware.

3d-gussner, 2023-08-23:
@Gabri988 This is not a bug but a feature! To speed up the filament changes we were able to use fast + normal load speeds.

We don't stop just before the gears as tests have shown that the fast speed works well and reliable **within** the bowden tube but causes from time to time issues at the transition from bowden tube to festo or plastic parts.

It is different during unload as the filament passed all transition zones and there we can use fast unload speeds.

Gabri988, 2023-08-23:
> This is not a bug but a feature! To speed up the filament changes we were able to use fast + normal load speeds.
>
> We don't stop just before the gears as tests have shown that the fast speed works [...]

Okay, that's a great explanation. As I said, I thought it should behave differently. Initially I also thought it was a problem with the ptfe tube being too long, but now that I know it's correct the way it works I'm calmer. 😅
Thank you

closed 2023-08-23 by Gabri988