prusa3d/Prusa-Firmware #4336: [BUG] Filament load leaves stepper motor energized indefinitely

fatmann66, 2023-08-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3

**Printer firmware version** - 3.12.2-5713

**Original or Custom firmware** - Original 

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
   USB

**Describe the bug**
  If you load Filament  but then do not print right away extruder motor stays energized indefinitely.  

**How to reproduce**
 preform  filament load routine.
once loaded issue: preheat\cool down. 
come back 1 day later and find extruder motor at 50c temp 

**Expected behavior**
  expected that extruder motor would be disabled at end of load/unload routine.  
if not then when cooldown  is issued  I would expect "disable steppers" command to be issue.
NOT TESTED:  If heating is disabled by safety timer (IE did a filament load and walked away) would extruder be disabled when preheat safety timer is hit??

**G-code**
 N/A

**Crash dump file**
  N/A

**Video**
 N/A

Prusa-Support, 2023-08-17:
As a clarification, this is not a bug but the intended behavior.
The extruder stepper motor will remain engaged after the first input (same as the X and Y stepper motors) and keep the assigned coordinate unless the user intentionally "disables the stepper motors" or reset/reboot the printer.
Also, the stepper motors may easily become warmer than that without posing a problem.

For the time being this is considered to be the correct behavior but the user has the option to disable the stepper motors (or reset the printer) if necessary. This shouldn't be intended as a bug but an enhancement request at this point.

Could you please clarify if the current behavior may be a cause of problems in your experience, or what would be the benefit of having the Extruder stepper motor disabled automatically after certain operations?

Michele Moramarco
Prusa Research

fatmann66, 2023-08-30:
why? for all the reasons it's standard practice  on end gcode to issue m84 as the last command I would think  that those reasons would apply to this scenario as well.  
disabling would reduce power consumption, wear and tear on the hardware, power supply and piece of mind for the user.  I don't really seen an advantage to the E motor being engaged until user interaction turns it off.  there are no forces on the motor that would require it to be engaged for an extended period. 

 I can see the reasoning from both sides.  it's just surprising to come back a day or two later to what I would consider an idle printer and find that  some or all of the steppers are enabled and  above ambient temperatures.

closed 2023-11-23 by github-actions[bot]