prusa3d/Prusa-Firmware #4337: [FEATURE REQUEST] Consistent "latest" link directly to binary firmware

endigma, 2023-08-17:
Using software like octoprint firmware updater, there is the option to "remember URL", this is not very useful since there is no stable URL to pull the latest version of firmware from. I propose that there be a way to automate downloading the most recent (stable, dev, etc) build of the firmware for a device.

Itrel, 2023-08-19:
The nightly/development builds would be very helpful. At the moment I can't run the latest build of MK3S firmware because it is expecting MMU2 build 3.0.1 but we only have access to [3.0.0](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.0/MMU3_3.0.0+814.hex) provided with release https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0.

endigma, 2023-08-19:
Someone could probably PR in a Github action to do this, I suspect that it could update a few different URLs in a template then push to GitHub pages or something.

3d-gussner, 2023-08-22:
The link https://github.com/prusa3d/Prusa-Firmware/releases/latest will always show you the latest stable release.

3d-gussner, 2023-08-22:
@endigma 

> Using software like octoprint firmware updater

Please DON'T USE the octoprint firmware updater as it is missing the https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325 to flash the MK3/S/+ correctly.

Flashing firmware is only supported via Prusa Slicer

endigma, 2023-08-22:
The releases/latest works for the latest release but the filenames inside the release artifacts aren't stable

3d-gussner, 2023-08-23:
> The releases/latest works for the latest release but the filenames inside the release artifacts aren't stable

We are not going to change the filenames as users may need to downgrade. Using same file name would result in multiple files with same name and an extra number behind when downloaded. There is no way for the user to find the correct version or the user has to create a folder for each version and this is not going to happen.

endigma, 2023-08-23:
I suppose every other software distribution on the planet offering predictable download URLs is just doing it wrong.

closed 2023-08-23 by endigma