prusa3d/Prusa-Firmware #4341: [FEATURE REQUEST]PreHeat Nozzle & Bed while gcode transfers over PrusaLink/PrusaConnect

mrjackson, 2023-08-20:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - PrusaLink compatible
**MMU Upgrade** - Any

**Is your feature request related to a problem? Please describe.**
  Larger print files can take a few minutes to transfer.
  
**Describe the solution you'd like**
  Start the printer preheating as soon as a file is sent with "Upload & Print" from PrusaSlicer.
  It could be either a pre-set temperature (based on loaded filament maybe), or actual temperatures from the gcode.
  This would minimize the time to start a print,

3d-gussner, 2023-08-22:
@mrjackson Thanks for the feature request and I like the idea.
This is a feature request for PrusaLink https://github.com/prusa3d/Prusa-Link
Please close it here.

mrjackson, 2023-08-22:
New request created for Prusa-Link, and closing this one.
https://github.com/prusa3d/Prusa-Link/issues/833

closed 2023-08-22 by mrjackson