prusa3d/Prusa-Firmware #4344: [FEATURE REQUEST] Update control panel to use new M701/M702 Z parameter

ttabi, 2023-08-22:
This is a follow-up to bug 3532.

Now that the firmware supports the Z and L parameters to M701 and M702, the control panel should allow the user to specify new values for these options.  This way, the control panel's unload menu option will use the specified Z and L values when sending the M702 command, and the firmware will also use these values when the user inserts new filament to be loaded.

Optionally, the control panel could allow the user to specify separate Z and L values for M701 and M702, so that load and unload behave differently.

ttabi, 2023-08-22:
See also https://github.com/prusa3d/Prusa-Firmware/issues/4115

sarusani, 2023-08-22:
At least for now, the raise when you use the LCD panel or sensor to load/unload filament and the raise you can define with M701/M702 are two separate things.

- M701/M702 will just raise z-axis by the value you provide for this particular load/unload, then lower it again. The value will not be stored.
- The LCD panel and sensor will check if z-axis is below the value that's defined in the configuration (50mm) and only raise z-axis up to that value and not lower it after load/unload. (So if z-axis is already at 50mm or above it won't move it at all. If it's at 40mm it will move it 10mm up.)

Bringing them together isn't as straight forward as it seems. (It's not difficult to code, but we would need to come up with an approach that makes sense for both cases and doesn't brake old g-code.)

It would probably be easier to add a new g-code param to overwrite the preconfigured values for min_z_for_load, min_z_for_unload, min_z_for_swap & min_z_for_preheat.

ttabi, 2023-11-15:
The request is still valid.  I still want some way to adjust the control panel's load/unload behavior.

timur-tabi, 2024-01-15:
The request is still valid. I still want some way to adjust the control panel's load/unload behavior.

Why do I need to keep posting a comment every two months, just because Prusa's developers are ignoring this issue?

3d-gussner, 2024-01-16:
> Why do I need to keep posting a comment every two months, just because Prusa's developers are ignoring this issue?

We use a bot to automatically close stale issues as often issues get forgotten by the owners. @sarusani Wrote a good comment and we didn't see any reaction on that one.

We are not ignoring this feature request but adding menus to the LCD are resource hungry. At this moment I would say  `M701 Z L` are barely used and we will not implement it.
Not every gcode setting makes sense to be added to the LCD menus. 

If you need different default load/unload/z values you can change the `FILAMENTCHANGE_FIRSTFEED` , ` FILAMENTCHANGE_FINALRETRACT`  and `MIN_Z_FOR_UNLOAD` in the variant and `Configuration.h` files.

3d-gussner, 2024-03-13:
Closing issue as no interaction

closed 2024-03-13 by 3d-gussner