prusa3d/Prusa-Firmware #4349: [BUG] MMU3 intermittently loads wrong filament after error condition

happykatz, 2023-08-27:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0
**MMU upgrade** - MMU3
**MMU upgrade firmware version** - 3.0.0

**SD card or USB/Octoprint**
 USB/Octoprint

**Describe the bug**
Intermittently, the MMU3 will attempt to load a filament to the right of the selector when the "retry" option is selected from the main interface after a load error. This does not occur on every error.

This is only an issue when a retry is attempted from the interface. When the middle button of the MMU3 device is pressed, the correct filament will load.
  
**To Reproduce**
 This does not always occur, but these are the conditions present when it does.
1) Wait for a filament load error. In the last instance, the filament passed the FINDA, and failed loading at the extruder: FI=1, FS=0)
2) Clear reason for load error, if applicable 
3) Press "retry" from the interface. In the last instance that I experienced this, the filament retracted past the selector (FI=0, FS=0). I do not clearly recall if the selector re-homed after this operation.
The selector will be aligned with the correct filament and the correct led will be lit. However, the filament to the right of the selector will load into the air, resulting in another fault.

**Workaround**
1) Gently guide the "air-loaded" filament back into position.
2) Press the middle button on the MMU3.
The correct filament will now load to the extruder.

  
**Expected behavior**
  The "retry" interface option should behave the same way as the MMU3 middle button and load the correct filament.
  
**G-code**
[White_Box_PLA_MK3SMMU3_11h1m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/12448462/White_Box_PLA_MK3SMMU3_11h1m.gcode.zip)

**Picture**
[image]

DRracer, 2023-08-28:
@happykatz this issue is usually caused by the misaligned Idler - can you please check, that the idler is homed correctly (when it is homing) and that it doesn't drift over time? Ideally put a camera to your MMU looking at the Idler barrel.

In order to home the Idler "reliably" (within the available limits of the TMC2130) we ignore the StallGuard signal when rolling over the inserted filaments. So the StallGuard signal is only checked when Idler is about to hit the far endstop or the front endstop. After hitting the front endstop, it rotates +18° if I remember correctly. That's the Disengaged position.

In your scenario - does the Idler try to rehome after you press "Retry"?

happykatz, 2023-08-28:
I'd need to check the idler state on a new print (I was focused on the selector). 

Intuitively, it makes sense to me that the idler alignment would cause this problem, as I tightened the tension screws considerably after encountering the issue, and did not see a recurrence during that print.

I am surprised, though, that the behavior of the middle button would be different from the behavior of the "retry" from the interface. (middle button re-homes selector, "retry" does not). Is this intentional ?

gudnimg, 2023-10-01:
@happykatz has the problem gone away? It's been a month and am wondering if we can close this Github issue

happykatz, 2023-10-02:
@gudnimg Yes, let's close this. I haven't encountered this again since I 1) increased the tension on the screws and 2) added auxiliary power to the MMU3 unit (cause of the "retry" failing to re-home the selector ?

closed 2023-10-02 by gudnimg