prusa3d/Prusa-Firmware #4351: [BUG] MMU3 - Intermittent failure to re-home on error recovery

happykatz, 2023-08-27:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0

**MMU upgrade** -MMU3
**MMU upgrade firmware version** - 3.0.0

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
I recently received my MMU3, which is replacing my old MMU2S on my MK3S+. I have successfully built the device and installed it, and have successfully completed the Sheep print with PLA. All of my equipment is stock, including the (much improved!!) filament buffer.

I am using the [rotating rings toy ](https://www.printables.com/model/7876-rotating-rings-toy) with Prusament and Overture PETG as my test print (gcode attached). I chose this object because it has multiple tool changes on each layer, for a total of 360+ tool changes, which makes it a good stress test. At this point, I am experiencing failure on average every 200 tool changes or so - with this print, I will have 1-2 load failures.

The load failures are all the same: the filament will form a string which then gets jammed in the selector as the selector moves, preventing the filament from loading properly. The selector will then try to re-home as a part of its recovery process, dragging the stuck filament with it (see pics). I will then:
1) Manually rotate the selector motor so that it is on the far right hand side (NOTE: the left and right buttons on the MMU are disabled during failure states, so manual rotation is the only option.)
2) Open the MMU by loosening the tension screws on either side of the device
3) Remove the cover for the blades on the selector
4) Cut the filament and remove any filament that remains in the selector
5) Select "retry" from the interface, and allow the selector to re-home and reload the filament.
Most of the time, the system then recovers and can continue.

Twice, however, the selector does not re-home after step 5) above, meaning that the selector will remain on the far right side, while the chosen filament is reloading into the air. In this case, the print is not recoverable, and I must reset the printer and start again, as from the error screen I cannot access any other function of the menu.
  
**To Reproduce**
NOTE: This error is intermittent, and will not always occur under these conditions.
 1) Wait for filament to jam.
 2) Recover filament jam as described above.
 3) Select "retry" from the interface.
If the error occurs, the selector will not re-home, but the filament to be loaded will load anyway into the air.
  
**Expected behavior**
 The selector should re-home, and the print proceed as normal.
  
**G-code**
[3_Cool_PETG_Rotating Rings Toy_0.4n_0.2mm_PETG,PETG,PETG,PETG,PETG_MK3SMMU3_13h56m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/12448487/3_Cool_PETG_Rotating.Rings.Toy_0.4n_0.2mm_PETG.PETG.PETG.PETG.PETG_MK3SMMU3_13h56m.gcode.zip)

**Pics**
[image]
[image]
[image]
[image]
[image]
[image]

DRracer, 2023-08-28:
@happykatz now that's a well documented bug report, thank you :+1: 

I'm wondering how it may happen that the selector doesn't rehome when it is supposed to do so every time ... the only thing which comes to my mind currently: when the selector doesn't rehome, what is the state of FINDA? Does it have it's LED on (I mean the LED inside the FINDA).

happykatz, 2023-08-28:
Let me check this again today. I don't have a setup that will allow me to take and store a video of the entire print, but if I get the filament jam with the rehoming error, I will take a video clip that shows the state of the FINDA. 

One last thought: the second time that I encountered this problem, I was able to successfully get the selector to rehome by pressing the middle button on the MMU3. This was odd as pressing "retry" from the interface did NOT initiate the rehome.

happykatz, 2023-08-29:
I was able to reproduce the error today - unfortunately, the error was not recoverable (MMU middle button produced the same result).

In this scenario, the filament load error was triggered with the filament in the correct position, and no obvious stringing on the filament. There was also no obvious blockage causing the FINDA to fire, although its LED was off and it reported its state as FI1. 

Briefly: I tried manually repositioning the selector, and then retrying first with the button and then with the interface "retry". In both cases the selector did not re-home and the filament loaded in air. I also tried tightening the tension screws with the same result. 

The steps I followed were these (photos / videos following each step) :

1) Notified of filament load error (FI1). 
[image]

2) Observed position of selector. 
[image]

3. Observed position / tip shape of other filaments.
[image]

4. Manually rotated the selector to neutral position to clear filament (none obvious, but following steps reported in the bug description). FI reads 1 at this step [image]

5. Cleared the selector with the small hex key. Afterwards, FI reads 0, and LED is on. 
[image]

6. Double checked that FINDA can still fire when filament present using small hex key. FI reads 1, LED off when hex key present. [image]

7.  Initiated a retry using the interface. Selector does not re-home and filament loads in the air (video) 
https://github.com/prusa3d/Prusa-Firmware/assets/71985281/30cd4286-af7e-4170-afdd-60b0ddf5e1ba

8. Intiated a retry using the MMU middle button. Selector does not re-home and filament loads in the air (video) https://github.com/prusa3d/Prusa-Firmware/assets/71985281/196354cf-d656-40af-93cc-52f745c27eb4

9. Checked whether tension screw position was related to failure by tightening the screws. 
[image]

10. Initiated a retry using the MMU middle button a second time, with the same result - no re-home, filament in air (video).
https://github.com/prusa3d/Prusa-Firmware/assets/71985281/4bb56ccb-ae71-4b8d-a282-c613c4134a33

11. Checked whether the problem exists in other than the neutral position - manually repositioned the selector and initiated a third retry with the same result. (video) https://github.com/prusa3d/Prusa-Firmware/assets/71985281/366867ab-dac7-47f4-914d-a5ec7692835d

12. Finally, I concluded that the error was unrecoverable, and took some additional pictures of the idler state with the top of the unit open, and of the build plate.
[image]
[image]
[image]
[image]

_A few notes -_ 

FI was at 0 throughout: there was never a case where the FINDA read the presence of a filament when it should not have (except possibly for the original error). In the latter tests, the idler advanced filament 2 slightly as well as filament 5 - this persisted even when the tension in on the screws was reduced.

[image]

[image]

I am hoping that this is an easy fix. Since the error is unrecoverable, it means that I can lose a multi-hour print mid-way with no recourse.

happykatz, 2023-09-08:
Adding a coda to this. I suspect that the issue that I am seeing is related to the following known issue in the release notes:

`Known issues
During final testing of the FW 3.13.0 and MMU3 3.0.0 developers discovered a rare issue, which we will address in a future release. In case the MMU print stops with an error and needs user interaction a power panic will cause issues to recover from the power panic.`

The description of the issue is somewhat vague (" a power panic causes issues to recover from power panic"), but my reading of this is that that the power draw on the steppers disrupts power to the arduino, causing a reset / partial reset of the arduino and results in the system losing state. 

My experience of arduino devices is that they will prefer the power supply running through USB (covering the case that they are being run in debug mode through a host computer). My assumption is that this behaves like any other arduino device and that by supplying continuous power through the USB port, I will be able to prevent the power disruption even under heavy draw.

I have since connected a 5V power supply through the USB port (just like connecting to a USB host for debugging purposes) and have not had a recurrence of this problem.

I would be interested to know if this analysis is correct, and if the planned upcoming fix described in the notes will address this issue w/o additional power supply.

gudnimg, 2023-09-08:
A power panic is a complete power outage. Meaning the whole printer shuts down while printing. Power panic recovers the XYZ positions and the position in the gcode file to resume correctly.

The known issue mentioned in the release notes is a problem on the printer side and not MMU. If the print is paused due to an MMU error screen, and a power panic occurs. Then recovering the print will fail as the printer will use incorrect XYZ positions. This will be fixed in 3.13.2. 

I think it’s not related to the issue you’re describing.

If only the the MMU is having power supply issues there is unfortunately not much the firmware can do except raise an error screen. You may get a TMC or MCU error screen in that case (or communication timeout). Randomly even.

cozmo14047, 2023-10-13:
So i have this problem where it dosent home but I have suspect that it only does it when the error is on port 4 or 5 I don't know if it's coincidence or not could you let me know what ports its doing it for you many thanks @happykatz

happykatz, 2023-10-16:
Hi @cozmo14047 - I saw this on port 5, but also on port 2 in other experiments. My "workaround" based on the observation from @gudnimg was to eliminate the possibility of an MMU-only power issue by plugging the control port to a continuous 5V power supply. Since that time, I have not had a recurrence of the not-homing-on-error problem; however, I  have also significantly reduced my error count, so this may be co-incidental. Hope this helps!

cozmo14047, 2023-10-19:
@happykatz 

Thanks what do you mean plugging controll port into power whst control port

cozmo14047, 2023-12-26:
> @happykatz
> 
> Thanks what do you mean plugging controll port into power whst control port

The usb port to update the mmu leave that connected to the mmu

cozmo14047, 2023-12-26:
> Hi @cozmo14047 - I saw this on port 5, but also on port 2 in other experiments. My "workaround" based on the observation from @gudnimg was to eliminate the possibility of an MMU-only power issue by [...]

I installed a new controller board and haven't had it since.

closed 2024-03-04 by github-actions[bot]