prusa3d/Prusa-Firmware #4352: [BUG]MK3S (fw3.13.0) with MMU2S (fw3.3.0): M403 causes filament choice

KeanWong, 2023-08-27:
**Printer type** - MK3S
**Printer firmware version** -  3.13.0

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.3.0

**SD card or USB/Octoprint**
  Tested on both, and by sending manually input terminal command via Octoprint

**Describe the bug**
  Sending M403 E0 F0 to the printer causes a filament choice operation to start. It should not (M403 is send 5 times at the start of a non-single print generated in PrusaSlicer, and T? should initiate a filament choice selection, not M403)
  
**To Reproduce**
  Let everything boot/reset. Send "M403 E0 F0" to the printer
  
**Expected behavior**
  Whatever the MMU does with M403 (I assume it is recalling timing settings that vary based on at the filament type?) should occur. It should NOT ask for a filament choice. Asking for a filament choice prevents the next T0 command from loading the first filament (in my example)
  
**G-code**
  ; Send the filament type to the MMU2.0 unit.
; E stands for extruder number, F stands for filament type (0: default; 1:flex; 2: PVA)
; M403 E0 F0
; M403 E1 F0
; M403 E2 F0
; M403 E3 F0
; M403 E4 F0

;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
; select extruder
T0

gbzh01, 2023-09-11:
Same issue here!

gudnimg, 2023-09-12:
Please try put 3.13.2 RC1. I think it is very likely it solves your problem. 

We found that G80 had an extrusion move in the firmware and could cause false FINDA runout event for some users. 

The issue affects MK3S specifically. 

https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1

PxT, 2023-09-12:
RC1 appears to have fixed the problem for me.

gbzh01, 2023-09-13:
Jep, it fixed it! Great job, Prusa team. I am working regularly again.

Best regards,

Gregor Buehrer							
Director
___________________________

￼

		
Sputnique Creative GmbH
Industriestrasse 44
8152 Glattbrugg 
Switzerland

+41 76 586 04 83
www.SPTNQ.com

closed 2023-09-13 by 3d-gussner