prusa3d/Prusa-Firmware #4353: [QUESTION] problem with compile the source code 1.0.6 of the MMU2S

Axolotelking, 2023-08-29:
I am working on my master thesis right now and I am new to programming firmware, for my project I wanted to change the MMU2S for using continuous fibers.

I would appreciate it verry much if you could help me, because I can’t compile the source code 1.0.6 of the MMU2S. The Problem is this specific line “static const uint16_t fw_buildnr =${GIT_PARENT_COMMITS};” in version.h, and I don’t know which number I need to compile this code into a .hex file.

The other possibility would be using the MMU3.0.0 source code but I couldn’t find it.

I hope you could help me with this problem.

3d-gussner, 2023-08-31:
Closed duplicate https://github.com/prusa3d/Prusa-Firmware/issues/4359

Axolotelking, 2023-08-31:
Ok thank you very much

gudnimg, 2023-09-17:
MMU FW 3.0.1 source code is public here: https://github.com/prusa3d/Prusa-Firmware-MMU

So I'm closing this issue as solved for now

closed 2023-09-17 by gudnimg