prusa3d/Prusa-Firmware #4361: [BUG] Thermal Anomaly with REVO HF & Prusament PC-CF

Qmodo, 2023-08-31:
**Printer type** - MK3S+ with Revo HF & 0.6mm ObXidian Nozzle
**Printer firmware version** -  MK3S-E3DREVO_HF_60W_ENGLISH version 3_13_0_6873

**SD card or USB/Octoprint**
SD card

**Describe the bug**
I've consistently had thermal anomalies while printing with higher-temperature materials. Mostly I'm using Prusament PC-CF with a Revo ObXidian nozzle and the High Flow heater.

It hasn't impacted the print or caused it to pause yet so it's mostly just inconvenient noise.

I'm currently printing with a 0.6mm nozzle however I've had the same thing happen while using 0.4mm, so that doesn't seem to be a relevant factor.

**To Reproduce**
1. swapped to the listed hardware
2. loaded PC-CF (I seem to get the anomaly here fairly regularly)
3. Recalibrated the thermal model
4. Started a print
5. The error happens most often while the printer laying down plastic quickly on the infil. It doesn't seem to happen on the slower features.
  
**Expected behavior**
Expected behavior would be not getting the error after recalibrating.
  
**G-code**
[Wheel_0.6n_0.15mm_PC_MK3S_5h15m.zip](https://github.com/prusa3d/Prusa-Firmware/files/12490707/Wheel_0.6n_0.15mm_PC_MK3S_5h15m.zip)

3d-gussner, 2023-11-15:
@Qmodo The PID and thermal model calibration on your setup is probably made to default temperatures and at the higher temps it may behave differently and "warn" more often.

I am using different PID and TM values for that high temperatures in my PrusaSlicer profiles

1.  Get default PID and TM values with `M503` and copy these somewhere you gonna need these later for the profiles
    1. `M301` are the PID values for the hotend
    2. `M304` are the PID values for the bed
    3. `M310` are for Thermal model
2.  Run hotend PID tuning for the high temperatures
    1. `M303 E0 S275 C5` to run the PID tuning for PC temp 275°C
    2. save the hotend PID values with `M301 P<P value from above> I<I valuefrom above> D<D value from above>` to RAM
    3. Optional run `M303 E-1 S110 C5` to run the PID tuning for the PC bed temp 110°C
    4. save the bed PID values with `M304 P<P value from above> I<I valuefrom above> D<D value from above>` to RAM
    5. copy the/both PID values again you gonna need these later for the profiles
    6. Run LCD TM calibration again. Verify new values with the results of 1.3
    7. Run test print with serial monitoring and watch for the `TM error:` messages if the PID tuning improves the behavior
 3. Add the PID tuning settings to PrusaSlicer profile
    1. add the gcode from line 2.2 (optional 2.3)  to your PrusaSlicer -> Printer Settings -> Start G-code
    2. add the gcode from line 1.1 (optional 1.2) to your PrusaSlicer -> Printer Settings -> End G-code to reset it to default
 4. If you still have `TM error:` warnings after the PID tuning for high temp
    1. Find the highest warning `TM error: <value>: 0.8500` it should not exceed the TM Error value [1.23](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/thermal_model/e3d_REVO_HF_60W.h#L16)
    2. send `M310 W<value you found>` to save it to RAM
    3. start a test print again until you get NO warnings with the new W value that works for you
 5.  Add the `M310 W<value you found>` gcode from 4.2 to your Start G-code and the default `M310 W0.83` to the End G-Code to reset to default again.

The TM Warning and Error values should be kept as low as possible to detect issues as fast as possible.

I hope this helps.

3d-gussner, 2023-11-29:
Closing as no response in  two weeks.

closed 2023-11-29 by 3d-gussner