prusa3d/Prusa-Firmware #4364: Unable to change steel sheet Z-offset for existing sheets after updating to 3.13.0 due to freeze when asked if filament is loaded.

ddbond, 2023-09-01:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

gudnimg, 2023-09-02:
This issue is fixed and will be released in 3.13.2

ddbond, 2023-09-02:
Thanks! For the quick response!!!

EeroFluge, 2023-09-10:
This issue was reported back in early August.  [BUG] FW 3.13.0 Cannot perform first layer calibration #4300 

Why is this not in the 3.13.1 build???   Still broken in 3.13.1 !

barrem95, 2023-09-12:
> This issue is fixed and will be released in 3.13.2

When should version 3.13.2 be released?

gudnimg, 2023-09-12:
@barrem95 it's out now https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1 :)

gudnimg, 2023-09-17:
Closing this issue as solved.

closed 2023-09-17 by gudnimg