prusa3d/Prusa-Firmware #4367: [BUG]<Enter comprehensive title>

staspika, 2023-09-03:
**Printer type** - MK3S+
**Printer firmware version** - 3.13.0

**MMU upgrade** - none
**MMU upgrade firmware version** - n/a

**SD card or USB/Octoprint**

  SD card is taken out and nothing is connected to USB port

**Describe the bug**

After starting thermal model calibration, one of the following usually happens (it seems random which one):

1. At some point the printer emits scratching noise and reboots. After reboot, LCD usually shows following message: "Firmware crash was detected. Contact support".
2. Sometimes (although I have no MMU and it is disabled in the settings) LCD shows following message: "MMU NOT RESPONDING" and proposes to either reset or disable MMU.
3. Although all of the Setup Wizard up to that point is completed, after reboot the printer invites to start Setup Wizard from scratch.
  
**To Reproduce**

  Run complete factory reset. After reboot, run the Setup Wizard as usual. Get to the thermal model calibration step. Choose "OK" and wait as the printer attempts to calibrate the thermal model. Wait for the scratching noise, reboot and a message about firmware crash.
  
**Expected behavior**
  Thermal calibration should finish without problems.
  
**Video**

Here is one video. I have more, but the size exceeds the 100 MB quote.
https://github.com/prusa3d/Prusa-Firmware/assets/26161325/107a58ef-667e-406d-81c0-d1ca00dd6e2b

staspika, 2023-09-04:
After downgrading to 3.12.2 or 3.12.1, I am unable to reproduce this exact behaviour. After upgrading again to 3.13.0 the same unpredictable behaviour is back.

vadim-ryazanov, 2023-09-12:
Same (or similar here) with REVO flavour: E3DREVO_fw_MK3S_MK3S+_3_13_1_6876.hex
After flashing firmware and trying to run Thermal Model Calibration, printer starts to heat up the nozzle, screams and quits after reaching around 50 C on the first step of the calibration.

3d-gussner, 2023-09-14:
@vadim-ryazanov Please contact E3D support. The issue is probably that you have E3D v6 values in the EEPROM and flashing the new E3DREVO_fw_MK3S_MK3S+_3_13_1_6876.hex doesn't reset these.

mmcglumphy, 2023-09-15:
Doing a factory reset before the firmware update cleared those values for me. After that, calibration finishes successfully.

staspika, 2023-09-20:
Not for me :) Also, I stress it once again: I do *not* have REVO on my printer, only a new stock V6 hotend. I replaced the old one as suggested by Prusa Support representative, whom I initially contacted about this problem. This did not fix the problem.

alextremblay, 2023-11-09:
I have an E3D REVO nozzle,
Did a factory Reset
flashed E3DREVO_fw_MK3S_MK3S+_3_13_2_7080.hex
Also experiencing a loud screech and printer reboot shortly after starting thermal calibration (when it gets to about 50-60C)

flashed 3.12.2 (first version with TM support for REVO), same outcome

For now, I've downgraded back to 3.11 so i can get back to a working printer.

If there's any other workaround I should test, please let me know. I'll re-flash 3.13.2 and try it

closed 2024-01-22 by github-actions[bot]